JMRI/JMRI #3862: UnsatisfiedLinkError: com.pi4j.wiringpi.Gpio on Win7

pischky, 2017-08-02:
JMRI 4.8 tries to load libpi4j.so on Windows 7.
To reproduce install 4.8 on Win7 and try to configure a new connection with "Raspberry Pi Foundation" as mfg.
See attachements.
After that you have to terminate DecoderPro by taskmanger.

pischky, 2017-08-02:
[image]
[session.zip](https://github.com/JMRI/JMRI/files/1192970/session.zip)

rhwood, 2017-08-02:
This is expected behavior if you attempt to configure a Raspberry Pi connection while not running JMRI on a Raspberry Pi. The Raspberry Pi connection type is used when JMRI is running on the Raspberry Pi to gain access to the GPIO on the Raspberry Pi device.

What are you attempting to do?

pischky, 2017-08-02:
It's expected behavior that the program dies when you browse
thru the available options of a drop-down box?

I just want to install LocoNetOverTcp connection but touched
the RPi option with the mouse. Than the program died.

Just try. It's easy to reproduce.

pischky, 2017-08-02:
Sorry may be my first report has been missleading. The program dies already when you touch the option in the drop down box.

rhwood, 2017-08-02:
@pischky Thanks for clarifying that.

We need to protect against accidentally selecting the RPi connection on non-RPi systems. This problem is reproducible on Windows and macOS systems (and probably non-ARM Linux systems).

pischky, 2017-08-02:
I have not looked in the sourcecode, but may be you can add an exeception handler in
jmri.jmrix.JmrixConfigPane.updateComboConnection(JmrixConfigPane.java:330) and ignore the selection when something is thrown.

rhwood, 2017-08-02:
Thank you for the suggestion, but we really should not fix this issue within the jmri.jmrix.pi package, as this is a manifestation of being unable to create a Raspberry Pi connection, and there are some other cases where we could find JMRI attempting to do so on the wrong platform.

SuzieTall, 2017-09-27:
This happens on a Raspberry Pi too.

rhwood, 2017-09-27:
@suzietall is this happening in nightly builds or in a specific version?

SuzieTall, 2017-09-27:
4.8

New install of latest OS and JMRI latestproduction version and running PanelPro as root. As soon as Raspberry Pi Foundation is selected it closes.

rhwood, 2017-09-27:
@Suzietall can you verify this is or is not an issue in 4.9.4?

SuzieTall, 2017-09-27:
Unfortunately not as I do not know how to install any version other than the current production. If you can point me to instructions I will have a go.

pabender, 2017-09-27:
Sent from my iPad

> On Sep 27, 2017, at 7:12 PM, SuzieTall <notifications@github.com> wrote:
> 
> 4.8
> 
> New install of latest OS and JMRI latestproduction version and running PanelPro as root. As soon as Raspberry Pi Foundation is selected it closes.
> 

Are you actually running as root or running using sudo?  There is a difference, and the library checks for it.

Paul

SuzieTall, 2017-09-27:
I used sudo

pabender, 2017-09-28:
Sent from my iPad

> On Sep 27, 2017, at 7:43 PM, SuzieTall <notifications@github.com> wrote:
> 
> I used sudo
> 

Ok, try it again by logging in as root.

Paul

DRChess, 2017-09-28:
This is the Raspberry kernel 4.8 and 4.9 reporting Hardware BCM2835 to the /Proc/cpuinfo file
that wiringpi uses to verify what pi version is running. My suspicion is that JMRI is using a working directory that has an old wiringpi version. To see this error more clearly open a terminal window and change directory to JMRI and "sudo bash DecoderPro", select connection preference and select Raspberry Foundation. The error will be in the open terminal window. If you "gpio -v"  and running out side of JMRI everything is good. This is why I think JMRI scripts are controlling the working path to wiringpi.

DRChess, 2017-09-28:
Found the problem. JMRI was compiled using an old jar file that is over a year old. In the LIB directory is  pi4j core code from 2016 version 2.0 needs to be 2.33 or 2.44 released. Needs to be recompiled with the corrected version. Readme docs says these are the jar files to recompile but these wont work for Raspbian kernel 4.9.35. Could someone correct this and re-release a working pi version of JMRI.

pabender, 2017-09-29:
Where are you finding those erosion numbers?

The latest production version of Pi4j is version 1.1, released in July 2016 ( see http://pi4j.com ).  This is the version we distribute with JMRI, and the library we use when building.

Paul

pabender, 2017-09-29:
Also, the latest production version does work on the Pi, but I haven't updated any of my Pi's to Raspbian Stretch yet.

rhwood, 2017-09-29:
See Pi4J/pi4j#349

pabender, 2017-09-29:
According to the issue @rhwood linked to above, you can tell pi4j to dynamically link to whatever version of wiringpi is installed locally by setting the system variable:

pi4j.linking=dynamic 

Randal, is that something we can put into the jmri.conf file as workaround until pi4j is updated?

rhwood, 2017-09-29:
> According to the issue @rhwood linked to above, you can tell pi4j to dynamically link to whatever version of wiringpi is installed locally by setting the system variable:
> 
> pi4j.linking=dynamic
> 
> Randal, is that something we can put into the jmri.conf file as workaround until pi4j is updated?
> 
This is something users can add to that file to work around this issue.

DRChess, 2017-09-29:
pi4j 1.1 also has the error BCM2835. You must use the snapshot 1.2. I have been working on this since raspberry kernel 4.8 came out trying to dynamic link the lib to JMRI. I came up with the idea that gpio was somehow wrapped to JMRI. With JMRI production version 4.8, the file in question is found in the JMRI/lib/pi4j core.jar the archive manager list wiringpi directory then list the modules as 2.0 version which is correct for pi4j 1.1 package. Gordon fixed this in December 2016. July 2016 is too old.

pabender, 2017-09-29:
@DRChess,

You are missing our point.  We do not distribute non-production versions of the support libraries we use.

Pi4j has a static link to wiringPi compiled in.  JMRI itself does not link to the wiringPi library.

The workaround to the issue you are seeing , as provided by Pi4j, is to
1) install a newer version of wiringPi on your system.
2) set the system variable pi4j.linking=dynamic

That will allow Pi4j 1.1 to use the version of wiringPi you install, and not the version it is statically linked against.

I'm in the process of updating one of my RaspberryPis to raspbian stretch.  I'll work on documenting this process further later today.

DRChess, 2017-09-29:
@pabender 
Paul,
I understand the need to distribute software in a production release but the latest NOOBS that is distributed by Raspberry Foundation has the command gpio -v to show 2.44 so wiringpi must be installed from Gordon's software. I have tried putting pi4j snapshot.1.2, setting the system variable for dynamic, and placing the setting in the desktop shortcut PanelPro,  but it has not worked. When I saw SuzieTall statement I thought it was time for me to chime in and believe me what's been going on with pi4j is too, too political for me (they know it's broke). I hope this issues is finally going to be fixed.
By the way. Kernel 4.8 was released without the proper stability checks and is why it was jumped to 4.9.35 according to articles I was reading. Kernel 4.9.35 seems to running smoother. I want get my JMRI platform running again which has not been able to used during last several months when the problem first broke. I'm just glad in the last 2 days finally someone on the JMRI team has noticed it. I think JMRI software is by far the best and I use it in many demonstration in the Pittsburg area showing the Pi doing block detection.

Signing off, the software is in good hands now.
Thanks again, 
Dave :)
PS I'll be glad to test.

pabender, 2017-09-29:
@DRChess @SuzieTall 

I just ran an experiment on my pi with stretch installed.  I did see the failure, and I was able to get JMRI to run without any changes to the distribution (I was using a current daily build).

Here's the exact command sequence I used to get this running again:
`sudo apt-get install wiringpi`
`echo "default_options=\"-Dpi4j.linking=dynamic\""`
'export WIRINGPI_GPIOMEM=1'
'./PanelPro'

(the export of WIRINGPI_GPIOMEM is optional, but since I wasn't running as root this time...)

Want proof?  I'm attaching the output of script showing first the failure and then the successful execution.

[typescript.txt](https://github.com/JMRI/JMRI/files/1345560/typescript.txt)

SuzieTall, 2017-09-29:
I think I must be missing something because my distribution already has 2.44 installed (first line does not install anything because it is already there at correct version), and the 3rd line throws an error "command not found"

rhwood, 2017-09-29:
I think the second line should be:

echo "default_options=\"-Dpi4j.linking=dynamic\"" > ~/.jmri/jmri.conf

You don’t want the single quotes around lines 3 & 4 (they are there to show the commands).

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

pabender, 2017-09-29:
Thanks @rhwood.  I fixed the commands in the previous comment

SuzieTall, 2017-09-29:
That's better. Works fine now. Thank you.

DRChess, 2017-09-30:
@pabender 
Paul,
As I have said in the past you guys are the best.
I'm up and running with full gpio support.
Thank You, Thank You very much Dave

pabender, 2017-10-01:
@SuzieTall and @DRChess Thanks for confirming this works for you.

I've documented the workaround in our documentation.  PR #4193 is putting that into place.

closed 2017-12-10 by bobjacobsen