JMRI/JMRI #3878: Add new item check

silverailscolo, 2017-08-05:
(Continued discussion from PR #3876)
There is some duplication in the regex needed for validating the user entry directly in the ValidatedTextField and what is used in the NormalizeSystemName code. However, the latter looks at a different (longer) string including the connection (system) prefix and bean type. I can think of the following steps to improve on this:

- [ ] As is, the regex variable in the NormalizeSystemName code can not be provided by itself via the manager (easy to fix with a getter).
- [ ] Additionally, to be of use for validation on the Add new item pane, the first 2 items in the regex need to be stored separately, and only joined for use inside the Address normalization methods. Also not difficult. 
- [ ] By the way, not all jmrix connection types have an Address class. Entry validation on the Add pane will use a default regex. Perhaps add normalization to more types?
- [ ] To further improve the user interaction, make the name entry validation error messages more specific (eg. Locate the faulty char, suggest a fix). Any effort here pays off in less support later and happy users.
- [ ] When that's in place, provide them not to the Console but to the pane interface.

bobjacobsen, 2017-08-06:
@silverailscolo: I don't understand what you're trying to do, I think.

It looks like you're trying to do something more complicated than "Is this an OK (system or user) name?".

Is that correct?  if so, what's it trying to do?

The [NormalizeSystemName](http://jmri.org/JavaDoc/doc/jmri/Manager.html#normalizeSystemName-java.lang.String-) and [normalizeUserName](http://jmri.org/JavaDoc/doc/jmri/NamedBean.html#normalizeUserName-java.lang.String-) methods do several things:
- If there are small problems that we've agreed to just fix in names (extra spaces at the end, for example), they just fix those for you.
- But if the string isn't a valid name, they tell you that too.

What else do you need?  What does exposing a regex provide?

silverailscolo, 2017-08-06:
@bobjacobsen did you try the Add new turnout pane, yet? You will now see a tooltip with valid formats for your system.
What I see could be improved is not (only) by magic in the background normalize the name after the user clicks [OK], but before that while typing in the hardware address field, receive clear feedback. Until.swing.ValidatedTextField is a nice trick we have for that. I looked at Java Formatters but the entry patterns for the hardware address are too diverse across systems, without a 1 to 1 match. The sort of thing they do provide is spacing - as in phone numbers or hex numbers - and ignoring entry of invalid characters. For instance, NormalizeAddress will catch a $ sign later, but it seems more instructive to beep while typing a $ (and explain why that is, of course).

bobjacobsen, 2017-08-06:
@silverailscolo - I played with the Turnout and Sensor tables a bit.

I think I see what you're trying to do. I think you're going to have a hard time doing keeping all the various forms of information consistent but if you're committed to doing that work, that's great.

That concern is because this approach now has at least 5 different sets of information that have to be kept consistent:
- The actual parsing code in the manager
- The validation (which might actually be more than one thing, don't quite understand that yet)
- The tooltip
- And the feedback-on-entry code (which solves a different problem than a validating regex and won't be the same)
- The web pages

(And five is not counting the multiple web pages that carry overlapping information, translations of the tooltips, etc)

It's going to be difficult to keep all this consistent because it's in too many places. 

Already, we're seeing signs of how hard this will be to make it all correct:  The tooltips on Internal turnouts say they need to be numbers: They don't, and many layouts use letters there (Both `ISCLOCKRUNNING` and `ISXMLIOHEARTBEAT` are auto-created; check the Names page and test XML files for additional)  C/MRI tooltips refer to nodes that don't (or might not) exist which will confuse people:  The examples of valid names should be, well, valid.  

I really do appreciate the effort you're putting into this. Code wins, and if this is the way you want to do it, more power to you.  But I think in the long run this is going to have to be simplified, and it seems like it would be better if we could start with that now.

As to beeping while typing:  I would be _amazed_ if you can get that to work well.  The two human factors guides I'm familiar with say to not attempt it unless you can do it extremely reliably.  People don't like it when sometimes they get a beep on invalid and other times they don't. That uncertainty has a cost people consider higher than the gain from the occasional early prompt: The program is considered "annoyingly stupid" instead of "almost smart".

In lots of cases, checking validity of the names can't be done character-by-character: 20 is valid, 201 is not, 2011 is valid (true of some C/MRI layouts)  Which character is invalid in 201?

Having a consistent, clear way of handling the entry once it's entirely entered seems much more important to me.  I think that's even more important than the tooltip:  Show examples on the dialog box when an invalid string is entered, because people will occasionally look at that.

silverailscolo, 2017-08-06:
Thanks for considering and reflecting. 
The 'beep' is meant as 'informing the user'. In many places we use a status bar on the pane as an elegant way for that; producing an actual sound is not the essence.
ValidatedTextField activates as soon as the focus leaves the field (and no earlier). Per character checking indeed is impossible to achieve for simple if-then-else code because we allow multiple entry formats for one connection type. Nor does it assess the numerical value, just the number of digits.

Internal items are a bit hard to find in the code, as the Javadoc says they're not connected to Internal connections (pun intended), which I expected to find in jmrix.internal. If what you describe is possible for naming Internal items, I can document that and customize the tooltip for Internal items.
i18n of the term "Internal" in the System connection combobox proved tricky, but it sure would look better in a translated GUI.

Of course, keeping stuff in sync is hard work behind the scenes. From experience, it's a little bit easier if one place is marked as "origin". So I hope to connect getEntryRegex to NormalizeSystemName and add useful `@link` pointers. When we get the item entry correct, that should prevent a range of later complications.

What I will to add to my list, is adjusting NormalizeSystemName to accept a customized system connection prefix, which the Connections preferences tab allows. We cannot assert that L, and only L, is for LocoNet, which the current regex does. But accepting an empty string as the type letter is better prevented at all cost.

bobjacobsen, 2017-08-06:
Here's another example of what I'm saying:  There are now multiple cases of code like this:
```
        if (turnManager.getClass().getName().contains("ProxyTurnoutManager")) {
            jmri.managers.ProxyTurnoutManager proxy = (jmri.managers.ProxyTurnoutManager) turnManager;
            List<Manager> managerList = proxy.getManagerList();
            String systemPrefix = ConnectionNameFromSystemName.getPrefixFromName(connectionChoice);
            for (int x = 0; x < managerList.size(); x++) {
                jmri.TurnoutManager mgr = (jmri.TurnoutManager) managerList.get(x);
                if (mgr.getSystemPrefix().equals(systemPrefix)) {
                    range.setEnabled(mgr.allowMultipleAdditions(systemPrefix));
                    // get tooltip from ProxyTurnoutManager
                    addEntryRegex = mgr.getEntryRegex();
                    addEntryToolTip = mgr.getEntryToolTip();
                    log.debug("T add box set");
                    break;
                }
            }
        } else if (turnManager.allowMultipleAdditions(ConnectionNameFromSystemName.getPrefixFromName(connectionChoice))) {
            range.setEnabled(true);
            log.debug("T add box enabled2");
            // get tooltip from turnout manager
            addEntryRegex = turnManager.getEntryRegex();
            addEntryToolTip = turnManager.getEntryToolTip();
            log.debug("TurnoutManager tip");
        }
```
and there will inevitably be more. All of that has to be maintained. 

And I still don't see what's added compared to just asking a manager (which may or may not be a ProxyManager) to say whether the name is OK or not.

silverailscolo, 2017-08-07:
Motivation can be found in https://www.smashingmagazine.com/2009/07/web-form-validation-best-practices-and-tutorials
"By combining [entry validation after submitting as well as before/during entry] we can get the best of the two: fast response, more secure validation and better user experience."
The tip to include the hints on the entry pane itself in a subtle style and not on a separate page or dialog would be addressed by using a status field. The tip to highlight the problem field (maybe even mark valid entry success) is just what the ValidatedTextField does.

rhwood, 2017-08-08:
Just a note about the Smashing Magazine article: They are referring to specific situation where the computer validating the input before/during entry is potentially half way around the world from the computer validating the input after entry, where the input validation can take dozens of seconds after submission.

This is *not* a situation being addressed by this validation in the JMRI tables. The connection managers can validate the input by calling normalizeSystemName on every keystroke effectively immediately, within single milliseconds of the keystroke.

Note that tooltips should not be displayed during keyboard entry (they should be where the mouse is, so they may obscure the input, and moving the mouse absent user input can cause accessibility issues, so displaying a tooltip to validate keyboard entry absent other indications will likely not help either.

bobjacobsen, 2017-08-08:
I think doing a validation is fine.  Much in favor of that.  

But I _still_ don't see what exposing a regex buys you.  Why can't the validation be done by calling one of the existing methods for doing validation? We can do that once, in a new kind of text field, and have all the details handled in a contained way instead of being spread across dozens of classes.

devel-bobm, 2017-08-08:
I coded the <ValidatedTextField> functionality in the 2011-2012 timeframe in support of the LocoNet "Duplex Group Configuration" tool.  I had a few different configuration fields to display and allow entry, with a rather diverse set of validity requirements.  So in the spirit of re-use, I coded a method which uses a regex as a parameter for the more-complex validity checking.  Looking back on this, "spirit of re-use" may map more to "lazy coding" than "object-oriented programming".

So far as I know, this is the only use of the "ValidatedTextField" until this recent effort.

While the "Duplex Group Configuration" tool uses two different regex's, and other ValidatedTextField forms which don't require regex parameters, it could have been done as base-class plus multiple derived sub-classes.  

If that is a primary "bone of contention", then the ValidatedTextField could be changed to use better o.o.p practices, without a regex as a parameter.

geowar1, 2017-08-12:
As much as I love me some regex… "valid" could include "does it exist". In other words is a name valid if the manager its prefixed with can't map it to something that actually exists?

I think that rather than have a manager return a regex just have it provide a method to validate the names. The ValidatedTextField could match against the appropriate manager prefix and then call its validator method (if it exists) or fall back on the (build in?) regex otherwise.

Plus, as far as feedback goes I've been changing the background color of the (editable) comboboxes to reflect if turnout and block names are valid (yellow) and exist (green)… or not (red). I find this much less intrusive than beeps.

devel-bobm, 2017-08-12:
Be aware that the ValidatedTextField class was initially defined for inputting values for JMRI configuration of something that does not directly relate to any JMRI bean.  As such, the original use cannot use a manager for validation.

silverailscolo, 2017-08-15:
No beeps. I've got something in the works using a dedicated CheckedTextField class including an InputVerifier that directly asks a (Turnout)manager to validate the entry using ``ValidSystemNameFormat()``
This adds user interaction (red arrows, compare the prior dotted lines). Where available, this will point to an xAddress class.

[image]

silverailscolo, 2017-08-19:
See PR #3955.

silverailscolo, 2017-09-19:
PR #4141 temporarily disables the Add New Item ``valid()`` check to silence logs while working on checkbox 4 and 5.

geowar1, 2017-09-19:
>  I've got something in the works using a dedicated CheckedTextField class including an InputVerifier that directly asks a (Turnout)manager to validate the entry using ValidSystemNameFormat()

If it can be used to select an existing turnout then perhaps you should use a JmriBeanComboBox instead?

geowar1, 2017-09-19:
Ah, read the title… it's for "new" items (so "no" to select existing…)… Nevermind, that's very different.

silverailscolo, 2017-09-19:
Correct, you should be able to use the ``ValidSystemNameFormat()`` check from anywhere, but ``New ItemEntry`` focuses on an application during creation of new items (in the tables GUI).
OT: I like the way you applied the green coloring as feedback in the LE toolbar item boxes.

geowar1, 2017-09-19:
On Sep 19, 2017, at 12:59 PM, Egbert Broerse <notifications@github.com> wrote:
> OT: I like the way you applied the green coloring as feedback in the LE toolbar item boxes.
Yeah, I’d still like to add ether help or tooltips to explain the colors: Green is an existing device; red is nonexistent (validation on) and yellow is nonexistent (validate off).

Note: The red and green colors are pastels; yellow is “borrowed” from "Sticky Notes":

        if (isEditable() && !comboBoxText.isEmpty()) {
            if (null != getNamedBean()) {	// <== existing bean
                c.setBackground(new Color(0xBDECB6));   //pastel green
            } else if (_validateMode) {
                c.setBackground(new Color(0xFFC0C0));   //pastel red
            } else {		// device doesn’t exist and not validate mode
                c.setBackground(new Color(0xFDFD96));   //pastel yellow
            }
        } else {
            c.setBackground(new Color(0xFFFFFF));   //white (pastel grey?)
        }

Feel free to plagiarize… ;-)
-- 
Enjoy,
George Warner
eMail/iChat: <geowar1@mac.com>

silverailscolo, 2018-11-07:
Name entry validation as of 4.13.4 seems pretty solid, so closing this issue.
The system connection memos - and sometime a separate SerialAddress class - take care of ``validateSystemNameFormat(String address)``, called through the respective manager.
Please follow up for any new connections and new valid entry formats.

closed 2018-11-07 by silverailscolo