JMRI/JMRI #3897: Recent commit to automatically create META-INT files breaks Eclipse IDE

DanielBoudreau, 2017-08-09:
Eclipse doesn't update the META-INT files correctly during a build using the internal Eclipse compiler. This latest commit makes the Eclipse IDE almost useless, we now have to use ant to build a jar file to test with ANT rather than the internal compiler.  This increases the time to make a change and test from seconds to almost a half an hour.  Running Eclipse in this mode is pretty much useless for me, and way too painful for any real work.  I can deal with it for simple bug fixes, but I'm pretty much sidelined until someone either figures out how to fix the Eclipse compiler or we go back to the way the META-INT files were managed before this commit.

KenC57, 2017-08-09:
I'm finding it makes trying to track down a problem with the CTI/Acela
connection impossible. Things now happen in the voodoo libraries and can't
be break pointed to see what is being found or not found correctly. It used
to be easy but this has made it a fruitless effort for me.

-ken c

pabender, 2017-08-09:
One of the eclipse users may want to look at this page:

https://wiki.jenkins.io/plugins/servlet/mobile?contentId=69271819#content/view/69271819

My reading of that is it is possible for eclipse to handle the notation processing and copy the information appropriately, but it has to be configured in eclipse.  I haven't gone through this process myself yet ( and I don't think my eclipse is currently setup to build JMRI)

bobjacobsen, 2017-08-09:
See issue #3899.  Until that's resolved, I'm taking myself off this Issue, and for reasons discussed below, muting this Issue so I don't get emails about it.

I agree that people who want to use Eclipse with JMRI should be able to. I agree that the recent change broke that, and hence is subtractive.  That situation needs to be resolved.  

Several ways of doing that have been proposed that range from making Eclipse work with annotations, to various ways of allowing Eclipse to do everything-but-annotation-processing.  My personal favorite is to set things up so there's default META-INF contents in the repository that can be used without regeneration by platforms that can't do that regeneration. Others may want to do other things.

The problem is that I, at least, can't work on this because #3899 prevents me. I've put more time than I really have into understand the Eclipse setup, and I have made negligible progress (see jmri-developers thread titled "More evidence that I just don't get Eclipse" and note that the timestamps are spread across _two_ _days_).

That experience shows there's a more basic problem here:  Eclipse support is _not_ something that's made available to JMRI users and/or developers. Apparently only people with already working setups can use JMRI with Eclipse. The bigger subtractive change is that the JMRI support for building with Eclipse hasn't kept up with changes.  Perhaps that's Eclipse changes, maybe JMRI changes, likely both.  Maybe the original Eclipse-to-JMRI connection wasn't robust,  or there's some other cause.  Whatever.  

So I'm happy to help with this Issue.  (And all the other things that people seem to expect that I will gladly do for them just because their hobby seems to be telling me what to do with my hobby)  But first the Eclipse users need to bring something to the table, and that's a working set of instructions that I can follow.

Will definitely keep an eye out for PR #3899 being closed.

bobjacobsen, 2017-08-10:
Instructions for using Eclipse via Ant are now available at [EverNote play by play](https://www.evernote.com/shard/s33/sh/79c94bac-a1c7-41e1-a5c0-e6e40ebe67fb/2be728b165d55a314872b83e17f68e82). See Issue #3899. That's not a solution, as some people want to use native Eclipse instead of Ant.  But it does let you reliably build and run JMRI  under Eclipse.

bobjacobsen, 2017-08-10:
After some additional use, the solution in #3899 seems quite reliable _if_ one starts with cloning a new Git repo.  Apparently use of previous versions of Eclipse can leave behind configuration information somewhere inside the development directory that prevents Ant from working in combination with Eclipse. I don't know where or what that information is, but it's pretty clear there's something there.

So now, having resolved #3899 on my own, I don't see a problem here.  "Recent commit to automatically create META-INT files breaks Eclipse IDE" is, as far as I can see, not a correct summary of the situation:  Once I have Eclipse working with Ant, it's working fine to create the META-INF files.

DanielBoudreau, 2017-08-10:
Bob,
I agree with you, Ant does create the META-INT files correctly under Eclipse.  And you can also launch PanelPro directly from Eclipse once those files have been created.  The issue is when you make a change to a file, it triggers an internal build, and the META-INT files (not sure what is really going on) seem to get modified or disappear. This means I have to use Ant again to restore the META-INT files, and that takes a long time.  That's the issue, I can't make changes and test in a reasonable time.

So for now I'm sidelined, I can fix minor bugs using Eclipse and Ant, but the IDE is broken from a development perspective.

bobjacobsen, 2017-08-11:
@DanielBoudreau I understand this isn't good for you. Could you (or somebody in a similar situation) _please_ put a bit of time in to helping me recreate this problem? I really don't think that's too much to ask.

KenC57, 2017-08-11:
Dan,

 

I've been using a script I put together at the hint from another Eclipse
user. It read protects the services directory after a clean and build. It
seems to hold up ok. And I can let Eclipse do the builds it wants during the
day ok. I only run the script if I think I've changed anything that would
have changed those files. So like after pulling a bunch of code down that
had anything to do with service. I figure this will stop gap until somebody
here figures out why/how to have Eclipse do the right thing.

 

Here is the example:

kcameron@KenC-5 ~/git/JMRI-eb

$ cat lockMeta.csh

#! /bin/sh

# build and freeze the META services

# Ken C

 

echo "unlocking META"

chmod a+w ./target/classes/META-INF/* -R

ant realclean

ant debug

echo "locking META"

chmod a-w ./target/classes/META-INF/* -R

ls -l ./target/classes/META-INF/services

 

kcameron@KenC-5 ~/git/JMRI-eb

$

DanielBoudreau, 2017-08-18:
How do I run this script?  Running on Windows 7.

DanielBoudreau, 2017-08-18:
Ken, would it be possible to create two new targets in the Ant Build.xml, one to lock the meta files, and another to unlock them?

DanielBoudreau, 2017-08-18:
Made the modifications to build.xml to lock and unlock the meta files.

mstevetodd, 2017-08-18:
All,
I may have found a solution. 
In Eclipse Neon.3, under Java, Compiler, Building, Output Folder, there is a flag "Scrub output folders when cleaning projects", which was enabled. I unchecked it.
I am now able to leave "Build Automatically" enabled so that my changes are available without having to run an Ant build. (I believe this uses the incremental compiler in Eclipse)
Of course, that means the Eclipse Clean no longer cleans everything, so after a Git Pull, I run the Ant Clean and Debug targets to freshen up everything.
Hope this helps someone else,
  SteveT

KenC57, 2017-08-18:
While working down to that option, I also see a section that talks able
annotation processors and for linking processer jars. This may be related to
how to make the Eclipse compiler also do the right annotation processing.
But I'm not real clear on how all those parts go together, so I'm not ready
to touch those yet.

-ken c

mstevetodd, 2017-08-23:
PR #3980 puts the now-disabled "Scrub output folders when cleaning projects" flag into project-specific settings. The Ant build is still needed (as before) after a Pull, but this allows the "Build Automatically" flag to be enabled, which lets the incremental compile work. For the Ant Build, I specify the Clean, Debug targets.

DanielBoudreau, 2017-10-24:
I find that for some commits, I need to scrub the output folder for the Eclipse build to be successful. So most of the time this fix works, but I don't feel we have the correct solution for this problem.

rhwood, 2017-11-04:
I believe that #4321 fixes this.

closed 2017-11-04 by rhwood