JMRI/JMRI #3899: Can't get Eclipse to work with older JMRI releases

bobjacobsen, 2017-08-09:
Following the instructions on the [JMRI Eclipse page](http://jmri.org/help/en/html/doc/Technical/Eclipse.shtml) with Eclipse Oxygen does not result in a working Eclipse+JMRI setup.

The instructions are from a previous version and it's not possible to follow them precisely.  Nor is it possible to go through them twice the same, as Eclipse seems to be saving state. And so far I haven't figured out how to remove all that state, as Eclipse seems to be saving it all over the place. 

Using multiple best guesses, I get to a basically blank Eclipse setup:
[image]

Selecting the Run menu, then External Tools, then build.xml:

[image]

runs the default Ant target, which builds and runs JMRI efficiently and correctly.  But I suspect that's not what people want.  Note that I can only run that deeply nested, non-intuitive thing.  If I select the top-level "Run" from the "Run" menu, I get an error dialog:

[image]

The Build operations are greyed-out.  There are no choices in the "external targets menu":

[image]

It's quite possible that my Eclipse Oxygen installation has something wrong with it, but it's a recent clean install (macOS) from the Eclipse.org download page.

bobjacobsen, 2017-08-09:
Most recent attempt: 
- Open Eclipse, see window like 1st above
- There's nothing to "Close" to clean up
[image]
- Use "Open Projects From File System" to select an existing JMRI repo clone:
[image]
- Get to yet another blank window

bobjacobsen, 2017-08-09:
From a brand new workspace.
- Launch Eclipse
- Instead of accepting the proffered workspace directory, click browse, then create a new directory somewhere, Open
- Then click Launch

[image]

- File menu > import 

[image]

- Select: Git, projects from Git
- Select: existing local repository
- Click add, then click browse, find JMRI repository

[image]

- check the box
- Click: Finish
- Gives you a chance to add another?  Click next
- “Select a wizard to use for importing projects”

[image]

Now what?

bobjacobsen, 2017-08-09:
This is an attempt to show what happens when I try to setup from scratch:
https://www.evernote.com/shard/s33/sh/79c94bac-a1c7-41e1-a5c0-e6e40ebe67fb/2be728b165d55a314872b83e17f68e82

pabender, 2017-08-09:
Bob,

Sent from my iPhone

> On Aug 9, 2017, at 2:59 PM, Bob Jacobsen <notifications@github.com> wrote:
> 
> This is an attempt to show what happens when I try to setup from scratch:
> https://www.evernote.com/shard/s33/sh/79c94bac-a1c7-41e1-a5c0-e6e40ebe67fb/2be728b165d55a314872b83e17f68e82
> 

To answer your "now what" question, run an ant build, which will generate the tmp directory and its contents.  To the best of my knowledge, this only needs to be done when generated code changes.

Paul

bobjacobsen, 2017-08-09:
@PABender:

> To answer your "now what" question, run an ant build, which will generate the tmp directory and its contents. To the best of my knowledge, this only needs to be done when generated code changes.

Thank you.  But I don’t know _how_ to “run an ant build”.  I’ve added some shots of menus and an error dialog to that EverNote URL in the previous note.  Any help on exactly how to “run an ant build” in this state, or how to get to a better Eclipse state, would be appreciated.  But please remember that I really _don’t_ understand Eclipse, so specifics will be helpful.

pabender, 2017-08-09:
Sent from my iPad

> On Aug 9, 2017, at 3:52 PM, Bob Jacobsen <notifications@github.com> wrote:
> 
> @PABender:
> 
> > To answer your "now what" question, run an ant build, which will generate the tmp directory and its contents. To the best of my knowledge, this only needs to be done when generated code changes.
> 
> Thank you. But I don’t know _how_ to “run an ant build”. I’ve added some shots of menus and an error dialog to that EverNote URL in the previous note. Any help on exactly how to “run an ant build” in this state, or how to get to a better Eclipse state, would be appreciated. But please remember that I really _don’t_ understand Eclipse, so specifics will be helpful.

I'll have setup eclipse again to figure that out.  Like you, I don't use eclipse regularly.

There are instructions for running the ant build outside of eclipse on our website, but they are windows centric.

Paul

KenC57, 2017-08-09:
I have to restart my Eclipse from time to time, new git workspace etc. So I
have done that a number of times now. I should likely print off the current
pages and markup what has changed next time I go through the loop. May be
soon. I was surprised that when I stepped up to Oxygen I didn't need to
start over this time. Last, Mars I think, it ended up being simple to start
over.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

bobjacobsen, 2017-08-10:
I've figured out how to hook "external build" targets to the Ant tasks in our usual build process. See the end of the [EverNote play by play](https://www.evernote.com/shard/s33/sh/79c94bac-a1c7-41e1-a5c0-e6e40ebe67fb/2be728b165d55a314872b83e17f68e82).

With this, I think Eclipse is working for JMRI development, including handling of annotations and all related items.

It's a bit of a mystery where Eclipse is storing this configuration information. `git status` doesn't show anything modified. There are lots of new files and directories under `.metadata`, but none of them seem to contain the obvious keywords to search for ("PanelPro", "headlesstest", etc) except perhaps a couple of opaque binary files.

bobjacobsen, 2017-08-10:
Title changed from "Can't get Eclipse to work with older JMRI releases" and comment added to 1st post.

bobjacobsen, 2017-08-24:
PR #3980 changed the default (from JMRI git) settings.

DanielBoudreau, 2017-10-25:
Bob,

I took a stab at correcting the Eclipse notes.  What I don't understand is why you couldn't easily run the ant default build by selecting "build.xml", then "Run As" then "2 Ant Build" from the package explorer.  That brings up the targets on my system allowing me the change the default "debug" to something else.

closed 2020-01-08 by bobjacobsen