JMRI/JMRI #3903: Strange error on AppVeyor

rhwood, 2017-08-10:
Recently, the error message:
```
[junit] *** java.lang.instrument ASSERTION FAILED ***: "!errorOutstanding" with message transform method call failed at JPLISAgent.c line: 844
```
has been occurring on AppVeyor, even on successful builds. Googling suggests this is a StackOverflow error in the JVM, and (were we using gradle) might indicate an instrumentation issue, but beyond that, I haven't a clue.

@pabender do you have any suggestions?

pabender, 2017-08-10:
Can you show me an example build.  I haven't seen this, and I just looked through a few successful builds for it and couldn't find an example.

rhwood, 2017-08-11:
https://ci.appveyor.com/project/JMRI/jmri/build/8458

bobjacobsen, 2017-08-11:
```
  [junit] WARN  - Cleaning up frame "Decoder Pro Wizard" (a class jmri.util.JmriJFrame) from earlier test. [main] jmri.util.JUnitUtil.?()
    [junit] WARN  - Cleaning up frame "Decoder Pro Wizard" (a class jmri.util.JmriJFrame) from earlier test. [main] jmri.util.JUnitUtil.?()
    [junit] WARN  - Cleaning up frame "JMRI" (a class apps.SplashWindow) from earlier test. [main] jmri.util.JUnitUtil.?()
    [junit] WARN  - Cleaning up frame "Test Launch Frame" (a class apps.AppsLaunchFrame) from earlier test. [main] jmri.util.JUnitUtil.?()
    [junit] ------------- ---------------- ---------------
    [junit] *** java.lang.instrument ASSERTION FAILED ***: "!errorOutstanding" with message transform method call failed at JPLISAgent.c line: 844
ci-test-appveyor-coveragecheck:
[jacoco:report] Loading execution data file C:\projects\jmri\jacoco.exec
[jacoco:report] Writing bundle 'AntTestReporting' with 6837 classes
[INFO] Executed tasks
```

The stack overflow seems to be happening right at the end of JUnit's echoing of the stored stdout output. That's well after anything in our code.  It's possible that we're giving it too much output, but that seems unlikely; it's just not that big.

Perhaps we're leaving a thread running that's going crazy during some subsequent step?  We could check for what's still running in the end-processing step of the overall class.

pabender, 2017-08-12:
There's an OpenJDK bug ticket (which is closed) that indicates a similar issue was fixed.
https://bugs.openjdk.java.net/browse/JDK-8061621

but that's OpenJDK and not the official Oracle compiler.

The only other thing I've seen that gives a clue as to what is happening is this blog about a Ruby on Rails project:
https://jolorenz.wordpress.com/2014/08/09/how-to-solve-java-lang-instrument-assertion-failed-erroroutstanding-with-message-transform-method-call-failed-at-srcshareinstrumentjplisagent-c-line/

Which seems to imply that forked test execution can lead to this.

rhwood, 2018-03-16:
OP is no longer concerned.

rhwood, 2020-07-20:
This is OBE as we no longer use AppVeyor

closed 2020-07-20 by rhwood