JMRI/JMRI #3913: Need NSIS 3 on Jenkins server

bobjacobsen, 2017-08-11:
Since PR #3724, we need to have NSIS version 3 on the Jenkins servers to build Windows packages.  Unfortunately, we don't, and the [Packages jobs]() are [failing]().

More unfortunately, the usual methods of installing don't seem to work. (Not even counting that a special version is required)  `apt-get` only has version 2, not 3.  A download of the source and build gives:
```
~/nsis-3.02.1-src$ scons
scons: Reading SConscript files ...
Delete("nsis-11-Aug-2017.cvs")
Delete(".instdist")
Delete(".test")
Using GNU tools configuration
Checking for linker flag $MAP_FLAG... yes
sh: 1: .sconf_temp/conftest_2: not found
Checking for linker flag -s... no
Checking for linker flag $MAP_FLAG... yes
sh: 1: .sconf_temp/conftest_5: not found
Checking for linker flag -s... no
Checking for memcpy requirement... yes
Checking for memset requirement... yes
Checking for linker flag -pthread... yes
Checking for __BIG_ENDIAN__... no
gcc --version
gcc (Ubuntu 4.8.4-2ubuntu1~14.04.3) 4.8.4
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

version
sh: 1: version: not found
Checking for C library gdi32... no
Checking for C library user32... no
Checking for C library pthread... yes
Checking for C library iconv... no
Checking for C library shlwapi... no
Checking for C library z... yes
Checking for C library iconv... no
Checking for C library z... yes
Please specify folder of zlib for Win32 via ZLIB_W32
```

Clearly I'm going to need some help on this. Anybody know how to do it?

The server is:
```
$ uname -a
Linux x...cloud.tagadab.com 3.19.0-25-generic #26~14.04.1-Ubuntu SMP Fri Jul 24 21:16:20 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
```

bobjacobsen, 2017-08-12:
#3914 reverts #3724, removing the requirement for NSIS 3 from master and releases.  

Still, #3724 was a good idea, and it would be good to find a way to build NSIS 3 on Ubuntu, hence this issue.

rhwood, 2017-08-12:
Did you try the instructions at http://nsis.sourceforge.net/Docs/AppendixG.html#build_posix They explicitly call out a need to not just call "scons".

[Fedora](https://src.fedoraproject.org/rpms/mingw-nsis/blob/master/f/mingw-nsis.spec) builds with:

```
scons VERSION=%{version} PREFIX=%{_prefix} PREFIX_CONF=%{_sysconfdir} ZLIB_W32=%{mingw32_sysroot}  SKIPUTILS='NSIS Menu' STRIP_CP=false NSIS_MAX_STRLEN=8192
```
with a dependency on mingw32-zlib (not sure what the Ubuntu equivalent package is).

I can't work on trying to build it until Sunday, hope this helps till then.

bobjacobsen, 2017-08-12:
```
~/nsis-3.02.1-src$ scons VERSION=%{version} PREFIX=%{_prefix} PREFIX_CONF=%{_sysconfdir} ZLIB_W32=%{mingw32_sysroot}  SKIPUTILS='NSIS Menu' STRIP_CP=false NSIS_MAX_STRLEN=8192
scons: Reading SConscript files ...

scons: *** Path for option ZLIB_W32 does not exist: %{mingw32_sysroot}
File "/home/bob/nsis-3.02.1-src/SConstruct", line 200, in <module>
```

See [this page](http://nsis.sourceforge.net/Zlib) suggests pointing to a .dll, which also doesn't work.

rhwood, 2017-08-12:
The following steps create and install makensis 3.01 packages for Ubuntu 14.04 LTS:

```
sudo apt-get install --yes devscripts
mkdir -p ~/nsis-pkg
cd ~/nsis-pkg
dget -ux http://http.debian.net/debian/pool/main/n/nsis/nsis_3.01-1.dsc
cd nsis-3.01/
dpkg-checkbuilddeps
# following are what above command reported required on a new 14.04 LTS Workstation edition
sudo apt-get install --yes debhelper docbook-xsl-ns docbook5-xml html2text libcppunit-dev mingw-w64 scons xsltproc zlib1g-dev
# comment out rule that forces failure
sed -i '210s/^/#/' debian/rules
# add large strings and advanced logging support
sed -i '/^SCONSOPTS =/s/\\/NSIS_MAX_STRLEN=8192 NSIS_CONFIG_LOG=yes \\/‘ debian/rules
dpkg-buildpackage -rfakeroot -us -uc
cd ~
sudo dpkg --install ~/nsis-pkg/nsis-common_3.01-1_all.deb ~/nsis-pkg/nsis_3.01-1_amd64.deb
```

- A build of JMRI generated with the above makensis installation from my jdk9-windows branch has been installed and run on Windows XP.
- This will probably work for other Debian and Ubuntu releases as well.

rhwood, 2017-08-12:
Zip archive of NSIS 3.01 packages for Ubuntu 14.04 LTS AMD64 architecture is attached.

[nsis_3.01-1_amd64.zip](https://github.com/JMRI/JMRI/files/1220050/nsis_3.01-1_amd64.zip)

bobjacobsen, 2017-08-18:
@rhwood Thanks for the instructions 2 above. The 2nd sed line didn't quite work (it wanted additional input, perhaps something in the posting process ate a quoting slash, but I made the change manually), but otherwise it ran OK on the Jenkins server.

[Packages build 1775](http://jmri.tagadab.com/jenkins/job/Development/job/Packages/1775/) was the first with this installed, and it failed:

```
package-windows:
    [mkdir] Created dir: /var/lib/jenkins/jobs/Development/jobs/Packages/workspace/dist/Windows/JMRI
     [copy] Copying 9141 files to /var/lib/jenkins/jobs/Development/jobs/Packages/workspace/dist/Windows/JMRI
     [copy] Copying 31 files to /var/lib/jenkins/jobs/Development/jobs/Packages/workspace/dist/Windows/JMRI
    [mkdir] Created dir: /var/lib/jenkins/jobs/Development/jobs/Packages/workspace/dist/Windows/JMRI/lib
     [copy] Copying 63 files to /var/lib/jenkins/jobs/Development/jobs/Packages/workspace/dist/Windows/JMRI/lib
    [mkdir] Created dir: /var/lib/jenkins/jobs/Development/jobs/Packages/workspace/dist/Windows/JMRI/lib/windows
     [copy] Copying 17 files to /var/lib/jenkins/jobs/Development/jobs/Packages/workspace/dist/Windows/JMRI/lib/windows
    [mkdir] Created dir: /var/lib/jenkins/jobs/Development/jobs/Packages/workspace/dist/release
     [exec] !define: "LVM_GETITEMCOUNT" already defined!
     [exec] !include: error in script: "/usr/share/nsis/Include/WinMessages.nsh" on line 612
     [exec] !include: error in script: "/usr/share/nsis/Contrib/Modern UI 2/MUI2.nsh" on line 25
     [exec] !include: error in script: "/usr/share/nsis/Include/MUI2.nsh" on line 1
     [exec] !include: error in script: "/usr/share/nsis/Include/MultiUser.nsh" on line 19
     [exec] Error in script "InstallJMRI.nsi" on line 350 -- aborting creation process

BUILD FAILED
/var/lib/jenkins/jobs/Development/jobs/Packages/workspace/build.xml:2028: exec returned: 1

Total time: 1 minute 16 seconds
Build step 'Invoke Ant' marked build as failure
```

Here are the messages from the similar section in the prior build, which succeeded with an older NSIS version:

````
package-windows:
    [mkdir] Created dir: /var/lib/jenkins/jobs/Development/jobs/Packages/workspace/dist/Windows/JMRI
     [copy] Copying 9141 files to /var/lib/jenkins/jobs/Development/jobs/Packages/workspace/dist/Windows/JMRI
     [copy] Copying 31 files to /var/lib/jenkins/jobs/Development/jobs/Packages/workspace/dist/Windows/JMRI
    [mkdir] Created dir: /var/lib/jenkins/jobs/Development/jobs/Packages/workspace/dist/Windows/JMRI/lib
     [copy] Copying 63 files to /var/lib/jenkins/jobs/Development/jobs/Packages/workspace/dist/Windows/JMRI/lib
    [mkdir] Created dir: /var/lib/jenkins/jobs/Development/jobs/Packages/workspace/dist/Windows/JMRI/lib/windows
     [copy] Copying 17 files to /var/lib/jenkins/jobs/Development/jobs/Packages/workspace/dist/Windows/JMRI/lib/windows
    [mkdir] Created dir: /var/lib/jenkins/jobs/Development/jobs/Packages/workspace/dist/release
     [exec] ----------------------------------------------------------------------
     [exec] NSIS String Functions Header File 1.09 - Copyright 2004 Diego Pedroso
     [exec] ----------------------------------------------------------------------
     [exec]  (/usr/share/nsis/Include/StrFunc.nsh:52)
     [exec] $ {StrStr} - Copyright 2004 Diego Pedroso - Based on functions by Ximon Eighteen (macro:STRFUNC_FUNC:11)
     [exec] $ {UnStrStr} - Copyright 2004 Diego Pedroso - Based on functions by Ximon Eighteen (macro:STRFUNC_FUNC:5)
     [exec] warning: File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:705)
     [exec] warning: File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:706)
```

I don't know enough about NSIS to know whether we're seeing a flaw in the installed NSIS version, or that this is telling me that the new version is working right but can't handle our installer source files.

bobjacobsen, 2017-08-18:
This is perhaps relevant to #3919 and #3949, so linking them in.

rhwood, 2017-08-18:
NSIS 3 will not compile our current LaunchJMRI.nsi in JMRI/master and NSIS 2.46 will not compile LaunchJMRI.nsi from #3919; the errors you are seeing are expected until #3919 is merged. You might want to have Jenkins build just that branch prior to merging to confirm.

Can you fix the sed statement in my comment so if anyone else needs to build NSIS 3, they can just do so?

bobjacobsen, 2017-08-18:
Here's @rhwood's recipe with a small fix to the sed line (different quoting character):

```
sudo apt-get install --yes devscripts
mkdir -p ~/nsis-pkg
cd ~/nsis-pkg
dget -ux http://http.debian.net/debian/pool/main/n/nsis/nsis_3.01-1.dsc
cd nsis-3.01/
dpkg-checkbuilddeps
# following are what above command reported required on a new 14.04 LTS Workstation edition
sudo apt-get install --yes debhelper docbook-xsl-ns docbook5-xml html2text libcppunit-dev mingw-w64 scons xsltproc zlib1g-dev
# comment out rule that forces failure
sed -i '210s/^/#/' debian/rules
# add large strings and advanced logging support
sed -i '/^SCONSOPTS =/s/\\/NSIS_MAX_STRLEN=8192 NSIS_CONFIG_LOG=yes \\/' debian/rules
dpkg-buildpackage -rfakeroot -us -uc
cd ~
sudo dpkg --install ~/nsis-pkg/nsis-common_3.01-1_all.deb ~/nsis-pkg/nsis_3.01-1_amd64.deb
```

The Jenkins server installed a shorter set of packages, as scons and html2text were already present.

bobjacobsen, 2017-08-18:
A Jenkins test run from the #3919 branch created a Windows installer: [Packages-test/1](http://jmri.tagadab.com/jenkins/job/Development/job/Packages-test/)

I'll ask on JMRI-developers if somebody can please test it.

```
package-windows:
    [mkdir] Created dir: /var/lib/jenkins/jobs/Development/jobs/Packages-test/workspace/dist/Windows/JMRI
     [copy] Copying 9140 files to /var/lib/jenkins/jobs/Development/jobs/Packages-test/workspace/dist/Windows/JMRI
     [copy] Copying 31 files to /var/lib/jenkins/jobs/Development/jobs/Packages-test/workspace/dist/Windows/JMRI
    [mkdir] Created dir: /var/lib/jenkins/jobs/Development/jobs/Packages-test/workspace/dist/Windows/JMRI/lib
     [copy] Copying 63 files to /var/lib/jenkins/jobs/Development/jobs/Packages-test/workspace/dist/Windows/JMRI/lib
    [mkdir] Created dir: /var/lib/jenkins/jobs/Development/jobs/Packages-test/workspace/dist/Windows/JMRI/lib/windows
     [copy] Copying 17 files to /var/lib/jenkins/jobs/Development/jobs/Packages-test/workspace/dist/Windows/JMRI/lib/windows
    [mkdir] Created dir: /var/lib/jenkins/jobs/Development/jobs/Packages-test/workspace/dist/release
     [exec] warning: File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:709)
     [exec] warning: File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:710)
     [exec] warning: File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:711)
     [exec] warning: File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:712)
     [exec] warning: File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:713)
     [exec] warning: File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:714)
     [exec] warning: File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:715)
     [exec] warning: File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:716)
     [exec] warning: File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:717)
     [exec] warning: File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:738)
     [exec] warning: File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:746)
     [exec] warning: File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:755)
     [exec] warning: File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:761)
     [exec] warning: File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:767)
     [exec] warning: File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:781)
     [exec] warning: File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:788)
     [exec] warning: File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:803)
     [exec] warning: File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:810)
     [exec] warning: unknown variable/constant "{MESSAGE_WIN2K_OR_LATER}" detected, ignoring (InstallJMRI.nsi:1069)
     [exec] warning: label "OfflineJREInstall" not used
     [exec] 
     [exec] 20 warnings:
     [exec]   File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:709)
     [exec]   File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:710)
     [exec]   File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:711)
     [exec]   File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:712)
     [exec]   File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:713)
     [exec]   File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:714)
     [exec]   File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:715)
     [exec]   File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:716)
     [exec]   File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:717)
     [exec]   File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:738)
     [exec]   File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:746)
     [exec]   File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:755)
     [exec]   File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:761)
     [exec]   File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:767)
     [exec]   File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:781)
     [exec]   File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:788)
     [exec]   File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:803)
     [exec]   File /a is disabled for non Win32 platforms. (InstallJMRI.nsi:810)
     [exec]   unknown variable/constant "{MESSAGE_WIN2K_OR_LATER}" detected, ignoring (InstallJMRI.nsi:1069)
     [exec]   label "OfflineJREInstall" not used

```

closed 2017-08-20 by bobjacobsen