JMRI/JMRI #3915: JDevHelp does not make TOC

petecressman, 2017-08-12:
Ran  'Run As Ant Build' from  help/en/build.xml in Eclipse.   New file was not in the TOC.
Then Ran manually and verified that new file was listed and missing file was not under the "Index Editor" tab

New file not in TOC, missing file still there.

bobjacobsen, 2017-08-12:
The search tab is automatically maintained.  The table of contents and index are manually maintained. Their contents are kept in the `JmriHelp_enTOC.xml` and `JmriHelp_enIndex.xml` files, respectively.  You can manually edit those, or use the `JHelpDev` GUI tool to update them.

See the [JMRI documentation page](http://jmri.org/help/en/html/doc/Technical/Help.shtml), the section on TOC and index near the bottom.

petecressman, 2017-08-12:
Didn't read closely enough. RTFM

closed 2017-08-12 by petecressman