JMRI/JMRI #3921: Could LayoutEditor draw tracks as… tracks?

geowar1, 2017-08-12:
For track widths greater than 2, tracks could be drawn as actually tracks (two "rails")… and at perhaps some width greater than that ties could be drawn also.

Discussion?

geowar1, 2017-10-28:
See PR https://github.com/JMRI/JMRI/pull/4317

pnbgit, 2017-10-29:
Please don't make this happen automatically - it would completely destroy existing panels that happen to use track widths greater than 2 (I have several of these).  By all means a user selectable option, but  please definitely not the default behaviour - thanks.

geowar1, 2017-10-29:
I completely understand that. Definitely a 100% opt-in. ;-)

I've been looking at layout track plans online and there are a lot of options…
I would welcome feedback/suggestions, etc. (post URL's to online track plan images you like.)

I'm somewhat pleased with what I have so far… 

[image]
[image]

Currently the space between the rails is getting erased but I will fix that.

geowar1, 2017-12-04:
Finally done!!! (Ignoring that "perfect code never ships"… ;-)

[image]

See #4433

geowar1, 2018-02-27:
This should be fixed by #4798.

closed 2018-02-27 by geowar1