JMRI/JMRI #3927: Do not Save Panels if a TrainIcon is visible

Cliff-Anderson, 2017-08-14:
In versions 4.9.2, 4.9.1, 4.8, and perhaps earlier, a Layout Editor panel cannot be loaded if it contains a TrainIcon object.  

There are two ERROR messages triggered by  single TrainIcon object in the following log.  It appears as if the first ERROR might not be critical but the second ERROR stops the panel from loading.

Thus far, my only known fix is to edit the xml file and delete all trainIcon lines.

This is not a high priority problem, as it only seems to hit me when editing the Operations Routes data for icon locations and save intermediate files to double check the results.  Double checking is a long burned in habit, but still not done enough.  
We do not perform a Save Panels at the club when ending an Ops Session.  

'''
2017-08-13 20:12:42,829 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2017-08-13 20:12:42,892 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\cliffaa-rr\JMRI\log\messages.log [main]
2017-08-13 20:12:42,892 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\cliffaa-rr\JMRI\log\session.log [main]
2017-08-13 20:12:42,923 apps.Apps                             INFO  - PanelPro version 4.9.2+Rff24066 starts under Java 1.8.0_144 on Windows 10 amd64 v10.0 at Sun Aug 13 20:12:42 PDT 2017 [main]
2017-08-13 20:12:46,096 apps.Apps                             INFO  - Starting with profile Layout_Ops_Test.3f053837 [main]
2017-08-13 20:12:46,908 node.NodeIdentity                     INFO  - Using jmri-F81654D51042-3f053837 as the JMRI Node identity [main]
2017-08-13 20:12:47,111 managers.JmriUserPreferencesManager   ERROR - Unable to retrieve property "windowRectangle2D" for window "jmri.jmrit.display.layoutEditor.LayoutEditor:Test TrainIcon" [main]
2017-08-13 20:12:49,268 hexfile.LnHexFilePort                 INFO  - LocoNet Simulator Started [Thread-8]
2017-08-13 20:12:50,064 loconet.LnTrafficController           DEBUG - notify of incoming LocoNet packet: B0 78 27 10 [AWT-EventQueue-0]
2017-08-13 20:12:50,080 loconet.LnTrafficController           DEBUG - notify of incoming LocoNet packet: BB 00 00 44 [AWT-EventQueue-0]
2017-08-13 20:12:50,080 loconet.LnTrafficController           DEBUG - notify of incoming LocoNet packet: B0 79 27 11 [AWT-EventQueue-0]
2017-08-13 20:12:50,736 apps.PerformFileModel                 INFO  - Loading file C:\Users\cliffaa-rr\JMRI\Config Profiles\Layout_Ops_Test\panels\Test TrainIcon 1.xml [main]
2017-08-13 20:12:50,877 loconet.LnTrafficController           DEBUG - notify of incoming LocoNet packet: B0 7A 27 12 [AWT-EventQueue-0]
2017-08-13 20:12:52,877 configurexml.ErrorHandler             ERROR - Parse error while parsing file /C:/Users/cliffaa-rr/JMRI/Config%20Profiles/Layout_Ops_Test/panels/Test%20TrainIcon%201.xml Exception: org.jdom2.input.JDOMParseException: Error on line 49: cvc-complex-type.2.4.a: Invalid content was found starting with element 'trainicon'. One of '{positionablelabel, positionablepoint, turnouticon, sensoricon, memoryicon, memoryInputIcon, signalheadicon, signalmasticon, reportericon, layoutturnout, layoutSlip, tracksegment, fastclock, multisensoricon, levelxing, layoutturntable, linkinglabel, locoicon, BlockContentsIcon, LightIcon}' is expected.
See http://jmri.org/help/en/package/jmri/configurexml/ErrorHandler.shtml for possibly more information. [AWT-EventQueue-0]
org.jdom2.input.JDOMParseException: Error on line 49: cvc-complex-type.2.4.a: Invalid content was found starting with element 'trainicon'. One of '{positionablelabel, positionablepoint, turnouticon, sensoricon, memoryicon, memoryInputIcon, signalheadicon, signalmasticon, reportericon, layoutturnout, layoutSlip, tracksegment, fastclock, multisensoricon, levelxing, layoutturntable, linkinglabel, locoicon, BlockContentsIcon, LightIcon}' is expected.
	at org.jdom2.input.sax.SAXBuilderEngine.build(SAXBuilderEngine.java:232)
	at org.jdom2.input.sax.SAXBuilderEngine.build(SAXBuilderEngine.java:253)
	at org.jdom2.input.SAXBuilder.build(SAXBuilder.java:1091)
	at jmri.jmrit.XmlFile.getRoot(XmlFile.java:197)
	at jmri.jmrit.XmlFile.rootFromURL(XmlFile.java:182)
	at jmri.configurexml.ConfigXmlManager.loadOnSwingThread(ConfigXmlManager.java:648)
	at jmri.configurexml.ConfigXmlManager.lambda$1(ConfigXmlManager.java:615)
	at java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$500(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)
Caused by: org.xml.sax.SAXParseException; lineNumber: 49; columnNumber: 354; cvc-complex-type.2.4.a: Invalid content was found starting with element 'trainicon'. One of '{positionablelabel, positionablepoint, turnouticon, sensoricon, memoryicon, memoryInputIcon, signalheadicon, signalmasticon, reportericon, layoutturnout, layoutSlip, tracksegment, fastclock, multisensoricon, levelxing, layoutturntable, linkinglabel, locoicon, BlockContentsIcon, LightIcon}' is expected.
	at org.apache.xerces.util.ErrorHandlerWrapper.createSAXParseException(Unknown Source)
	at org.apache.xerces.util.ErrorHandlerWrapper.error(Unknown Source)
	at org.apache.xerces.impl.XMLErrorReporter.reportError(Unknown Source)
	at org.apache.xerces.impl.XMLErrorReporter.reportError(Unknown Source)
	at org.apache.xerces.impl.XMLErrorReporter.reportError(Unknown Source)
	at org.apache.xerces.impl.xs.XMLSchemaValidator$XSIErrorReporter.reportError(Unknown Source)
	at org.apache.xerces.impl.xs.XMLSchemaValidator.reportSchemaError(Unknown Source)
	at org.apache.xerces.impl.xs.XMLSchemaValidator.handleStartElement(Unknown Source)
	at org.apache.xerces.impl.xs.XMLSchemaValidator.startElement(Unknown Source)
	at org.apache.xerces.xinclude.XIncludeHandler.startElement(Unknown Source)
	at org.apache.xerces.impl.XMLNSDocumentScannerImpl.scanStartElement(Unknown Source)
	at org.apache.xerces.impl.XMLDocumentFragmentScannerImpl$FragmentContentDispatcher.dispatch(Unknown Source)
	at org.apache.xerces.impl.XMLDocumentFragmentScannerImpl.scanDocument(Unknown Source)
	at org.apache.xerces.parsers.XML11Configuration.parse(Unknown Source)
	at org.apache.xerces.parsers.XML11Configuration.parse(Unknown Source)
	at org.apache.xerces.parsers.XMLParser.parse(Unknown Source)
	at org.apache.xerces.parsers.AbstractSAXParser.parse(Unknown Source)
	at org.jdom2.input.sax.SAXBuilderEngine.build(SAXBuilderEngine.java:217)
	... 20 more
'''

bobjacobsen, 2017-08-14:
> Thus far, my only known fix is to edit the xml file and delete all trainIcon lines.

Running the `jython/TurnOffXmlValidation.py` script before reading the file is another workaround (that should work, but I haven't confirmed that).

bobjacobsen, 2017-08-14:
Also, could you please upload the file that can't be read (you might have to change the file extension to .txt or .zip it).  Thanks!

Cliff-Anderson, 2017-08-14:
Panel file with txt extension.

[Test TrainIcon 1.txt](https://github.com/JMRI/JMRI/files/1221364/Test.TrainIcon.1.txt)

Also the xml files from the profile "operations" subfolder.

[operations.zip](https://github.com/JMRI/JMRI/files/1221368/operations.zip)

PR #3928 (2017-08-15): Fix issue #3927: panel file does verify if it contains "trainicon" element
Add 'trainicon' element to LayoutEditor schema.

Includes test case from @Cliff-Anderson, thank you!

Also includes incidental update to release instructions (convenient to include it here instead of another PR)

Closes #3927
closed 2017-08-15 by bobjacobsen