JMRI/JMRI #3941: Unable to delete Turnouts/Sensor if SSL exists.

dsand47, 2017-08-17:
While working on a user problem with SSL errors during loading, I discovered that I could not delete sensors, turnouts, and probably others.

I finally isolated the problem to **getDrivenSignal** in **BlockBossLogic**.

The challenge was that a null pointer exception was occurring on a separate thread that died with no messages on the JMRI console.

My tentative fix is to return an empty string.

>     public String getDrivenSignal() {
>         if (driveSignal == null) {
>             return "";
>         }
>         return driveSignal.getName();
>     }
> 

Since I have never used SSL and this area had a change in 4.9.2, someone knowledgeable in this area should make the change if it is appropriate.

I also noticed a number of warnings when I deleted a test sensor.

[session.txt](https://github.com/JMRI/JMRI/files/1230084/session.txt)
 
Dave Sand

bobjacobsen, 2017-08-17:
@dsand47 Thanks.

I was trying to reproduce this and so far haven't managed.  I'm probably trying too-simple cases.  Any suggestions?  Is there a panel file that could serve as a starting point?  

Also, which thread is throwing the null pointer exception? Or where does it show up in the code?  Perhaps that's enough for me to track it down.

Thanks again.

dsand47, 2017-08-17:
The file I was using was uploaded by Craig Goodyear:  ProblemsBeingWorkedOn/Craig G/Dover_Oakland_SH_2.xml.

Here is my novice take on the process:
When a bean is being deleted, such as a sensor, a **CanDelete** process starts in a separate thread.  If there are SSL objects, BlockBossLogic:vetoableChange() is called.  This line is executed near the beginning:
>message.append(Bundle.getMessage("InUseBlockBossHeader", getDrivenSignal())); //IN18N

This calls getDrivenSignal.  Since there might not be an active **driveSignal**, a NPE can occur.

Since the thread dies, there is no confirmation prompt to the user asking if it is OK to delete based on the current usage list.

Dave Sand

bobjacobsen, 2017-08-17:
Thanks. That helps a lot.

The fundamental problem (which I didn’t realize until now) is that BeanTableDataModel is deferring this operation to another thread via it’s internal DeleteBeanWorker class. That’s got some implications that has to be worked through.

Will put a patch in place tonight or perhaps tomorrow.

PR #4516 (2017-12-06): Fix failure to delete Turnout
Protect against hidden NPE when a SSL has no signal defined.

Closes #3941 and #4514
closed 2017-12-06 by bobjacobsen