JMRI/JMRI #3966: VSDecoder Manager will not launch more than once if closed using the X button

klk32003, 2017-08-21:
A JMRI user reported the following issue ...

> I see a similar problem of not being able to re-open the VSDecoder Manager in Panel Pro after exiting the program using the X button that is part of the window in the upper right. If I use the "Close" button in the lower right I can re-enter the program with no problems. If I use the "Close" button in the lower right I can re-enter the program with no problems.

See https://groups.yahoo.com/neo/groups/jmriusers/conversations/messages/140710

I think this issue was introduced with the first check-in / commit of [VSDManagerFrame.java](https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrit/vsdecoder/swing/VSDManagerFrame.java#L366) on 14 Nov 2012 and never detected since today.

Since I'm not able to solve this I'm asking for support by developer experts. Thank you.

Klaus Killinger

klk32003, 2017-08-22:
Understand this a bit better now and I'll try to solve it.

rhwood, 2017-09-18:
Fixed by #4004.

closed 2017-09-18 by rhwood