JMRI/JMRI #3971: Zimo SW Update V36: Swiss Mapping Groups 11 to 17 added

alain355, 2017-08-22:
Hi Team

I added some updates to my previous "Zimo Unified software (version 36+)" files definition included in JMRI 4.9.3 (SW Version in CV7 from 36 to 40):
1) I added a default value 36 for the SW Version, allowing to create an entry from JMRI without a command station. This allows user to access Swiss Mapping Groups 11 to 17 using the 2 additionnal files in ZImo subfolder.
2) I created using previous zimo subfolder files:
2.1) a CVSwissMapping_V36.xml with the CVs for all Swiss Mapping groups (CV 431 to 507, CV 800 to 823)
2.2) a PaneSwissMapping_V36.xml file to display SwissMapping Pane for the swiss Mapping groups 1 to 17.
As Zimo files are quite complex, I didn't include any other updates shwon onn the Zimo Web page.

Thanks to include them in a future release if OK.

Alain
[Zimo_V36_SwissMapping.zip](https://github.com/JMRI/JMRI/files/1241558/Zimo_V36_SwissMapping.zip)

alain355, 2017-08-27:
Thanks Ken, the error is found and corrected in the 5 decoder definitions: I defined an already defined item, which was unique.

the 5 files with error were:
      <variable CV="7" readOnly="yes" item="Decoder Version"
[Zimo_V36_SwissMapping.zip](https://github.com/JMRI/JMRI/files/1254229/Zimo_V36_SwissMapping.zip)
 default="36">

the 5 newer files without error on my PC versions are:
      <variable CV="7" readOnly="yes" default="36">

updated files attached below:
[Zimo_V36_SwissMapping.zip](https://github.com/JMRI/JMRI/files/1254230/Zimo_V36_SwissMapping.zip)

alain355, 2017-08-27:
Do not update yet, there is not error, but swiss mapping groups 11 to 17 do not appear on the pane.

Still WIP, sorry

alain355, 2017-08-28:
Latest version of those 5 new decoder files, with the 2 related subfolders CVSwissMapping_v36 and PaneSwissMapping_v36 to go into zimo subfolders. Test OK with JMRI 4.9.3
[Zimo_V36_SwissMapping.zip](https://github.com/JMRI/JMRI/files/1255753/Zimo_V36_SwissMapping.zip)

alain355, 2017-08-28:
Sorry I closed it by error while adding the updated file, so I reopened it.

KenC57, 2017-08-28:
It has now been merged.

closed 2017-08-28 by KenC57