JMRI/JMRI #3985: JMRI interface freeze on secondary display.

ClockLoop, 2017-08-24:
JMRI interface freeze on secondary display if second monitor is on left side of main display.
Nvidida geforce 1080gtx, driver 358.28, windows 7x64
Java64bit

If display is on right side of main monitor the jmri interface and all panels respond properly.

We need some code to allow windows placed into negative space.

 Even if any part of any panel or the panel pro window is sitting in the negative space on the left side of the main display, it will not respond.  So when the panel pro window is partly placed between the main display and the display on the left, the items in the menu on the main display respond, meaning you can click things, like debug, and loconet and window and help and the menus will show.  but any item partly or totally inside the second monitor to the left of the main display will not respond at all, and if you resize any panel when its on the left secondary monitor it will graphically be glitched until you drag the window to the main monitor.  

The only fix is to use the monitor on the left as the primary display.  that way the main monitor is to the right of the primary display, and the jmri panels are in positive display space.
Did someone break negative space?   I havent tested negative space above the monitor for freezing, but everything to the right works.

danielb987, 2017-10-14:
Main display on the right and secondary display on the left works for me. Java 8 on Windows 10 64-bit. Intel HD Graphics. I can move LayoutEditor to secondary display on the left and move around turnouts.

ClockLoop, 2017-10-20:
This problem has been resolved in 4.9.4+R3bcfc65

closed 2017-10-20 by ClockLoop