JMRI/JMRI #3986: Appveyor builds no longer completing successfully on a regular basis

pabender, 2017-08-24:
Something has changed that has severely impacted our Appveyor builds.

This started happening about a week ago, where we would see a build take an hour to time out and then after a restart we would see builds complete without warning, but then things started looking like they were getting back to normal...

Now it seems that every single build hangs, and restarts don't help.  (Looking at the history right now, I see TWO successful builds in the last 24 hours, and those may have been from PRs that haven't been fully updated).

Something is hanging in the build, and we need to find out what.

@rhwood you probably know the setup here better than any of us.  Do you have a good feeling for how we can debug this?

pabender, 2017-08-24:
BTW, I can't debug this locally because running the appveyor maven build dies pretty quickly on my Linux box (that was somewhat expected, since appveyor is running windows...)

Does anyone have the ability to run the appveyor maven build on a windows machine?

rhwood, 2017-08-24:
> BTW, I can't debug this locally because running the appveyor maven build dies pretty quickly on my Linux box (that was somewhat expected, since appveyor is running windows...)
> 
This is not expected, since maven is used on appveyor to 1. run the file line ending tests, and 2. bootstrap ant and run the “appveyor” ant target. There is nothing in that target that can’t run on Linux, so maybe the failures to run it on your Linux box point to a problem, that if fixed, will address the problems on appveyor.

rhwood, 2017-08-24:
> Something has changed that has severely impacted our Appveyor builds.
>
> This started happening about a week ago, where we would see a build take an hour to time out and then after a restart we would [...]
There are a couple of things we could try:
- flip appveyor to running straight maven instead of using maven to bootstrap ant — this would loose any CI testing of builds in ant (or of any PackageTest classes), but give us logging that shows which test is timing out.
- figure out how to get ant to actually dump “live” test results so we can see:
  1. when a test starts
  2. when a test ends (and how long it took)
  (this would at least show which test is taking so long that it dies on us)

rhwood, 2017-08-24:
Attached is the appveyor run from my Mac (not Windows -- that would still be a valid check) from a current copy of JMRI/master.
[appveyor-on-rhwood-mac.txt](https://github.com/JMRI/JMRI/files/1248323/appveyor-on-rhwood-mac.txt)

pabender, 2017-08-24:
PR #3987 turns on live output for the ci-test-appveyor ant target.

I'm trying the appveyor maven build locally after a clean.

pabender, 2017-08-24:
looking at the first appveyor build on PR #3979, there is an interrupted exception near the start of the build output and the garbage collection exception at the end.

That was a quick glance, but it's more information than we had before.

bobjacobsen, 2017-08-24:
From @rhwood's log:

```
[INFO] Total time: 18:05 min
[INFO] Finished at: 2017-08-24T05:17:47-04:00
[INFO] Final Memory: 93M/1429M
```

Could we be running up against a process memory limit?  Java used to handle that _really_ poorly,  in some cases just locking up.

geowar1, 2017-08-24:
FYI: One of my appveyor fails ended with a message about a garbage collection exception. If I can find it again I'll put a link to it here…

pabender, 2017-08-24:
@bobjacobsen that might be a problem.

I've seen similar garbage collection issues on my Raspberry Pi's with the default memory settings in the  build.xml file, so I had to change those.

I just pushed an update to appveyor.yml that increases the memory limit in maven.  I want to see how that behaves before I change the build.xml file.  That's building as part of PR #3987 now.

geowar1, 2017-08-24:
Found it: [[junit] Exception in thread "main" java.lang.OutOfMemoryError: GC overhead limit exceeded](https://ci.appveyor.com/project/JMRI/jmri/build/8807)

pabender, 2017-08-24:
Thanks @geowar1 

Looking at ant junit options some more, it looks like we need to set the maxmemory parameter:

> Maximum amount of memory to allocate to the forked VM. Ignored if fork is disabled. Note: If you get java.lang.OutOfMemoryError: Java heap space in some of your tests then you need to raise the size like maxmemory="128m"

The build on my last change finished in 21 minutes and 29 seconds, so it looks like we were running out of memory.

I'm going to go ahead and commit that, and then pull it into my other PR to see if that succeeds.

We may still want to set the maxmemory parameter on the junit tests.

bobjacobsen, 2017-08-24:
"GC overhead limit exceeded" means that Java is (basically) doing nothing but garbage collecting.  Usually, that means there isn't enough heap space and it's trying to over and over to free more with little success.  

In the short term, more heap space will help.  (That might be telling Java to allocation more, Ant to allocate more or might require a larger process limit; I don't know where the restriction comes from)

But in the long run, we should probably look at memory consumption in our test sequences.  We might be leaving lots of things lying around that we don't need to.  Randall's work at moving static instances to InstanceManager should be helping, but there might be other places were references are rooted. (A long time ago, there was a root in JDOM that kept XML stuff around for no visible reason, for example; that's been fixed, but there might be something else like that)

pabender, 2017-08-24:
@bobjacobsen, 

You are right, we need to do better cleanup on our tests.  I think we're working on that (things like cleaning up windows help a great deal) but we've been lax about it for a long time, and It's going to take time to discover all the interdependencies.

for what it's worth, on my Pi, I acually had to lower all of the memory limits so that it would compile and run tests.  With the defaults in the build.xml, ant tests ran for about two hours then failed.  With the memory set to 384m, it compiles in about 45 minutes.

bobjacobsen, 2017-08-25:
See comments in #3984.  Egbert may have found a clue on one of his AppVeyor runs.

pabender, 2017-08-25:
@bobjacobsen I've seen that on a couple of my test builds as well, and I've been working on tracking down extraneous error messages that shouldn't be in the output (which is the focus of my latest PR #3991)  I'll make the change to JmriJFrame and push that up.

There are a number of the tests that make plain JFrames to hold components.

pabender, 2017-08-25:
Oh wait, I see you already did that.

And yes, there are a lot of JmriJFrame objects disposed in tests just in the test runs themselves, mostly in methods marked After.

pabender, 2017-08-26:
So it looks like setting the. maxmemmory parameter on the JUnit ant task does fix the immediate problem noted by this PR.  There might still be some other things we need to look at here, but The Appveyor builds seem to be back to normal after merging #3994.

closed 2017-08-26 by pabender

geowar1, 2017-08-26:
Isn't this just deferring the real issue? Shouldn't this trigger a "seek out and destroy" memory hogs/leakers effort? (For example, I've noticed a bunch of the xml load routines stuff values into temporary variables in the object being loaded and then call an initialize method in that object to call the appropriate setters passing those temporary variables… Those temporary variables are wasted space after initialization time. Should they (and the initialize method?) be moved into the xml loader to be released after the object is done load'ng?)

pabender, 2017-08-26:
Sent from my iPad
> On Aug 26, 2017, at 1:35 PM, George Warner <notifications@github.com> wrote:
> Isn't this just deferring the real issue? Shouldn't this trigger a "seek out and destroy" memory hogs/leakers effort?
> 
Ideally, every test that runs will completely clean up everything it creates.  Unfortunately, all you ever do in Java is remove references to objects.  The garbage collector has to be involved to remove the actual object.

Are there tests where we need to clean up better?  I am certain there are.  Are there tests that don't clean up at all? I am certain there are some of those as well.

As people go through tests to add better coverage, they should make sure they clean up as best they can. ( we have many objects with only constructor tests.  We can do better than that. )
> (For example, I've noticed a bunch of the xml load routines stuff values into temporary variables in the object being loaded and then call an initialize method in that object to call the appropriate [...]
If those temporary variables are local variables, they go away when the method ends.

Also, remember that java creates references to objects  ( effectively, pointers ) and those references are actually what gets passed around, not the object itself. ( yes, you can make copies of objects, but that isn't done automatically)

Paul

geowar1, 2017-08-26:
Those "temporary variables" are class variables set by the Xml load routine (in the target object). It then calls an initialize routine in the target object that used those temporary variables to call the setters for the actually class variables for that object. Most of the values are boolean's and a few strings but it still seems wasteful to have redundant copies in the target object. IMHO these variables should be kept in the xml loader (as local variables) and then used to call the setters after the target object's initialize method has been called. That way they're released when the xml load/store class object is released.

geowar1, 2017-08-26:
> Ideally, every test that runs will completely clean up everything it creates.

Actually, I'm a little less concerned about test memory usage than JMRI memory usage. The test environments can all have as much memory as needed to run their tests… Our JMRI target machines only have whatever memory they have. Being memory conservative helps them more than the test machines.

pabender, 2017-08-26:
Sent from my iPad

> On Aug 26, 2017, at 2:36 PM, George Warner <notifications@github.com> wrote:
> Ideally, every test that runs will completely clean up everything it creates.
> 
> Actually, I'm a little less concerned about test memory usage than JMRI memory usage. The test environments can all have as much memory as needed to run their tests… Our JMRI target machines only have whatever memory they have. Being memory conservative helps them more than the test machines.
> 
This particular PR only has to do with the test environment.

Runtime is much more conservative than the changes made in this 
PR, as far as memory allocation is concerned.

One of the issues with the testing environments is that we are creating and destroying a very large number of objects in a short period of time.  That just doesn't happen on real layouts.  ( Even on the largest layouts, panels are load once, then use for a long time objects. )

pabender, 2017-08-26:
Sent from my iPad

> On Aug 26, 2017, at 2:32 PM, George Warner <notifications@github.com> wrote:
> Those "temporary variables" are class variables set by the Xml load routine (in the target object). It then calls an initialize routine in the target object that used those temporary variables to call the setters for the actually class variables for that object.
> 
We should probably open a separate issue for this if that is a concern.  Can you give us a specific example?

Paul

geowar1, 2017-08-26:
LayoutTurnoutXml sets these temporary variables in the LayoutTurnout class:
```
    // initialization instance variables (used when loading a LayoutTurnout)
    public String connectAName = "";
    public String connectBName = "";
    public String connectCName = "";
    public String connectDName = "";
    public String tBlockName = "";
    public String tBlockBName = "";
    public String tBlockCName = "";
    public String tBlockDName = "";
    public String tTurnoutName = "";
    public String tSecondTurnoutName = "";
```
It then calls the (LayoutTurnout).setObjects() method which uses those temporary variables to call the setters for their appropriate variables. I suspect this was necessary at one time because all the connects, blocks and turnouts may not have existed at the time this LayoutTurnout object is loaded. I don't know if this is still the case (as Blocks and Turnout have already been created before Layout Turnouts are loaded)… But connect's OTOH (which are all TrackSegments) have not been loaded yet.
One way around this would be to wrap all those temporary variables in a container class that's created by the xml loader and passed to the object it's loading so at setObjects() time the temporary variables can be accessed from the wrapper class and then the wrapper class disposed to free its memory.

Or… the xml loader could save a reference to the element it's loading into the object it's creating and then at setObject() time the object could extract the additional element values it needs and then release its reference to the element before returning.

Thoughts?

pabender, 2017-08-26:
Sent from my iPad

> On Aug 26, 2017, at 4:10 PM, George Warner <notifications@github.com> wrote:
> 
> LayoutTurnoutXml sets these temporary variables in the LayoutTurnout class:
> 
Are you talking about jmri.jmrit.display.layoutEditor.LayoutTurnout and JMRI.jmrit.display.layputEditor.configurexml.LayoutTurnoutXml?

Are you talking about lines like this:

a = element.getAttribute("connectaname");
if (a != null) {
l.connectAName = a.getValue();
}

If you are, that's the correct way to verify the xml attribute exists before trying to use it.

The only thing I would question is directly setting connectAName in theturnout object ( 
there should be a setter ).

Paul

a = element.getAttribute("connectaname");

geowar1, 2017-08-26:
Verification isn't the issue: the l.connectAName variable is.
"l.connectAName" is one of those "temporary variables". It's only use in LayoutTurnout is in the setObjects() method when it calls  p.getFinder().findTrackSegmentByName to set connectA.
```
    public void setObjects(LayoutEditor p) {
        connectA = p.getFinder().findTrackSegmentByName(connectAName);
```
IMHO: the xml loader should just save a reference to the element it's loading in the object it's creating and then in that object's setObjects() routine it will pull out the attributes it needs at that time. Then when it's done with the element release it (by setting it to null). So it's only storing one reference to the objects element temporarily (until setObject is called) instead of all those other temporary variables until the object is destroyed and releases them.

pabender, 2017-08-27:
George,

This REALLY should be a separate issue.  It doesn't have anything to do 
with this one.

On 08/26/2017 07:22 PM, George Warner wrote:
> Verification isn't the issue: the l.connectAName variable is.
> "l.connectAName" is one of those "temporary variables". It's only use in
> LayoutTurnout is in the setObjects() method when it calls
> p.getFinder().findTrackSegmentByName to set connectA.

I do see what you are talking about now.

On the surface, I don't see a problem with connectAName being an 
attribute on the turnout block.  A block has to know the user and/or 
system name(s) of the blocks to which it connects so the layout 
connectivity tools can work.

I definitely don't think that connectAName and kin should be public, 
which they are now.  That's exposing internal details to external 
classes (even if it is only used by the XML persistance class)

> |public void setObjects(LayoutEditor p) { connectA =
> p.getFinder().findTrackSegmentByName(connectAName); |
>
> IMHO: the xml loader should just save a reference to the element it's
> loading in the [...]
I really don't think storing the attributes is right either.  There's no 
reason to keep the XML details around for any longer than it takes to 
read the file.

Looking through the code, I THINK the reason it is done the way it is 
now is to allow a one pass read through the XML data.  If you make two 
passes, the first pass can create all the objects, and then you can have 
the second pass create all the connections.

Paul

bobjacobsen, 2017-08-27:
(I agree with @pabender that this should be elsewhere, but while we're here)

Any references to a JDOM2 Element or Attribute keeps the entire JDOM tree (the parsed and expanded version of the file) from being garbage collected.  The Element and Attribute objects have "up and down" references, so that as long as you have a reference to one object keeping it from being gc'd, you've got the whole thing.

The original idea of the [configureXML](http://jmri.org/help/en/html/doc/Technical/XmlPersistance.shtml) system was that it would hide all the XML load and store operations in separate classes. That in turn made it simple to make sure all the references to the XML objects were localized, would go away promptly, and that memory would be reused.

I'm not really familiar with the code, but it looks like that `layoutEditor.BlockValueFile` class just works in a completely different (and ungood) way. Instead of using the `configureXML` patterns, it pulls XML JDOM operations up into a separate file.

Maybe that was thought to needed for some reason, maybe it's just historical, but I think it's not a good approach.  For one thing, exposing JDOM like this has a long history of leading to trouble....

bobjacobsen, 2017-08-27:
(Still catching up) Referring to:

> LayoutTurnoutXml sets these temporary variables in the LayoutTurnout class:
> ```
>     // initialization instance variables (used when loading a LayoutTurnout)
>     public String connectAName = "";
> ...
>     public String tSecondTurnoutName = "";
> ```
> It then calls the (LayoutTurnout).setObjects() method which uses those temporary variables to call the setters for their appropriate variables

This is just bad code.  It's maintaining two sets of state.

The right solution is to have the data marshaled in `LayoutTurnoutXml ` and provided as arguments (or an argument-holding object of some custom class) when `LayoutTurnout` is constructed (better) or afterwards in a init/configure call, or even via a sequence of setFoo calls.

> Or… the xml loader could save a reference to the element it's loading into the object it's creating and then at setObject() time the object could extract the additional element values it needs and then release its reference to the element before returning.

Please don't do this, for all the reasons described above.

pabender, 2017-08-27:
Sent from my iPad

> On Aug 27, 2017, at 12:42 AM, Bob Jacobsen <notifications@github.com> wrote:
> 
> (Still catching up) Referring to:
> 
> LayoutTurnoutXml sets these temporary variables in the LayoutTurnout class:
> 
>     // initialization instance variables (used when loading a LayoutTurnout)
>     public String connectAName = "";
> ...
>     public String tSecondTurnoutName = "";
> It then calls the (LayoutTurnout).setObjects() method which uses those temporary variables to call the setters for their appropriate variables
> 
> This is just bad code. It's maintaining two sets of state.
> 
> The right solution is to have the data marshaled in LayoutTurnoutXml and provided as arguments (or an argument-holding object of some custom class) when LayoutTurnout is constructed (better) or afterwards in a init/configure call, or even via a sequence of setFoo calls.
> 

Unfortunately, the bad code that George is describing is the status quo.

Now my question ( and this is somewhat rhetorical, because I know the answer already... ) is do we have enough tests in place to capture that any improvements in this code produces the current result?

( in terms of raw uncovered line count, the jmri.jmrit.display.layoutEditor package is the least covered package in all of JMRI... but it also has the second highest number of covered lines... which gives it all of 21% line coverage )

bobjacobsen, 2017-08-27:
> Unfortunately, the bad code that George is describing is the status quo.

True, and there's no hurry removing it.  But the discussion was, in part, whether this was something of a good model or not.  It's not.

Note that I don't think saving a few String variables is going to make the difference of a GB of memory during tests, though I haven't checked that.  And if it does, [interning the Strings](http://docs.oracle.com/javase/7/docs/api/java/lang/String.html#intern%28%29) as part of loading them would probably gain a lot of it back.  

> Now my question ( and this is somewhat rhetorical, because I know the answer already... ) is do we have enough tests in place to capture that any improvements in this code produces the current result?

Probably, though I don't know for sure.  The load/store/compare tests should make sure of simple stuff like data being loaded into the right place.  We might not have comprehensive enough test files for LayoutEditor panels though.

geowar1, 2017-08-27:
> in terms of raw uncovered line count, the jmri.jmrit.display.layoutEditor package is the least covered package in all of JMRI... but it also has the second highest number of covered lines... which gives it all of 21% line coverage 

This is what I was working on for https://github.com/JMRI/JMRI/pull/3940. I was hopeful that just getting the LoadAndStoreTest to work would bump the code coverage by a fair amount. Just adding LayoutSlipTest.java bumped it's coverage amount from zero to 37%… that (and the LayoutEditor) will get better as I learn how to drive/test all the dialogs GUI code.
I'm not going to work on any new features until I've gotten the LayoutEditor sources code coverage above the JMRI average (with hopefully some "breathing room" (>= 5%?)).

geowar1, 2017-08-27:
>Note that I don't think saving a few String variables is going to make the difference of a GB of memory during tests
While I do think that memory conservation for testing is good I'm actually more concerned about it for the JMRI apps. The integration checks are the "canary in to coal mine" that we're about to start running into memory issues in the applications.
IOW: "Out of Memory" issues during testing are annoying to the JMRI developers. During an operation session? Ouch. We (the developers) could get lynched. Lets not let that happen.

pabender, 2017-08-27:
Sent from my iPad

> On Aug 27, 2017, at 12:59 PM, George Warner <notifications@github.com> wrote:
> 
> in terms of raw uncovered line count, the jmri.jmrit.display.layoutEditor package is the least covered package in all of JMRI... but it also has the second highest number of covered lines... which gives it all of 21% line coverage
> 
> This is what I was working on for #3940. I was hopeful that just getting the LoadAndStoreTest to work would bump the code coverage by a fair amount. Just adding LayoutSlipTest.java bumped it's coverage amount from zero to 37%… that (and the LayoutEditor) will get better as I learn how to drive/test all the dialogs GUI code.
> 

We are using Jemmy for new GUI tests.

http://jmri.org/help/en/html/doc/Technical/JUnit.shtml#jemmy

The best example we have is probably the speedometer ( which is linked from above ) I built a custom Jemmy operator to drive that.

Paul