JMRI/JMRI #3995: Loconet Stationary Decoder Address 16000 and above not handled correctly in V4.8.

n3ix, 2017-08-26:
On Loconet systems long addresses up to 16383 are valid for stationary decoders.  In JMRI 4.8 doing an Ops Mode Read of CV2 on a Loconet stationary decoder at address 16001 produces the following incorrect Loconet message:

02:43:46.643: [EF 0E 7C 2F 00 7D 01 00 00 01 00 7F 7F 30]  Byte Read on Main Track (Ops Mode): Decoder address 1(short): CV2.

The same operation in JMRI 4.7.1ish produces the expected Loconet message:

02:50:57.535: [EF 0E 7C 2F 00 7D 01 00 02 01 7F 7F 7F 4D]  Programming Track Request: Read Byte in OP's Mode variable  CV2 of Loco 16001 value 255 (0xff, 11111111).

The descriptive text is certainly better in 4.8, however the programming message itself in 4.8 is incorrect.  4.7 shows the correct message.  I tried addresses 16000, 16001, 16100 and 16383 and each exhibited the problem, but just guessing that all addresses >= 16000 are problematic.

A Tam Valley Depot customer ran into this issue recently.  Hopefully he will be fine once he reprograms the board address.  In truth I don’t think there are many of our stationary decoders in the field set for addresses in the range 16000-16383 since we send the decoders out with 11000 as the default address.  However stationary decoders from other manufacturers that are out in the field might have this issue as well.   For example I checked the decoder files for a couple legacy RR-cirkits TC-64 Tower Controllers I own and the tooltip help for the Long Address entry on the “Tower Controller” tab Long Address says “Use long address from 0-16383…”.  Checking the decoder file for the newer TC Mark II shows “Use a 5 digit address from 10000-16383…”.  So there may be some impact.

devel-bobm, 2017-08-26:
I expect to look into this during the week of 28Aug.

n3ix, 2017-08-26:
Bob,

 

Much appreciated.  If it matters, the stationary decoder in question only has a long address (no CV29 usage).  Top of the decoder file looks like this:

 

    <decoder model="Quad-LN_S" family="Quad-LN_S" comment="" />

    <locoaddress>

      <dcclocoaddress number="16001" longaddress="yes" />

      <number>16001</number>

      <protocol>dcc_long</protocol>

    </locoaddress>

 

 

Thanks.

 

Robin

devel-bobm, 2017-08-27:
@n3ix,
I suspect that this this solely a problem of JMRI's LocoNet Monitor displaying a message "differently" than before, and that it does not affect LocoNet programming.  Does that match your understanding of the problem?

If I recall correctly, I spent a bit of time understanding how LocoNet throttle devices operate, and coded LocoNet Monitor to mimic the behavior of the devices.  As such, I mimic'd the "aliasing" and "short address" decoding behavior of the early LocoNet throttles, and the resulting interpretation is displayed.

I do not recall having made changes which would affect how addresses are passed to the programming slot, but I might have done so.

There may be some misunderstanding on my part of LocoNet loco address encoding.  Or there may be some mistaken hardware implementations and/or "spec" interpretations which lead to the odd (but technically correct) reporting in the current LocoNet monitor implementation.

n3ix, 2017-08-27:
Bob,

Comparing the message byte values I posted from the two Jmri versions that I posted I concluded that the message itself had changed for addresses of 16000 and above?   

Robin

From: Bob Milhaupt

Sent: Sunday, August 27, 8:12 AM

Subject: Re: [JMRI/JMRI] Loconet Stationary Decoder Address 16000 and above not handled correctly in V4.8. (#3995)

To: JMRI/JMRI

Cc: Robin, Mention

@n3ix,

I suspect that this this solely a problem of JMRI's LocoNet Monitor displaying a message "differently" than before, and that it does not affect LocoNet programming. Does that match your understanding of the problem?

If I recall correctly, I spent a bit of time understanding how LocoNet throttle devices operate, and coded LocoNet Monitor to mimic the behavior of the devices. As such, I mimic'd the "aliasing" and "short address" decoding behavior of the early LocoNet throttles, and the resulting interpretation is displayed.

I do not recall having made changes which would affect how addresses are passed to the programming slot, but I might have done so.

There may be some misunderstanding on my part of LocoNet loco address encoding. Or there may be some mistaken hardware implementations and/or "spec" interpretations which lead to the odd (but technically correct) reporting in the current LocoNet monitor implementation.

devel-bobm, 2017-08-28:
@n3ix, perhaps I am missing the point, because I don't understand what the problem is.

The _only_ difference I see between the two LocoNet messages you report is in the `<DATA7>` byte and the "extension bit" `<CVH.1>`.  And, for a JMRI "read" of a CV, the value in `<DATA7>` and `<CVH.1>` should not make any difference.

Based on my reverse-engineering of Digitrax command stations, the loco address reported in the decoded text from JMRI 4.8 is consistent with "short address" encoding.

So, if the LocoNet message content is wrong, I will need more information on exactly what user operation is being done, and what "parameters" are in-play, that causes generation of the "wrong" LocoNet message content.  Because I just don't understand the "problem" that needs to be fixed.

devel-bobm, 2017-08-28:
Are we talking about the "message" or the "message"?  
 
I mean "LocoNet byte" or "JMRI's interpretation and User-presentation" of those bytes?

n3ix, 2017-08-29:
Well shoot, I thought I had captured enough info on this problem.   When the customer reported the problem with 4.7.4 reading a board at address 16001, I captured the loconet traffic for an Ops Mode read of CV2 at address 16001 using 4.7.1 and a couple later versions.  It turned out that both the loconet data and description shown in the Monitor Loconet window was different in the later versions from 4.7.1.  I _think_ I also found that a read of CV2 at address 11001 produced the same loconet data in all JMRI versions I tried - only the description had changed.  The read at address 11001 worked fine so it appeared that the issue was the changed loconet bytes when reading address 16001.  I stopped at that point since I didn't have any hardware at address 16001 readily available.  I'm on travel now so I won't be able to do further testing for a couple weeks.

n3ix, 2017-08-30:
Bob, want to clarify that I believe there is a bug in JMRI.  The problem is not with the Loconet Monitor which is fine in 4.8.  The problem is with the Loconet Ops Mode Read message that is generated by JMRI when a user does a Read Sheet on a decoder with address from 16000 to 16383.  Let me try and restate the original info:

Here is the Loconet message sent by JMRI 4.8 to read CV2 when doing an Ops Mode Read Sheet for a decoder with address 16001 using JMRI 4.8

02:43:46.643: [EF 0E 7C 2F 00 7D 01 00 00 01 00 7F 7F 30] Byte Read on Main Track (Ops Mode): Decoder address 1(short): CV2.

Here is the Loconet message sent by JMRI 4.7.1 or earlier for the exact same Ops Mode read sheet operation using JMRI 4.7.1 or earlier

02:50:57.535: [EF 0E 7C 2F 00 7D 01 00 02 01 7F 7F 7F 4D] Programming Track Request: Read Byte in OP's Mode variable CV2 of Loco 16001 value 255 (0xff, 11111111).

Version 4.8 is requesting a read at Short address 1 which is incorrect, it should be requesting a read at Long address 16001.  Version 4.7.1 is correctly requesting a read at Long address 16001.  The change in behavior occurred at approximately Version 4.7.2.

Hope that clarifies the issue?  Thanks.

n3ix, 2017-08-30:
Guess I'm just not able to get this while on the road.  Will give it a rest until I get back in front of a system.

bobjacobsen, 2017-08-30:
Thanks for the detail, made it easy to create a test and run ‘git bisect’ (see below for details).  That says the change in behavior was here:

https://github.com/JMRI/JMRI/commit/aab79ae147c6b528387959cd2ed0b1d03eb1e8ec

That’s one of mine, so I’ll figure out what’s going wrong and fix it shortly.

Sorry for the trouble.

Bob

> On Aug 30, 2017, at 5:18 AM, Robin <notifications@github.com> wrote:
> 
> Bob, want to clarify that I believe there is a bug in JMRI. The problem is not with the Loconet Monitor which is fine in 4.8. The problem is with the Loconet Ops Mode Read message that is generated by JMRI when a user does a Read Sheet on a decoder with address from 16000 to 16383. Let me try and restate the original info:
> 
> Here is the Loconet message sent by JMRI 4.8 to read CV2 when doing an Ops Mode Read Sheet for a decoder with address 16001 using JMRI 4.8
> 
> 02:43:46.643: [EF 0E 7C 2F 00 7D 01 00 00 01 00 7F 7F 30] Byte Read on Main Track (Ops Mode): Decoder address 1(short): CV2.
> 
> Here is the Loconet message sent by JMRI 4.7.1 or earlier for the exact same Ops Mode read sheet operation using JMRI 4.7.1 or earlier
> 
> 02:50:57.535: [EF 0E 7C 2F 00 7D 01 00 02 01 7F 7F 7F 4D] Programming Track Request: Read Byte in OP's Mode variable CV2 of Loco 16001 value 255 (0xff, 11111111).
> 
> Version 4.8 is requesting a read at Short address 1 which is incorrect, it should be requesting a read at Long address 16001. Version 4.7.1 is correctly requesting a read at Long address 16001. The change in behavior occurred at approximately Version 4.7.2.
> 
> Hope that clarifies the issue? Thanks.

The test I added to LnOpsModeProgrammerTest:

     public void testSOps16001Read() throws ProgrammerException {
        LnOpsModeProgrammer lnopsmodeprogrammer = new LnOpsModeProgrammer(sm, memo, 16001, true);
        
        lnopsmodeprogrammer.readCV("2",pl);
        
        // should have written and not returned
        Assert.assertEquals("one message sent", 1, lnis.outbound.size());
        Assert.assertEquals("No programming reply", 0, pl.getRcvdInvoked());

        Assert.assertEquals("message", "[EF 0E 7C 2F 00 7D 01 00 02 01 7F 7F 7F 00]", lnis.outbound.toString());
                
     }

Full bisect result:

aab79ae147c6b528387959cd2ed0b1d03eb1e8ec is the first bad commit
commit aab79ae147c6b528387959cd2ed0b1d03eb1e8ec
Author: Bob Jacobsen <jacobsen@mac.com>
Date:   Mon Mar 27 19:42:43 2017 -0700

    assume 0 value for read cycle

:040000 040000 c5e41dbb06b0a9d09943cba7bee7ed24f7f5e023 b0b96e7c4cff59f3c404fb9155916b73b0c3a037 M	java

--
Bob Jacobsen
jacobsen@mac.com +1-510-708-5988 AIM, Skype JacobsenRG

bobjacobsen, 2017-08-30:
Oops, spoke too soon.

Having looked into it, I think that commit wasn’t the problem.

1) The difference between [EF 0E 7C 2F 00 7D 01 00 00 01 00 7F 7F 30] and  [EF 0E 7C 2F 00 7D 01 00 02 01 7F 7F 7F 4D] is the assumed value for the read operation.

“Read” and “Write” both carry data values. For write, that’s obvious.  For read, not so much.  When the read reply comes back, it has the value-actually-read if the read succeeded.  If not, the value that comes back was the one in the request.  Before this change, that was 0; after this change, it was 0xFF.  That shouldn’t have _anything_ to do with the locomotive address.

(See LocoNet PE page 13 and 14)

Either message is OK

2) So this means that the message should _work_.  Can somebody please confirm if there’s a problem with the message working, and if so what hardware this is happening with? I’ve lost track of that part of the thread.  If there is a problem with actual hardware operations, we need to look at that in more detail.

3) Still, there is a problem in JMRI:  The decoding of the message broke.  ‘git bisect’ says that happened in this commit:

https://github.com/JMRI/JMRI/commit/962d9ca7caba8540a77010e2f17111a7636c7e86

Unfortunately, that’s a huge commit, so bisect can’t break it down any further to localize what went wrong. Somebody else will have to find that.

The test that located this was added to LlmonTest:

        l = new LocoNetMessage(new int[] {0xEf, 0x0E, 0x7c, 0x2F, 0, 0x7D, 0x01, 0x00, 0x02, 0x01, 0x7F, 0x7F, 0x7F, 0x4D}); //!!
        assertEquals("Bit mode direct read test 16001",
                "Programming Track Request: Read Byte in OP's Mode variable  CV2 of Loco 16001 value 255 (0xff, 11111111).\n",
                f.displayMessage(l));

That test is now in the bobjacobsen/loconet_tests_for_Issue_3995 branch and PR #4020 if anybody wants to pick it up.  It’s not passing, so I don’t want to merge the PR until somebody fixes the presentation issue.

n3ix, 2017-08-30:
Bob J and Bob M, thanks for sorting through my less than stellar info.  The changed message text in the Monitor Loconet window threw me and I  should have known better than to post while on travel.  I will test on hardware when I get back and report then.  FYI the customer is using a DCS240 and reported other issues like being unable to Read/Write stationary decoders when track power is off.  I will attempt to sort that out when I'm back in the office, and I fear it means getting a 240 for the test bench.

devel-bobm, 2017-08-30:
The _information presentation_ problem with LocoNet monitor is new, and relates to code I applied.  It is NOT an insurmountable problem, though.

The problem goes like this:  Historically, Digitrax "short" addresses (7-bit) are differentiated from Digitrax "long" addresses (14-bit) by the encoding of the most-significant 7-bits of the address.  My reverse-engineering revealed that in LocoNet messages, the more-significant byte (upper 7 bits) would encode the value 0x7D when encoding a "short" address.  Long address 16000 is encoded in binary as 11111010000000, and when this is broken into two 7-bit quantities, you get 0x7D (most-significant 7 bits) and 0x00(least-significant 7 bits).

Notice this interesting clash - "long" addresses between 16000 and 16127 will be encoded on LocoNet exactly identically to "short" addresses 0 thru 127.

LocoNet Monitor's code was changed, as part of I18N efforts, to rigorously decode the "short" address format.  I did this without regard to the possibility of use of an overlapping range of "long" addresses.

I will take an action to update the LocoNet monitor so that it displays a message something like:

'[EF 0E 7C 2F 00 7D 01 00 00 01 00 7F 7F 30] Byte Read on Main Track (Ops Mode): Decoder address 1(short) (or 16001 long): CV2.'

devel-bobm, 2017-09-11:
@n3ix - any update from your end-user or your own testing?

n3ix, 2017-09-11:
Bob,

Hi.  I'm out of the office for a fewmore days, but there will be a new DCS240 waiting there so I'll be able to do the testing myself.  I will report back on this...

Fyi in the meantime we have established that the 240 does not route Ops Mode (main track) programming messages received on the USB PORT out to the Loconet unless track power is on.  This has created additional confusion with our users trying to program stationary decoders.

Robin

From: Bob Milhaupt

Sent: Monday, September 11, 12:36 PM

Subject: Re: [JMRI/JMRI] Loconet Stationary Decoder Address 16000 and above not handled correctly in V4.8. (#3995)

To: JMRI/JMRI

Cc: Robin, Mention

@n3ix - any update from your end-user or your own testing?

bobjacobsen, 2017-09-11:
> Fyi in the meantime we have established that the 240 does not route Ops Mode (main track) programming messages received on the USB PORT out to the Loconet unless track power is on.  This has created additional confusion with our users trying to program stationary decoders.

Wow. 

JMRI could (maybe) attempt to help people with this by e.g. prompting the user if they are using a DCS240 and track power off. Or maybe turn it on automatically as needed. Worth thinking about. 

What would be _really_ useful would be a list of new and/or unexpected behaviors from the DCS240 so we can address them as a group.

n3ix, 2017-09-14:
I’ve got a DCS240 up and running.  Turns out the behavior with Track Power OFF is not quite what it appeared from the tester reports.  Here is the Loconet Monitor output for an Ops Mode (Main Track) Read of CV17 from a stationary decoder at address 11001 with track power OFF and with track power ON.  Think it sheds some light on what is happening:

 

TRACK POWER OFF

09:51:45.301: [EF 0E 7C 2F 00 55 79 00 00 10 00 7F 7F 71]  Byte Read on Main Track (Ops Mode): Decoder address 11001: CV17.

09:51:45.301: [B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.

09:51:45.301: [E7 0E 7C 2F 00 55 79 00 02 10 6A 7F 7F 11]  Programming Response: Byte Read on Main Track (Ops Mode) Was Successful: Decoder address 11001: CV17 value 234 (0xEA, 11101010b).

 

TRACK POWER ON

09:53:30.723: [EF 0E 7C 2F 00 55 79 00 00 10 00 7F 7F 71]  Byte Read on Main Track (Ops Mode): Decoder address 11001: CV17.

09:53:30.723: [B4 6F 40 64]  LONG_ACK: The Slot Write command was accepted blind (no response will be sent).

09:53:30.739: [E7 0E 7C 2F 00 55 79 00 02 10 6A 7F 7F 11]  Programming Response: Byte Read on Main Track (Ops Mode) Was Successful: Decoder address 11001: CV17 value 234 (0xEA, 11101010b).

 

 

So the good news is that the DCS240 does route the incoming messages from JMRI on its USB port out to Loconet whether or not track power is On and the stationary decoder responds in both cases.  The difference is that the Long Ack from DCS240 has response code 0x00 when track power is OFF and response code 0x40 when track power is ON.   When JMRI sees the 0x00 response code it appears that it discards the response from the stationary decoder.

 

 

(Bob M - the “Slot Write command was …” text the Monitor is using for the LONG ACK description is not the clearest in this context since there never was a Slot Write.   Just noting it.)

 

Thanks.

 

Robin

bobjacobsen, 2017-09-14:
Thanks for the trace.

That "LONG_ACK: The Slot Write command was rejected" is in some sense the right decoding, in that a programmer write operation is done via a Slot Write message.  LocoNet isn't particularly good about disentangling all that stuff, and in particular you _can't_ assume with 100% reliability that the LONG_ACK is a reply to the most recent message you saw.

The bigger issue is that the "LONG_ACK ... rejected" message means different things in different contexts.  With other command stations, it means just that:  It really was understood by the command station and rejected.  The other reply (accepted blind) is what's needed for regular ops-mode operation where the message goes out over DCC, the decoder replies via transcending, and a BDL168 or similar sends the reply message.  

Since the hardware you're working with is apparently just looking at LocoNet, not DCC (rails or rail sync) it still works even if the command station rejected it. 

The logic to decode all this is in "handleLongAck(..)", line 508 of jmrix.loconet.SlotManager.  It's a bit messy....

n3ix, 2017-09-14:
Bob,

 

Hi.  My DCS100 gives the same LONG_ACK response for Ops Mode Reads so this goes way back.  Since JMRI is asking for a (Ops Mode) Read in this particular case, to me the “Slot Write…” description for the LONG_ACK adds a bit of confusion (even if technically accurate).  Not a big deal in any case.

 

In a successful Ops Mode Read sequence, what the command station’s LONG_ACK response code 0x40 “accepted blind” message indicates to me is that the command station itself doesn’t have the ability to do an Ops Mode Read and so it (the command station) will not be sending a read response.  Somebody else _may_ send a read response, whether directly on Loconet by a stationary decoder or by translating some DCC behavior to a Loconet message.   Just IMHO of course!  At least that is what makes sense to me as a Loconet device developer :)

 

Anyway, as you noted to underlying issue is whether to try to handle the 0x00 response code in this context so that the read response is utilized by JMRI.

 

Robin

devel-bobm, 2017-09-21:
@n3ix ,

The difference between 

`[B4 6F 00 24] LONG_ACK: The Slot Write command was rejected.`

and

`[B4 6F 40 64] LONG_ACK: The Slot Write command was accepted blind (no response will be sent).`

is problematic.

The first of these can happen when a DCS210 or DCS240 (and perhaps a DCS51) has track power turned off.  But it can also happen for some other reason when the command station must "reject" the programming operation; this is historic logic that I believe pre-dates the DCS240 and DCS210 and DCS51.

jmrix.loconet.SlotManager treats the two cases differently - the first as a sort of "abort" and the second as a "wait-and-see".  From a historic perspective, this is presumably a correct thing to do.  

But now the more-modern command stations come along and give different behavior for the first of the two messages.  

What's the right thing to do in this case?  I honestly don't know.  Perhaps if I knew what cases could cause the first type of LACK, across the variety of Digitrax and 3rd-party command stations...

In the absence of better information on command station behavior, I propose that we make no changes to JMRI.  Here's my reasoning:

- Presumably this issue does not affect "standalone" LocoNet installations.  There is nothing to issue LACK messages!  No workaround needed.

- Presumably this issue does not affect "historic" Digitrax command stations, else this would have been a problem a long time ago.  No workaround needed.

- Presumably this issue does not affect "historic" 3rd-party command stations, else this would have been a problem a long time ago.  No workaround needed.

- Presumably this issue affects only those most-recent command stations (DCS240, DCS210 and perhaps DCS51).  But there is a work-around - simply turn on track power.

Does this seem reasonable to you, @n3ix ?

devel-bobm, 2017-09-21:
@RR-CirKits , do you have any insights on this?

bobjacobsen, 2017-09-21:
JMRI does know the command station type (assuming it's configured right) and the track power status, so it could handle a special case:  "If DCS240 and track power off, treat 'LONG_ACK ... rejected' as if it's 'LONG_ACK ... blind'

The problem is that we don't know whether the device being programmed is on the LocoNet (unusual, but does happen) in which case this works, or on the track (more common) in which case we've now done the wrong thing: You can't do an ops-mode write to a locomotive decoder if track power is off.

n3ix, 2017-09-21:
BobJ:  Understand that not knowing whether the target is a Loconet device or a DCC on-track decoder makes any attempt to just complete the OpsMode request problematic.  So instead of the special case you noted below, could the test be this: “If DCS240 and track power off, handle any OpsMode/MainTrack programming request with a warning (pop-up dialog for instance) that track power is off and then abort the programming request” ?  At least that would clearly let the user know why the programming operation wasn’t completed.

 

Robin

n3ix, 2017-09-21:
BobJ:  Understand that not knowing whether the target is a Loconet device or a DCC on-track decoder makes any attempt to just complete the OpsMode request problematic.  So instead of the special case you noted below, could the test be this: “If DCS240 and track power off and any OpsMode/MainTrack programming request, then show a warning (pop-up dialog for instance) that track power is off and then abort the programming request” ?  At least that would clearly let the user know why the programming operation wasn’t completed.

 

Robin

devel-bobm, 2017-09-23:
Here are some "sideband" comments I've received from @RR-CirKits:

> I don't think the 16,000 plus issue effects many of our users because I 
> think folks tend to stay below that space. However the DCS240 issue is a 
> new problem. (compared with the Chief) I think that the Zephyr Xtra also 
> does similar things. It is annoying and causes us support calls when 
> folks are trying to setup boards and I need to remember to tell them to 
> be sure that their track power is turned on.
> 
> I don't have a DCS240, and probably will not get one in the future, so I 
> think you guys know more than I do about how it responds to track power 
> on/off, and how JMRI should react to that. E.g. does it buffer the 
> commands and then reject when the buffer is full, (like turnout 
> commands) or immediately reject them if track power is off? What does AJ 
> expect to happen when he sends the 'rejected' message? Is that a 
> reasonable expectation?
> 
> Is this a case where the 'Once Only' option would be useful? Or maybe 
> add a new option into that set of exceptions to normal LONG_ACK processing.
> 
> What happens to all the Ethernet traffic when a router gets turned off? 
> Mine didn't come equipped with a bit bucket or drain. Does the internet 
> try to wait and/or resend the packets? If so, for how long?
> 
> Karen just now hit some button on our printer to clear a print job that 
> she wanted to cancel, and it started spewing out pages from the 8th of 
> the month. Apparently it was holding them all in memory someplace 
> waiting on some stalled job, or the correct paper size, or who knows 
> what?? It has printed lots of things from many different computers since 
> the 8th, so I have no idea where they were waiting or why.
> 
> The point being, what should happen if someone sends OPS mode 
> programming commands when track power is off? Ideally they shouldn't pop 
> out two weeks later, like the printer, or just not do anything like the 
> internet. Maybe an error message? However that could get annoying if the 
> buffers get full very often.
> 
> I am looking at this as just one more reason that the LocoNet may not be 
> the ideal layout control bus for us to support in the future. However, I 
> suppose that any other bus will have the same limitations when part of 
> the transmission channel is disabled for one reason or another.

danielb987, 2020-08-19:
Can this PR be closed? It has not had any comments since 23 Sept 2017.

closed 2020-08-24 by danielb987