JMRI/JMRI #3999: Loop in startup

bobjacobsen, 2017-08-27:
The long-running "jumping window" jmriusers thread is actually about something else:  The startup configuration process goes into a loop of self-creation, and ends up dropping out at some random point when it finally hits an exception.

To recreate, unzip the attached and put the resulting directory into your preferences directory where it will appear as a new 'LocoNet_Simulator_Empty` profile. Start up using that.

On my Mac, after _minutes_ with the fan running, this throws an exception.  In the meantime it's been in a loop doing 
```
     [java] 	at jmri.configurexml.ConfigXmlManager.load(ConfigXmlManager.java:610)
     [java] 	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:230)
     [java] 	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:170)
     [java] 	at jmri.managers.JmriUserPreferencesManager.readUserPreferences(JmriUserPreferencesManager.java:957)
     [java] 	at jmri.managers.JmriUserPreferencesManager$Initializer.getDefault(JmriUserPreferencesManager.java:1453)
     [java] 	at jmri.InstanceManager.getNullableDefault(InstanceManager.java:236)
     [java] 	at jmri.InstanceManager.getDefault(InstanceManager.java:183)
     [java] 	at jmri.managers.configurexml.DefaultUserMessagePreferencesXml.load(DefaultUserMessagePreferencesXml.java:227)
     [java] 	at jmri.configurexml.ConfigXmlManager.loadOnSwingThread(ConfigXmlManager.java:705)
     [java] 	at jmri.configurexml.ConfigXmlManager.load(ConfigXmlManager.java:610)
     [java] 	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:230)
     [java] 	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:170)
     [java] 	at jmri.managers.JmriUserPreferencesManager.readUserPreferences(JmriUserPreferencesManager.java:957)
     [java] 	at jmri.managers.JmriUserPreferencesManager$Initializer.getDefault(JmriUserPreferencesManager.java:1453)
     [java] 	at jmri.InstanceManager.getNullableDefault(InstanceManager.java:236)
     [java] 	at jmri.InstanceManager.getDefault(InstanceManager.java:183)
     [java] 	at jmri.managers.configurexml.DefaultUserMessagePreferencesXml.load(DefaultUserMessagePreferencesXml.java:227)
     [java] 	at jmri.configurexml.ConfigXmlManager.loadOnSwingThread(ConfigXmlManager.java:705)
     [java] 	at jmri.configurexml.ConfigXmlManager.load(ConfigXmlManager.java:610)
     [java] 	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:230)
     [java] 	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:170)
     [java] 	at jmri.managers.JmriUserPreferencesManager.readUserPreferences(JmriUserPreferencesManager.java:957)
     [java] 	at jmri.managers.JmriUserPreferencesManager$Initializer.getDefault(JmriUserPreferencesManager.java:1453)
     [java] 	at jmri.InstanceManager.getNullableDefault(InstanceManager.java:236)
     [java] 	at jmri.InstanceManager.getDefault(InstanceManager.java:183)
     [java] 	at jmri.managers.configurexml.DefaultUserMessagePreferencesXml.load(DefaultUserMessagePreferencesXml.java:227)
     [java] 	at jmri.configurexml.ConfigXmlManager.loadOnSwingThread(ConfigXmlManager.java:705)
     [java] 	at jmri.configurexml.ConfigXmlManager.load(ConfigXmlManager.java:610)
```

over and over.

bobjacobsen, 2017-08-27:
.zip file to demonstrate above:

[LocoNet_Simulator_Empty.zip](https://github.com/JMRI/JMRI/files/1254161/LocoNet_Simulator_Empty.zip)

bobjacobsen, 2017-08-27:
This may be a case where asking for a default object -> automatic creation -> automatic loading is a step too far.  The problem is, starting three lines from the bottom of the trace above:  
- managers.configurexml.DefaultUserMessagePreferencesXml, as part of loading, checks that InstanceManager has a UserPreferencesManager exists (DefaultUserMessagePreferencesXml.java:227) and wants to call `setLoading()` on it
- but that triggers a creation operation via the InstanceManager
- which triggers a call to managers.JmriUserPreferencesManager.readUserPreferences(JmriUserPreferencesManager.java:957)
- which invokes a load operation
- which eventually repeats the whole thing

rhwood, 2017-08-27:
> This may be a case where asking for a default object -> automatic creation -> automatic loading is a step too far. The problem is, starting three lines from the bottom of the trace above: [...]

I’ll take a look at this over the next week. A DefaultUserMessagePreferences object shouldn’t be created ever (except possibly during a migration from < 4.5.4 to newer) sine it was obsoleted by JmriUserPreferencesManager in that time frame), so its likely there is a logic failure in the migration code.

PR #4005 (2017-08-29): Delay initialization of the UserPreferencesManager.
Allow the InstanceManager to initialize the default instance of the UserPreferencesManager after it has created the default instance.

Fixes #3999.

The problem here was that, in the absence of per-profile/per-computer stored user preferences, and in the presence of the older-style per-profile/all-computers stored user preferences, creating and initializing the default instance before registering it was causing the infinite looping. The solution was to implement the InstanceManagerAutoInitialize interface for JmriUserPreferencesManager and to [... cut]
closed 2017-08-29 by rhwood