JMRI/JMRI #4010: Operations - Conductor Edits Train

KenC57, 2017-08-29:
Right now conductor only works if all the cars listed are found and picked up. It would be a great improvement if the conductor could skip cars (think bad order or more model like, not found). It would then adjust the weight and length of the train too. A second option would be to allow adding a car not on the list, but not as critical. I also don't think that a car forced to pick up would need to be processed for routing, but if it could see where it might go along the route, great. Otherwise just take it to the final destination.

I've always though this was missing and I keep running into others who think it would be a great improvement. Once this idea of editing is added two additional tools for layout support would be nice. One would work by listing all cars that should be at a location. You could quickly validate the list and cars not found but listed would be considered lost. The other would be a way to enter a car id and see where it should be or let a location be set for the car. These would help deal with layouts that get a fair bit of movement between normal sessions or other things that cause stuff to magically move all by itself.

All of these would be best for both desktop and for web operation.

DanielBoudreau, 2017-10-24:
To add a car, select the car you want to add from the "Cars" window, and press "Set" and fill in the car's destination and train that you want to car added to.  You can add cars to manifests before you print them, or if using conductor, the window will be updated in the desktop version of conductor.

Also if a car in the conductor window isn't at the right location, the "Set" button allows you to modify the car's location.

DanielBoudreau, 2017-10-25:
I just commits some changes to the desktop version to make it easier to understand how to add and remove cars.

closed 2018-12-03 by DanielBoudreau