JMRI/JMRI #4011: Copy jmri.conf to "Settings" directory as part of Windows install

devel-bobm, 2017-08-29:
Perhaps the NSIS configuration could be changed so that, as part of JMRI install, a copy of a "default" copy of `jmri.conf` is installed into the "Settings" directory.

This would make it easier for "advanced" users to update those things which can be configured via `jmri.conf`.

rhwood, 2017-08-29:
I'm going to suggest not.
- We do not know the settings directory for the user running the installer is at installation time, only what we would default it to; the "advanced" user could also choose to override that.
- If the installer is running with elevated privileges, it may be installing JMRI as a different user than the user who would need this file pre-staged, so the wrong user would get a copy of the jmri.conf file.
- We really want users who are modifying this to think about what they are doing, reading the jmri.conf file and copying it to the correct location is part of that.
- We really do not want to override the user's changes if they copied a file manually (ignoring the problems above).

devel-bobm, 2017-09-13:
@rhwood's comment convinces me.  Closing the issue as "Won't Fix".

closed 2017-09-13 by devel-bobm