JMRI/JMRI #4013: Dispatcher won't properly stop a train before it reaches an occupied section.

ClockLoop, 2017-08-29:
When running multiple trains on the same track each train in a different transit that uses the same track or track that crosses the transit of another train, the dispatcher will not stop the train if it moves into a block before an occupied block.

And with math, proper speed profiles, and correct block lengths, it should be able to stop before the occupied block, is the maths function to do this broken, even the TEST speed step under speed profiling does not work, are these 'lack of block length calculations" related?

I have verified that my track block lengths are correct.
I have tried allocate as far as possible, this doesn't work.

Other issues with dispatcher, when creating a new train, sometimes it will start in manual mode, and not show any blocks as the next block, after it allocates the initial blocks, it will also start in auto mode and do this too, either situation the train will not start, or show any blocks as NEXT.

Pugwash1, 2017-08-29:
Hi
I know most of these, can you assign to me. It needs to be done in bits.

mstevetodd, 2017-08-29:
Dispatcher stops based on signals. It does not look directly at block occupancy. @clockloop,  do you have signalheads or signalmasts set up properly ?

Pugwash1, 2017-08-29:
Actually a speed profile loco hardly ever stops. Unless the profile has extremely small steps. For now I think we just  dont use speed profiling for stopping so at least trains wont runaway in 4.9.4, and then see if we can get it properly fixed in 4.9.n. Unless I find its real easy. It applies to masts and heads.

Pugwash1, 2017-08-29:
Steve T.
This is a diff of what appears to work and what doesn't.
Can you confirm? It looks like the stopping on profile has been disabled at some point. I may be barking up non existant tree here so Id like your opinion.
diff --git a/java/src/jmri/jmrit/dispatcher/AutoActiveTrain.java b/java/src/jmri/jmrit/dispatcher/AutoActiveTrain.java
index cdf7452fad..d39bfced05 100644
--- a/java/src/jmri/jmrit/dispatcher/AutoActiveTrain.java
+++ b/java/src/jmri/jmrit/dispatcher/AutoActiveTrain.java
@@ -1468,13 +1468,18 @@ public class AutoActiveTrain implements ThrottleListener {
                     _currentSpeed = 0.0f;
                     _halted = true;
                 } else if (_slowToStop) {
-//                    re.getSpeedProfile().setExtraInitialDelay(1500f);
-//                    re.getSpeedProfile().changeLocoSpeed(_throttle, _currentBlock, _targetSpeed);
-                    _currentSpeed = _throttle.getSpeedSetting();
+                    if (useSpeedProfile) {
+                        re.getSpeedProfile().setExtraInitialDelay(1500f);
+                        re.getSpeedProfile().changeLocoSpeed(_throttle, _currentBlock, _targetSpeed);
+                    }else {
+                        _currentSpeed = _throttle.getSpeedSetting();
+                    }
                     if (_currentBlock != _lastBlock) {
                         _lastBlock = _currentBlock;
-//                        re.getSpeedProfile().setExtraInitialDelay(1500f);
-//                        re.getSpeedProfile().changeLocoSpeed(_throttle, _currentBlock, _targetSpeed);
+                        if (useSpeedProfile) {
+                            re.getSpeedProfile().setExtraInitialDelay(1500f);
+                            re.getSpeedProfile().changeLocoSpeed(_throttle, _currentBlock, _targetSpeed);
+                        }
                     } else {
                         if (_currentSpeed <= _targetSpeed) {
                             _halted = true;

ClockLoop, 2017-08-29:
I have all signal masts.  With logic that I had to double check due to a layout that has 5 loops, each loop is connected to the next loop with two crossover switches one left and one right, the auto logic gets confused from crossovers, OR its due to the way ANYRAIL6 creates the track node connections, out of order confusing the auto logic?  
Anyway..

The file attached is my layout file.  Not sure if more files are needed to answer more questions.
[Rocky Mountain.zip](https://github.com/JMRI/JMRI/files/1262003/Rocky.Mountain.zip)

When trains don't deal with other occupied blocks or trains in them, they run fine, however it looks like they still overrun the next block if the dispatcher still hasn't claimed them.   (train dosen't stop when blocks aren't acquired or available.)

Sometimes when I **just run my trains on the loops** _and never allow them to cross each others path_, they still once in a while will not acquire future blocks in their transit, run right into the last acquired block and keep going into unclaimed territory, not stopping in the last available block.

I have configured my blocks all with signal masts and the logic does show on the mast lights that it works as trains run through the signals.

Pugwash1, 2017-08-30:
Clock loop. 
Locos that have been speed profiled will not stop. We are working on it. Until we fix it don't waste your time. Use a loco that has never (not a speed profile that has been deleted), but has never  been speed profiled and see how that goes.

mstevetodd, 2017-08-30:
@Pugwash1 the formatting for the diff doesn't show correctly, so I'm not certain of your change, but..... 
I am certain that I disabled the speed profile logic a while back when I was making major changes to Dispatcher. My intent was to set up profiles and reenable it once I could test. Unfortunately, I never reached that point. I suspect part of my issue was that I did not catch the "never had a profile" distinction you point out.
If you have working speed profiles and can test Dispatcher using them, please reenable the logic and give it a go.

mstevetodd, 2017-08-30:
@ClockLoop It'll be a bit before I can look at your panel file, but here are my thoughts on your questions:
I too am experiencing memory crash with recent JMRI with my layout editor panel open. I have not had time to debug, so I've rolled back to 4.8 for running Dispatcher.
Yes, an auto-train should stop at any "Stop" signalmast, whether manually set or sensor-based. If yours does not, investigate and solve that before worrying about loops and multiple trains and such.
I would suggest enabling debug for Dispatcher by adding the line 
`log4j.category.jmri.jmrit.dispatcher=DEBUG `
to your default.lcf file. This will provide some useful details about what Dispatcher is seeing from your signalmasts.

ClockLoop, 2017-08-30:
Oh.   I can just make some fake profiles for engines that have been already profiled by making a new nmra standard cv definitions profile which uses the same address as my currently profiled engines.

I will do some more testing, but initial findings are that MAKING A PROFILE WITH NO SPEED PROFILE, did work to make the train respond to the signal mast.   The train is also responding to speed limits set on blocks now.      

It seems im running into multiple dispatcher bugs, so i will make report on each, instead of piling them all here, thinking they are related, they might be but they must be tracked separate i suppose.  I do search for prior reports best i can so im not duplicating one...

A Debug out for dispatcher, excellent..  will test more asap with that too.

ClockLoop, 2017-08-30:
I got some debug info going and it IS showing informaton that is helpful but why the virtual train man stopped my train at the end of this log I can't figure out.

`23:54:09.261 (174517) DEBUG [AWT-EventQueue-0] TrainInfoFile.java:352  - directory of TrainInfoFiles is /home/pi/.jmri/My_JMRI_Railroad/dispatcher/traininfo/
23:54:12.525 (177781) DEBUG [AWT-EventQueue-0] TrainInfoFile.java:58  - entered readTrainInfo for SantaFe307.xml
23:54:24.918 (190174) DEBUG [AWT-EventQueue-0] AutoActiveTrain.java:246  - 1F - NoSpeedProfile - 307: requesting throttle address=307
23:54:24.977 (190233) DEBUG [AWT-EventQueue-0] AutoActiveTrain.java:897  - 1F - NoSpeedProfile - 307: StopInCurrentSection called for IY:AUTO:0066(Loop3C)
23:54:24.983 (190239) DEBUG [AWT-EventQueue-0] AutoActiveTrain.java:899  - 1F - NoSpeedProfile - 307: train is already stopped or stopping.
23:54:25.048 (190304) DEBUG [AWT-EventQueue-0] AutoActiveTrain.java:659  - 1F - NoSpeedProfile - 307: new current signalmast Loop3Cright(Clear) for section IY:AUTO:0066(Loop3C)
23:54:25.108 (190364) DEBUG [AWT-EventQueue-0] AutoTurnouts.java:214  - 1F - NoSpeedProfile - 307: setting turnout LT2(Cross2) to Closed
23:54:25.258 (190514) DEBUG [AWT-EventQueue-0] AutoTurnouts.java:214  - 1F - NoSpeedProfile - 307: setting turnout LT3(Cross3) to Closed
23:54:25.407 (190663) DEBUG [AWT-EventQueue-0] AutoTurnouts.java:214  - 1F - NoSpeedProfile - 307: setting turnout LT7(Cross7) to Closed
23:54:25.560 (190816) DEBUG [AWT-EventQueue-0] AutoTurnouts.java:214  - 1F - NoSpeedProfile - 307: setting turnout LT6(Cross6) to Closed
23:54:26.897 (192153) DEBUG [AWT-EventQueue-0] AutoActiveTrain.java:282  - 1F - NoSpeedProfile - 307: New AutoEngineer, address=307(L), length=500.0, factor=1.0
23:54:26.904 (192160) DEBUG [Auto Engineer 307] AutoActiveTrain.java:1440  - 1F - NoSpeedProfile - 307: _speedIncrement=0.005
23:54:26.911 (192167) DEBUG [Auto Engineer 307] AutoActiveTrain.java:1443  - 1F - NoSpeedProfile - 307: AutoEngineer.setIsForward(true)
23:54:26.931 (192187) ERROR [Thread-10] LnThrottleManager.java:86  - No response to slot request for 307(L) after 2 attempts.
23:54:26.935 (192191) ERROR [Thread-10] AutoActiveTrain.java:306  - Throttle request failed for 307(L) because Failed to get response from command station
23:54:26.987 (192243) DEBUG [AWT-EventQueue-0] AutoActiveTrain.java:829  - 1F - NoSpeedProfile - 307: Loop3Cright(Clear) Normal(100), Dest: Cross27right, path max: 50
23:54:26.996 (192252) DEBUG [AWT-EventQueue-0] AutoActiveTrain.java:849  - 1F - NoSpeedProfile - 307: Signal Loop3Cright setting speed to 50.0 for next
23:54:47.727 (212983) DEBUG [Allocated Section Block Change on 27] AutoActiveTrain.java:659  - 1F - NoSpeedProfile - 307: new current signalmast Cross27right(Clear) for section IY:AUTO:0058(Cross2high)
23:54:47.731 (212987) DEBUG [Allocated Section Block Change on 27] AutoActiveTrain.java:829  - 1F - NoSpeedProfile - 307: Cross27right(Clear) Normal(100), Dest: Cross28right, path max: 50
23:54:47.737 (212993) DEBUG [Allocated Section Block Change on 27] AutoActiveTrain.java:853  - 1F - NoSpeedProfile - 307: Signal Cross27right setting speed to 50.0 for previous
23:54:50.393 (215649) DEBUG [Allocated Section Block Change on 28] AutoActiveTrain.java:659  - 1F - NoSpeedProfile - 307: new current signalmast Cross28right(Clear) for section IY:AUTO:0059(Cross3low)
23:54:50.398 (215654) DEBUG [Allocated Section Block Change on 28] AutoActiveTrain.java:829  - 1F - NoSpeedProfile - 307: Cross28right(Clear) Normal(100), Dest: Cross29right, path max: 50
23:54:50.404 (215660) DEBUG [Allocated Section Block Change on 28] AutoActiveTrain.java:853  - 1F - NoSpeedProfile - 307: Signal Cross28right setting speed to 50.0 for previous
23:54:53.346 (218602) DEBUG [Allocated Section Block Change on 29] AutoActiveTrain.java:659  - 1F - NoSpeedProfile - 307: new current signalmast Cross29right(Clear) for section IY:AUTO:0060(Cross7low)
23:54:53.353 (218609) DEBUG [Allocated Section Block Change on 29] AutoActiveTrain.java:829  - 1F - NoSpeedProfile - 307: Cross29right(Clear) Normal(100), Dest: unknown, path max: -1
23:54:53.358 (218614) DEBUG [Allocated Section Block Change on 29] AutoActiveTrain.java:853  - 1F - NoSpeedProfile - 307: Signal Cross29right setting speed to 100.0 for previous
23:54:54.252 (219508) DEBUG [Allocated Section Block Change on 30] AutoActiveTrain.java:659  - 1F - NoSpeedProfile - 307: new current signalmast Cross30right(Clear) for section IY:AUTO:0061(Cross6high)
23:54:54.256 (219512) DEBUG [Allocated Section Block Change on 30] AutoActiveTrain.java:829  - 1F - NoSpeedProfile - 307: Cross30right(Clear) Normal(100), Dest: Loop3Aright, path max: 119
23:54:54.259 (219515) DEBUG [Allocated Section Block Change on 30] AutoActiveTrain.java:853  - 1F - NoSpeedProfile - 307: Signal Cross30right setting speed to 100.0 for previous
23:54:55.511 (220767) DEBUG [Allocated Section Block Change on 7] AutoActiveTrain.java:659  - 1F - NoSpeedProfile - 307: new current signalmast Loop3Aright(Approach) for section IY:AUTO:0064(Loop3A)
23:54:55.527 (220783) DEBUG [Allocated Section Block Change on 7] AutoActiveTrain.java:829  - 1F - NoSpeedProfile - 307: Loop3Aright(Approach) Medium(50), Dest: Loop3Bleft, path max: 119
23:54:55.530 (220786) DEBUG [Allocated Section Block Change on 7] AutoActiveTrain.java:849  - 1F - NoSpeedProfile - 307: Signal Loop3Aright setting speed to 50.0 for next
[... cut ...]
23:55:22.825 (248081) DEBUG [Allocated Section Block Change on 9] AutoActiveTrain.java:659  - 1F - NoSpeedProfile - 307: new current signalmast Loop3Cright(Clear) for section IY:AUTO:0066(Loop3C)
23:55:22.830 (248086) DEBUG [Allocated Section Block Change on 9] AutoActiveTrain.java:829  - 1F - NoSpeedProfile - 307: Loop3Cright(Clear) Normal(100), Dest: Cross23right, path max: 0
23:55:22.835 (248091) DEBUG [Allocated Section Block Change on 9] AutoActiveTrain.java:849  - 1F - NoSpeedProfile - 307: Signal Loop3Cright setting speed to 0.0 for next`

rhwood, 2017-08-30:
I have edited some comments in this discussion so they are readable on the GitHub site (if not in email). Please use GitHub's [Markdown](https://guides.github.com/features/mastering-markdown/) formatting to ensure readability.

mstevetodd, 2017-08-30:
@ClockLoop the "path max: x" debug message is showing the lesser of the speed from the signal aspect vs. the minimum speed of the path to the target signal. This path includes block speeds and turnout speeds. So it's basically the min of aspect speed, block speed and turnout speed for items on that path.

I opened your panel and found the masts in question, but I cannot discern how you expect it to work.
Maybe we can discuss this offline? Send me an email if so (mstevetodd at gmail)

ClockLoop, 2017-08-30:
@mstevetodd 
After making a new profile in JMRI, and a new layout file with all new blocks and signal masts, I think I may have found the problem, my signal mast information on some masts is missing.
Heres from my layout file, its missing info.
```  <signalmasts class="jmri.managers.configurexml.DefaultSignalMastManagerXml">
    <virtualsignalmast class="jmri.implementation.configurexml.VirtualSignalMastXml">
      <systemName>IF$vsm:basic:one-low($0029)</systemName>
      <userName>Loop2-A-East</userName>
      <unlit allowed="no" />
    </virtualsignalmast>
```

Heres from a new file i just created...
```
  <signalmasts class="jmri.managers.configurexml.DefaultSignalMastManagerXml">
    <virtualsignalmast class="jmri.implementation.configurexml.VirtualSignalMastXml">
      <systemName>IF$vsm:basic:one-low($0001)</systemName>
      <userName>1a</userName>
      <properties>
        <property>
          <key>intermediateSignal</key>
          <value class="java.lang.Boolean">true</value>
        </property>
      </properties>
      <unlit allowed="no" />
    </virtualsignalmast>
```

Some of my masts are missing this info.
```
      <properties>
        <property>
          <key>intermediateSignal</key>
          <value class="java.lang.Boolean">true</value>
        </property>
      </properties>
```
I don't know if that can explain it, but I think I'll just redo the profile, that'll keep me busy for a while.

ClockLoop, 2017-08-31:
I found that not only is it missing those properties, the logic is messed up on that and some other masts.
I could have sworn it wasn't at some point.  ah well. forge ahead.  I am currently making a new layout file to replace the rockymountain one and I am going to see if I can fix the old to find out what is going wrong so i know.

bobjacobsen, 2017-09-04:
#4019 has been merged.  Does that close this?

Pugwash1, 2017-09-20:
#4025 deals with any outstanding issues. So this can be closed.

closed 2017-09-20 by bobjacobsen