JMRI/JMRI #4016: Openning Layout Panel with Signal Mast Logic is slow

Pugwash1, 2017-08-29:
When openning a layout panel with signal mast logic the LayoutEditor.LayoutEditorTools  invoke later:
 public LayoutEditorTools(LayoutEditor thePanel) {
        layoutEditor = thePanel;

        // defer these actions until all these actually exist...
        SwingUtilities.invokeLater(() -> {
is invoked multiple times, each costing 7 seconds on my PC. for a total of over 2 minutes for a small test layout with 5 blocks and eight masts, during which time the system is unresponsive.
Later, when running trains, it is occasionally invoked again. If the trains are running under autoengineer the system misses various changes in states. I am not familiar enough with this part of the code to competently resolve it.

geowar1, 2017-08-29:
Once upon a time (when I converted all the JTextFields to JComboBoxes) I was getting a null pointer exception on the first call to setupComboBox from here… Deferring those calls till later resolved the issue… Sometime later I found and moved all the inline initializations for those comboboxes to the class variable declaration and could have removed that invokeLater() (but didn't). I think I can safely do so now. Let me double check and I'll fix that ASAP. (Wow! 7 seconds performance hit?!? That's just wrong… LayoutEditor has a few of these as well.)

Pugwash1, 2017-08-29:
> ---------- Original Message ----------

I feel your pain George, after 50 years of programming assembler, Cobol, C, C++
 and Visual Studio languages Java and the long history of JMRI brings me to
tears some days.
Steve G.

geowar1, 2017-08-29:
That 7 second overhead bugs me… I'm going to double check the other places where I've "invokeLater"'d… and see if I can remove them also. Thanks for the heads up.

bobjacobsen, 2017-08-29:
> When openning a layout panel with signal mast logic the LayoutEditor.LayoutEditorTools invoke later:
> public LayoutEditorTools(LayoutEditor thePanel) {
> layoutEditor = thePanel;
>
>     // defer these [...]

!!!

InvokeLater should be a delay of _milliseconds_ at most in a well-running program. It’s not a delay per se; it just puts the routine on the event queue instead of doing it now.  Perhaps it should have been called “invokeAsNextThing()”.

If there’s a 7-second-long event queue, something else terrible is happening.

We’ve seen that in two recent cases:

JMRI 4.9.1 was spending a huge amount of CPU time doing listener updates at a certain phase in initialization.  I don’t remember the details, but basically a bunch of “listen for any new sensor” listeners were installed, and as each new sensor was read from a panel file, a huge amount of work was being done by each sensor that had already been created.  This grew as N^2, slowing the whole system down.

JMRI 4.9.2 was extremely show at painting and repainting the display when there were overlapping windows.

One diagnostic is to move the mouse cursor around on top of JMRI windows during the slow startup. If it’s sluggish, that indicates that there’s too much (too many things or too bug things) happening on the event queue.  

Another clue is the CPU pegging at a constant value (might be 100%, but it might also be 50% or 25% if it’s a multi-core system)

bobjacobsen, 2017-08-30:
Is there a sample file, or ideally a sample profile directory, for this?

Pugwash1, 2017-08-30:
Here you go. (loconit)

rhwood, 2017-08-30:
@Pugwash1 To add an attachment to an issue, you need to view the issue on the GitHub website. Sorry about that.

Pugwash1, 2017-08-30:
Here we go again.

[IsabelleAndMicheals.zip](https://github.com/JMRI/JMRI/files/1263543/IsabelleAndMicheals.zip)

mstevetodd, 2017-08-30:
[session.outofmemory.wcl w debug.log.txt](https://github.com/JMRI/JMRI/files/1264763/session.outofmemory.wcl.w.debug.log.txt)
On the assumption that I'm seeing the same problem, I have been unable to open my layout editor panel with JMRI using latest master. I'm opening it from Eclipse. I can load the 4.8 executables and they open the panel and run fine. This is on Windows 10 and on Linux.

Set up: 
1) Start JMRI with no panels loading. No problems, seems to work fine, UI is responsive.

2) Open a Layout Editor panel with SignalMasts (I've tried mine and another).
3) Panel window opens in initial state.
4) UI becomes VERY sluggish, if it responds at all. CPU stays above 95%.
5) Eventually, I will get an "out of memory" or "gc" error. (see logs). Not always the same error message, but always memory-related.
6) I have to kill the program externally.

To (hopefully) help track down the issue, I turned on ALL debugging and captured the session logs. I will attach two logs, one from my WCL layout, and another "Rocky Mountain" that is attached to issue #4013 

[session.outofmemory.wcl w debug.log.txt](https://github.com/JMRI/JMRI/files/1264768/session.outofmemory.wcl.w.debug.log.txt)
[session.outofmemory.rocky mountain w debug.log.txt](https://github.com/JMRI/JMRI/files/1264770/session.outofmemory.rocky.mountain.w.debug.log.txt)

Pugwash1, 2017-08-30:
Hi Steve 
I filed a github issue on this yesterday or the day before.
Steve G.

On August 30, 2017 3:10:09 PM EDT, SteveT <notifications@github.com> wrote:
> [session.outofmemory.wcl w
> debug.log.txt](https://github.com/JMRI/JMRI/files/1264763/session.outofmemory.wcl.w.debug.log.txt)
> On the assumption that I'm seeing the same problem, I have been unable [...]

-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

geowar1, 2017-08-30:
Just from a performance standpoint I found/fixed one issue: In the past I'd added code to "grow" the layout if a new thing was added outside the current panel size. What I didn't realize was that this code was being called for every part added at layout load time. So I rewrote it to not do the grow check until after everything has been loaded.

bobjacobsen, 2017-08-31:
Thanks for the logs!  From the first one:

```
2017-08-30 14:24:58,773 layoutEditor.TrackSegment             DEBUG - Block boundary  ('LayoutBlock ILB149'<->'LayoutBlock ILB157') found at TrackSegment T129 c1:{TO20 (3}, c2:{A109 (1} [AWT-EventQueue-0]
2017-08-30 14:24:58,774 layoutEditor.TrackSegment             DEBUG - Block boundary  ('LayoutBlock ILB150'<->'LayoutBlock ILB157') found at TrackSegment T132 c1:{TO20 (4}, c2:{A112 (1} [AWT-EventQueue-0]
2017-08-30 14:24:58,774 layoutEditor.ConnectivityUtil         INFO  - trackSegmentLeadsTo(T81, TO4): entry [AWT-EventQueue-0]
2017-08-30 14:24:58,774 layoutEditor.ConnectivityUtil         INFO  - trackSegmentLeadsTo(T80, TO4): entry [AWT-EventQueue-0]
2017-08-30 14:24:58,774 layoutEditor.ConnectivityUtil         INFO  - trackSegmentLeadsTo(T81, TO4): entry [AWT-EventQueue-0]
2017-08-30 14:24:58,774 layoutEditor.ConnectivityUtil         INFO  - trackSegmentLeadsTo(T80, TO4): entry [AWT-EventQueue-0]
2017-08-30 14:24:58,775 layoutEditor.LayoutBlockManager       DEBUG - end initializeLayoutBlockRouting and initializeLayoutBlockPaths [AWT-EventQueue-0]
2017-08-30 14:24:58,776 igurexml.DefaultCatalogTreeManagerXml DEBUG - entered readCatalogTrees [AWT-EventQueue-0]
2017-08-30 14:24:58,777 igurexml.DefaultCatalogTreeManagerXml DEBUG - File: C:\Dropbox\jmri\catalogTrees.xml not Found [AWT-EventQueue-0]
2017-08-30 14:24:58,777 configurexml.LoadXmlUserAction        DEBUG - load was successful [AWT-EventQueue-0]
2017-08-30 14:25:00,024 layoutEditor.LayoutBlockManager       DEBUG - routing table not stable after 0:00:04 in Routing stabilising timer [Routing stabilising timer]
2017-08-30 14:25:00,142 layoutEditor.LayoutEditor             DEBUG - jmri.jmrit.display.layoutEditor.LayoutEditor:Waccamaw Coast Line zoom is null [AWT-EventQueue-0]
2017-08-30 14:25:00,173 layoutEditor.LayoutEditor             DEBUG - jmri.jmrit.display.layoutEditor.LayoutEditor:Waccamaw Coast Line.toolBarThin is null [AWT-EventQueue-0]
2017-08-30 14:25:00,209 layoutEditor.LayoutEditor             DEBUG - prefsMgr.prefsWindowLocation(jmri.jmrit.display.layoutEditor.LayoutEditor:Waccamaw Coast Line) is java.awt.Point[x=786,y=62] [AWT-EventQueue-0]
2017-08-30 14:25:00,209 layoutEditor.LayoutEditor             DEBUG - prefsMgr.prefsWindowSize is(jmri.jmrit.display.layoutEditor.LayoutEditor:Waccamaw Coast Line) java.awt.Dimension[width=0,height=0] [AWT-EventQueue-0]
2017-08-30 14:25:00,209 layoutEditor.LayoutEditor             DEBUG - prefsMgr.getProperty(jmri.jmrit.display.layoutEditor.LayoutEditor:Waccamaw Coast Line, "windowRectangle2D") is null [AWT-EventQueue-0]
2017-08-30 14:25:02,027 layoutEditor.LayoutBlockManager       DEBUG - routing table has now been stable for 2 seconds [Routing stabilising timer]
2017-08-30 14:25:02,028 layoutEditor.LayoutBlockManager       DEBUG - Stable, no sensor to set [Routing stabilising timer]
2017-08-30 14:25:10,899 implementation.DefaultSignalMastLogic DEBUG - true false [AWT-EventQueue-0]
2017-08-30 14:25:10,900 implementation.DefaultSignalMastLogic DEBUG - SM-167-170 Layout name Waccamaw Coast Line [AWT-EventQueue-0]
2017-08-30 14:25:10,941 layoutEditor.LayoutBlockManager       DEBUG - found facing block by positionable point [AWT-EventQueue-0]
2017-08-30 14:25:10,975 layoutEditor.LayoutBlockManager       DEBUG - found signalmast at turnout West:Yard 3 [AWT-EventQueue-0]
2017-08-30 14:25:10,975 layoutEditor.LayoutBlockManager       DEBUG - return block ILB167 [AWT-EventQueue-0]
2017-08-30 14:25:10,975 utEditor.LayoutBlockConnectivityTools DEBUG - faci ILB148 [AWT-EventQueue-0]
2017-08-30 14:25:10,976 utEditor.LayoutBlockConnectivityTools DEBUG - next ILB147 [AWT-EventQueue-0]
```

The LayoutEditor works really, really hard (maybe too hard) to set up, but finally it succeeds about 14:24:58,775.  There's a bit more work, but finally "load was successful" for the panel file. 

There's a short wait to 14:25:00,024, then the LayoutEditor looks up some preferences, and there's another pause.

At 14:25:02,027, "routing table has now been stable for 2 seconds" and a bit more happens.

Then, at 14:25:10,899, the signal mast logic gets triggered. (I think that's a call to `setupLayoutEditorDetails` but am not 100% certain) Maybe something took that long to come in from the layout?  

LayoutEditor now seems to go through a bunch of work and navigation _again_. The, starting around 2017-08-30 14:25:33,057, it tries to load some Operations files. That goes on until (about) 14:25:43,224 with a lot of the action on a separate [Operations Auto Save] thread; could that be the cause of some confusion?

Meanwhile, the LayoutEditor process is chugging along, and finally it runs out of memory.

So there are a couple possibilities:
- Somehow, LayoutEditor is kicking off Operations quite late, and that confuses LayoutEditor.  This seems unlikely, but possible
- LayoutEditor isn't converging, and just keeps going around forever on whatever it's doing
- Something is causing LayoutEditor to recompute whatever it's computing.

Some other observations:
- `end initializeLayoutBlockRouting` appears twice in the log
- `routing table not stable` appears three times.  Maybe it's kicking off something before the previous part was done? (It seems to wait roughly 2 seconds; maybe that's not long enough?)

Pugwash1, 2017-09-01:
@bobjacobsen @geowar1 @rhwood 
Hi 
There are 2 Issues. 
A: The layout editor tools provides 2 distinct types of methods. The first is the obvious Toolbox on the screen, the other is Block Connectivity Tools  such as boolean block1IsWestEnd = tools.isAtWestEndOfAnchor(tr, p); used by LayoutBlockManager. This second set of uses is always proceed by blah = new LayoutEditorTools(). This is the repeated invocations of the swing later.These routines do not need the combo boxes and stuff. 
B: LayoutEditor.setupComboBox takes " forever"
I think both need to be fixed. (A) can be done by the constructor having a boolean for whether the user interface is required. Ugly, bad but should work, or splitting LayoutEditorTools into 2. B is out of my league.

geowar1, 2017-09-01:
I may have fixed LayoutEditor.setupComboBox issue. I conditionalized the getMaximumWindowBounds call with (!GraphicsEnvironment.isHeadless()). See PR #3997

silverailscolo, 2017-09-27:
@Pugwash1 can you confirm this was fixed/reduced to an acceptable amount of time after #4131 by opening your panel file in current head of master? If so, please close this issue.

closed 2017-09-28 by Pugwash1