JMRI/JMRI #4023: 4.9.3 made this user-interface.xml file, which causes a JMRI unrecoverable error.

ClockLoop, 2017-08-30:
4.9.3 somehow made this user-interface.xml file, which causes JMRI unrecoverable error after selecting profile that uses that file.

I used the bak file to verifiy that JMRI 4.9.3 did in fact do something to this file that caused the profile to encounter this error.

Heres the modifed file that causes the unrecoverable error.
[user-interface.xml.bad.zip](https://github.com/JMRI/JMRI/files/1265166/user-interface.xml.bad.zip)

Heres the original that it modified.
[user-interface.xml.good.zip](https://github.com/JMRI/JMRI/files/1265167/user-interface.xml.good.zip)

And heres a shot of the error.
[image]

At the time I was adding block lengths to the panel, and jmri froze.
The next time I started, it showed the error.
Not sure if any of this helps, if not ignore it.

rhwood, 2017-08-31:
I think this is fixed in the current nightly builds. Can you please test a build from http://jmri.tagadab.com/jenkins/job/Development/job/Packages/lastBuild/ and confirm you still see this behavior?

ClockLoop, 2017-08-31:
@rhwood   Sure.  Its nice to know those builds are available,   I tried to compile myself using eclipse but the pi only allows an eclipse version that is too old for the jmri code.

silverailscolo, 2017-08-31:
I notice BlockTableAction lines 83 and 100 in the error. Have been working on that table for the UI, but was convinced this was caught. Will check again.

silverailscolo, 2017-08-31:
@rhwood line 83 applies the result of line 82, including: ``jmri.InstanceManager.getDefault(jmri.BlockManager.class).getDefaultSpeed()``
These lines have been here for some time, but might need a null check or a better way to call instanceManager, which has indeed seen changes of late.

ClockLoop, 2017-08-31:
@rhwood the last build **will** properly load the seemingly corrupt user-interface.xml file, it also seems much speedier in loading.

rhwood, 2017-08-31:
@ClockLoop Thank you.

This would appear to be fixed by #4005.

closed 2017-08-31 by rhwood