JMRI/JMRI #4024: Dispatcher stops train when it should not.

ClockLoop, 2017-08-30:
JMRI v4.8Rbe0ff15
Java v1.8.0_65
OS: RaspberryPi(stretch) and windows 7 64-bit, (i test problems with both before making an issue)

I have made a desktop video capture of the bug.
Link here when upload to youtube finishes.

Attached is the layout file im using called circle.
[Circle.zip](https://github.com/JMRI/JMRI/files/1265304/Circle.zip)

Its a simple circle with 7 blocks.
I have a digitrax 240 command station.

Heres the dispatcher log.
'''
16:45:36.084 (151073) DEBUG [AWT-EventQueue-0] OptionsFile.java:60  - Reading Dispatcher options from file /home/pi/.jmri/Circle/dispatcheroptions.xml
16:45:36.422 (151411) DEBUG [AWT-EventQueue-0] DispatcherFrame.java:103  - Loading saved trains flagged as LoadAtStartup
16:45:36.427 (151416) DEBUG [AWT-EventQueue-0] TrainInfoFile.java:352  - directory of TrainInfoFiles is /home/pi/.jmri/Circle/dispatcher/traininfo/
16:45:36.434 (151423) DEBUG [AWT-EventQueue-0] TrainInfoFile.java:58  - entered readTrainInfo for SF307.xml
16:46:02.799 (177788) DEBUG [AWT-EventQueue-0] TrainInfoFile.java:352  - directory of TrainInfoFiles is /home/pi/.jmri/Circle/dispatcher/traininfo/
16:46:04.749 (179738) DEBUG [AWT-EventQueue-0] TrainInfoFile.java:58  - entered readTrainInfo for SF307.xml
16:46:19.790 (194779) DEBUG [AWT-EventQueue-0] AutoActiveTrain.java:246  - SantaFe307: requesting throttle address=307
16:46:19.853 (194842) DEBUG [AWT-EventQueue-0] AutoActiveTrain.java:897  - SantaFe307: StopInCurrentSection called for IY:AUTO:0009(Circle1)
16:46:19.858 (194847) DEBUG [AWT-EventQueue-0] AutoActiveTrain.java:899  - SantaFe307: train is already stopped or stopping.
16:46:19.911 (194900) DEBUG [AWT-EventQueue-0] AutoActiveTrain.java:659  - SantaFe307: new current signalmast 2a(Clear) for section IY:AUTO:0009(Circle1)
16:46:19.971 (194960) DEBUG [AWT-EventQueue-0] AutoAllocate.java:430  - SantaFe307: auto allocating Section IY:AUTO:0010(Circle2)
16:46:20.018 (195007) DEBUG [AWT-EventQueue-0] AutoAllocate.java:430  - SantaFe307: auto allocating Section IY:AUTO:0011(Circle3)
16:46:20.065 (195054) DEBUG [AWT-EventQueue-0] AutoAllocate.java:430  - SantaFe307: auto allocating Section IY:AUTO:0012(Circle4)
16:46:20.754 (195743) DEBUG [AWT-EventQueue-0] AutoActiveTrain.java:282  - SantaFe307: New AutoEngineer, address=307(L), length=200.0, factor=1.0
16:46:20.767 (195756) DEBUG [Auto Engineer 307] AutoActiveTrain.java:1440  - SantaFe307: _speedIncrement=0.01
16:46:20.782 (195771) DEBUG [Auto Engineer 307] AutoActiveTrain.java:1443  - SantaFe307: AutoEngineer.setIsForward(true)
16:46:20.845 (195834) DEBUG [AWT-EventQueue-0] AutoActiveTrain.java:829  - SantaFe307: 2a(Clear) Normal(100), Dest: 3a, path max: 100
16:46:20.850 (195839) DEBUG [AWT-EventQueue-0] AutoActiveTrain.java:849  - SantaFe307: Signal 2a setting speed to 100.0 for next
16:46:20.926 (195915) DEBUG [AWT-EventQueue-0] AutoActiveTrain.java:282  - SantaFe307: New AutoEngineer, address=307(L), length=200.0, factor=1.0
16:46:20.937 (195926) DEBUG [Auto Engineer 307] AutoActiveTrain.java:1440  - SantaFe307: _speedIncrement=0.01
16:46:20.942 (195931) DEBUG [Auto Engineer 307] AutoActiveTrain.java:1443  - SantaFe307: AutoEngineer.setIsForward(true)
16:46:20.937 (195926) DEBUG [AWT-EventQueue-0] AutoActiveTrain.java:829  - SantaFe307: 2a(Clear) Normal(100), Dest: 3a, path max: 100
16:46:20.953 (195942) DEBUG [AWT-EventQueue-0] AutoActiveTrain.java:853  - SantaFe307: Signal 2a setting speed to 100.0 for previous
16:46:20.992 (195981) WARN  [Auto Engineer 307] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.046 (196035) WARN  [AWT-EventQueue-0] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.096 (196085) WARN  [Auto Engineer 307] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.146 (196135) WARN  [Auto Engineer 307] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.160 (196149) WARN  [Auto Engineer 307] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.171 (196160) WARN  [AWT-EventQueue-0] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.200 (196189) WARN  [AWT-EventQueue-0] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.200 (196189) WARN  [Auto Engineer 307] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.265 (196254) WARN  [Auto Engineer 307] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.310 (196299) WARN  [AWT-EventQueue-0] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.310 (196299) WARN  [Auto Engineer 307] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.386 (196375) WARN  [AWT-EventQueue-0] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.395 (196384) WARN  [Auto Engineer 307] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.417 (196406) WARN  [AWT-EventQueue-0] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.418 (196407) WARN  [Auto Engineer 307] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.510 (196499) WARN  [AWT-EventQueue-0] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.512 (196501) WARN  [Auto Engineer 307] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.525 (196514) WARN  [Auto Engineer 307] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.620 (196609) WARN  [Auto Engineer 307] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.641 (196630) WARN  [Auto Engineer 307] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.668 (196657) WARN  [AWT-EventQueue-0] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.674 (196663) WARN  [AWT-EventQueue-0] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.681 (196670) WARN  [AWT-EventQueue-0] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.761 (196750) WARN  [Auto Engineer 307] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.764 (196753) WARN  [Auto Engineer 307] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.845 (196834) WARN  [AWT-EventQueue-0] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.851 (196840) WARN  [AWT-EventQueue-0] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.866 (196855) WARN  [Auto Engineer 307] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.869 (196858) WARN  [Auto Engineer 307] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.976 (196965) WARN  [Auto Engineer 307] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:21.979 (196968) WARN  [Auto Engineer 307] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:46:22.020 (197009) WARN  [AWT-EventQueue-0] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
[... 546 lines cut ...]
16:47:04.912 (239901) WARN  [AWT-EventQueue-0] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:47:04.952 (239941) WARN  [Auto Engineer 307] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:47:05.023 (240012) WARN  [AWT-EventQueue-0] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:47:05.055 (240044) WARN  [Auto Engineer 307] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:47:05.120 (240109) WARN  [AWT-EventQueue-0] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:47:05.158 (240147) WARN  [Auto Engineer 307] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:47:05.223 (240212) WARN  [AWT-EventQueue-0] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:47:05.261 (240250) WARN  [Auto Engineer 307] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
[... cut ...]
16:47:06.010 (240999) WARN  [Auto Engineer 307] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:47:06.041 (241030) WARN  [AWT-EventQueue-0] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:47:06.113 (241102) WARN  [Auto Engineer 307] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
16:47:06.145 (241134) WARN  [AWT-EventQueue-0] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
'''

ClockLoop, 2017-08-30:
It seems one of my blocks is not occupied or unoccupied, I looked at the block and the sensor was set proper so I looked at the sensor and it showed unknown, when all other sensors showed proper.  When I clicked the sensor state to force a state, it not only started showing its proper state but the train started back up.

I would consider this a bug because the sensor didn't actually check its hardware value for some reason it just kept unknown.

This time I missed the problem in the log before I made all this effort with the video and log.
```
16:46:43.628 (218617) WARN  [AWT-EventQueue-0] Section.java:249  - Occupancy of block IB6 is not OCCUPIED or UNOCCUPIED in Section - IY:AUTO:0014
16:46:43.682 (218671) WARN  [AWT-EventQueue-0] Section.java:249  - Occupancy of block IB6 is not OCCUPIED or UNOCCUPIED in Section - IY:AUTO:0014
16:46:43.922 (218911) WARN  [AWT-EventQueue-0] Section.java:249  - Occupancy of block IB6 is not OCCUPIED or UNOCCUPIED in Section - IY:AUTO:0014
16:46:47.822 (222811) WARN  [AWT-EventQueue-0] Section.java:249  - Occupancy of block IB6 is not OCCUPIED or UNOCCUPIED in Section - IY:AUTO:0014
```

ClockLoop, 2017-08-30:
So I forced the state of the sensor LS69  which is block 29 to unoccupied and saved the layout.
I then restarted jmri, and the same sensor comes up as unknown.

So I took a coal car with resisors on the wheels and rolled it over the sensor in question, it changed from unknown to occupied, then to unoccupied once I remove the car.
I then saved the layout file, and reloaded it, now the sensor shows as unoccupied when I load the layout file.

The first attached post with circle.zip has the layout file with the sensor that wouldn't work even though it had its hardware address and was originally entered into the sensor table by a train going thru the block.
So why i needed to roll another car over the sensor to get the sensor to read its state i don't know.

ClockLoop, 2017-08-30:
If the sensor state showed as occupied even if it was not (if the state is unknown set it as occupied).
This way the user (me) knows something is wrong with the sensor, or something to do with that sensor configuration.   
Instead the state was set to unknown making the track color black.   

Or

If the sensor state is unknown can the track color be set to the occupied color, not the unoccupied color?

bobjacobsen, 2017-08-31:
What hardware is that sensor connected to?  E.g. a BDL168, or something else?

ClockLoop, 2017-08-31:
Its a digitrax BXP88, actually I have 4 of them.
http://www.digitrax.com/products/detection-signaling/bxp88/

They all work great, this is the first time I have ever had this happen.  
The led occupancy indicator on the DXP88 not only shows occupied, but transponding.

ClockLoop, 2017-08-31:
Im guessing that this is just some jmri thing not related to my digitrax hardware.
```
16:46:20.992 (195981) WARN  [Auto Engineer 307] AutoTrainsFrame.java:151  - handleThrottleChange - cannot find throttle index
```

Pugwash1, 2017-08-31:
@ClockLoop Switch the wires for the block with a problem to its neighbour and see if problem follows the move. Check opsw 41 to 48 to make sure it's reporting correctly. The throttle messages can be caused by having stolen a throttle. If you post these problems first on the Yahoo groups you will access a much larger and diverse knowledge base.

KenC57, 2017-08-31:
Does the BXP88 have an option for if it responds to the 'GPON'?? Some
devices make it an option that when the system starts it will create and
send sensor messages to say what is its current status. However I've seen
some systems where that is a flood of messages and some devices seem to miss
that first round, my guess traffic collisions.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

ClockLoop, 2017-10-20:
It seems the problem was caused by a bxp88 slowly loosing its mind.

One day the unit stopped working and flashed red and green.   I later found after it sat on my desk for a month, that digitrax released a firmware update, so i tried that.  It made the unit come back to life, so i suspect the problem is physical for this one.  Its good that digitrax releases teh firmware for these, saved me shipping money and time.

closed 2017-10-20 by ClockLoop