JMRI/JMRI #4025: Dispatcher needs maximum block aquire, rule change and number entry...

ClockLoop, 2017-08-31:
Sometimes my train uses 3 blocks when running due to multiple blocks being crossover switches that have many blocks in a short area of track.

The dispatcher will stop acquiring blocks if it has already acquired and occupied 3 blocks.
The train will not stop when it gets to the last block it can acquire it just runs into unclaimed territory.

So I propose that all trains will acquire at least one block ahead in its transit, regardless of the maximum number of blocks its allowed, if the block is available.

This will allow for condensed blocks in switch yards and the like.

Currently we only have two options, 3 blocks acquired, or all of them on your whole darn transit darn the torpedoes and other trains here i come and i claim it all.

Having the ability to enter in the amount of blocks appropriate to the layout, **train length** would be GREAT!

The minimum of 1 block ahead of the train should be acquired if available even if the train has already reached its maximum, or at least stop the train for this situation, the max block entry would let you fix train length and many compressed blocks.

Pugwash1, 2017-08-31:
Before we start writing code try putting the blocks into sections, as many blocks per section as you like. The system allocates 3 sections ahead.

ClockLoop, 2017-08-31:
I thought I tried the sections and it still didn't work but I can try again.
Yeah I didn't want to suggest code changes but sections was the other other way, I will try that again.

ClockLoop, 2017-09-01:
@Pugwash1 
I think i know why i thought the sections didn't work, the speed profile bug, I was using all speed profiled trains since I started using jmri a month ago, it would have been nice to know the whole time my dispatched trains were Casey Jones'in.   

I just tried to redo my sections and transits and doing like you said has worked so far I have not had to use the allocate as far as possible option.
I think it might still be nice to have the option to choose the allocation rules a bit more, perhaps put that Idea on the shelf, along with the million other.

closed 2017-10-20 by ClockLoop