JMRI/JMRI #4032: RaspberryPi Only - JMRI.4.9.3.Rcadf9de and JMRI.4.9.4ish unusable.

ClockLoop, 2017-08-31:
FYI
On the Raspberrypi 3, Raspbian Stretch.

-JMRI.4.9.3.Rcadf9de

-JMRI.4.9.4ish+jenkins+20170831T1143Z+Rf6edb7b

Both of the above versions with my layout file become too slow, the interface and actions all respond slowly or delayed, ON THE PI ONLY.

4.9.3 and up uses twice the amount of ram JMRI v4.8Rbe0ff15 uses. It peaks at max usage on a pi3 before its starts to miss sensor changes and allocation releases and requests.

I the pi3 runs JMRI v4.8Rbe0ff15 with my layout file acceptably.

bobjacobsen, 2017-09-04:
This might be an initialization loop found and fixed in #3963.  Could you try it again once that it merged?

If that's not it, it would be very helpful to get a thread and heap dump (ctrl-\) while it's slow on the RPi.

ClockLoop, 2017-10-20:
I will give it test, now that my bxp88 is fixed.

closed 2017-12-10 by bobjacobsen