JMRI/JMRI #4056: Issue with order of preference loading during PanelPro startup

bobjacobsen, 2017-09-05:
Getting an intermittent error during PanelPro startup (in PanelProTest) that seems to be due to the parallel nature of the preferences and main threads. (See also #4054).  This was perhaps always a possibility, but the current test makes it more reproducible (about 20%).

The "init prefs thread" calls for ProgrammerConfigPane through a service loader at TabbedPreferences.java:184.  That fails unless a jmri.jmrit.symbolicprog.ProgrammerConfigManager already exists (see trace).  That class is properly registered in "target/classes/META-INF/services//jmri.spi.PreferencesManager", so it should be loaded at some point, but I haven't been able to figure out where.  Any suggestions on tracking this down greatly appreciated.

```
Exception in thread "init prefs" java.util.ServiceConfigurationError: jmri.swing.PreferencesPanel: Provider jmri.jmrit.symbolicprog.ProgrammerConfigPane could not be instantiated
	at java.util.ServiceLoader.fail(ServiceLoader.java:232)

	at java.util.ServiceLoader.access$100(ServiceLoader.java:185)
	at java.util.ServiceLoader$LazyIterator.nextService(ServiceLoader.java:384)
	at java.util.ServiceLoader$LazyIterator.next(ServiceLoader.java:404)
	at java.util.ServiceLoader$1.next(ServiceLoader.java:480)
	at apps.gui3.TabbedPreferences.init(TabbedPreferences.java:184)
	at apps.Apps$3.lambda$run$4(Apps.java:366)
	at java.util.Optional.ifPresent(Optional.java:159)
	at apps.Apps$3.run(Apps.java:365)
	at java.lang.Thread.run(Thread.java:748)
Time: 11.178

OKCaused by: java.lang.NullPointerException: Required nonnull default for jmri.jmrit.symbolicprog.ProgrammerConfigManager does not exist. (2 tests)

	at jmri.InstanceManager.getDefault(InstanceManager.java:195)
	at jmri.jmrit.symbolicprog.ProgDefault.getDefaultProgFile(ProgDefault.java:96)
	at jmri.jmrit.symbolicprog.ProgrammerConfigPane.<init>(ProgrammerConfigPane.java:36)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at java.lang.Class.newInstance(Class.java:442)
	at java.util.ServiceLoader$LazyIterator.nextService(ServiceLoader.java:380)
	... 7 more
```

rhwood, 2017-09-05:
The traces on this and #4057 are identical. Is that intentional or an error?

bobjacobsen, 2017-09-05:
(#4057 was meant to be a trace with similar sequence where `ManagerDefaultSelector` wasn't loaded; probably due to the same underlying cause as this one)

Currently, I think this isn't so much about preferences being in a parallel thread, vs somehow service loading not working as intended in 2nd and later tests of the PanelPro class.  

#4054 about synchronization is also an issue still, it's just not what's happening here.

rhwood, 2017-09-06:
We may need to call [ServiceLoader.reload()](https://docs.oracle.com/javase/8/docs/api/java/util/ServiceLoader.html#reload--) before JmriConfigurationManager instances are constructed in the tests and possibly within the JmriConfigurationManager itself (I think that if JmriConfigurationManager implemented jmri.Disposable, we could do that in the dispose() method).

closed 2017-10-30 by bobjacobsen