JMRI/JMRI #4062: LocoNet throttle acquisition does not honor command station rejection

devel-bobm, 2017-09-05:
When a script attempts to acquire a JMRI LocoNet throttle for a given mobile decoder address, and that script does not have support for the recently-added "steal" feature, and that mobile decoder address is already "in-use" by another throttle, the default JMRI behavior is to attempt to complete the Throttle acquisition.  

The LocoNet throttle object attempts to set the LocoNet slot to "in-use", even though it already is "in-use".  The command station returns a LONG_ACK "refusal" of the attempt.  JMRI's LocoNet support does not recognize this refusal.  The throttle object is "associated" with the slot, and the LocoNet throttle object s returned to the script.  The script may use the slot, even though another LocoNet agent also has control of the slot.

This may be seen in JMRI code (with the "steal" functionality) via hardware test.  A DCS210, a UT4R, and a LocoBuffer-USB were used to dial-up engine 14.  Then the jython script `BackAndForthTimed.py` was run (which uses engine 14), with the following LocoNet monitor log:

**UT4R throttle acquires loco address 14:**

`[BF 00 0E 4E]  Request slot for loco address 14.`
`[E7 0E 4D 33 0E 00 00 47 00 00 00 1B 74 4E]  Report of slot 77 information:
	Loco 14 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x74 0x1B (14875).`
`[EF 0E 4D 33 0E 00 00 47 00 00 00 1B 74 46]  Write slot 77 information:
	Loco 14 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x74 0x1B (14875).`
`[B4 6F 7F 5B]  LONG_ACK: Function not implemented, no reply will follow.`
`[DF 00 00 00 00 20]  Query Tetherless Receivers.`
`[A0 4D 00 12]  Set speed of loco in slot 77 to 0.`
`[A0 4D 03 11]  Set speed of loco in slot 77 to 3.`
`[A0 4D 05 17]  Set speed of loco in slot 77 to 5.`
`[A0 4D 07 15]  Set speed of loco in slot 77 to 7.`
`[A0 4D 09 1B]  Set speed of loco in slot 77 to 9.`
`[A0 4D 0B 19]  Set speed of loco in slot 77 to 11.`
`[A0 4D 06 14]  Set speed of loco in slot 77 to 6.`
`[A0 4D 03 11]  Set speed of loco in slot 77 to 3.`
`[A0 4D 00 12]  Set speed of loco in slot 77 to 0.`

**Jython script `BackAndForthTimed.py` is started:**

`[BF 00 0E 4E]  Request slot for loco address 14.`
`[E7 0E 4D 33 0E 00 00 47 00 00 00 1B 74 4E]  Report of slot 77 information:
	Loco 14 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x74 0x1B (14875).`

**Note above that the requested slot is already "in-use" and assigned to throttle 0x74 0x1B.**

**JMRI's throttle new "Steal" functionality was invoked as a result of the "in-use" reply above, but the script does not have any "Steal" callback functions.  The script blindly goes forward, not knowing that there is a problem.**

**Below, JMRI continues to acquire the slot.  First, it writes the ThrottleID value for JMRI to the slot:**

`[EF 0E 4D 33 0E 00 00 47 00 00 00 71 02 5A]  Write slot 77 information:
	Loco 14 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x02 0x71 (369).`
`[B4 6F 7F 5B]  LONG_ACK: Function not implemented, no reply will follow.`

**Next, JMRI sets the slot to "in-use", even though it already is.  The command station rejects this.**

`[BA 4D 4D 45]  Set status of slot 77 to IN_USE.`
`[B4 3A 00 71]  LONG_ACK: The Move Slots command was rejected.`

**Next, the script blindly goes forward, making use of the slot as if all were well:**

`[A1 4D 00 13]  Set loco in slot 77 direction Forward F0=Off F1=Off F2=Off F3=Off F4=Off.`
`[A0 4D 5A 48]  Set speed of loco in slot 77 to 90.`
`[A0 4D 00 12]  Set speed of loco in slot 77 to 0.`
`[A1 4D 20 33]  Set loco in slot 77 direction Reverse F0=Off F1=Off F2=Off F3=Off F4=Off.`
`[A0 4D 5A 48]  Set speed of loco in slot 77 to 90.`
`[A0 4D 00 12]  Set speed of loco in slot 77 to 0.`
`[A1 4D 00 13]  Set loco in slot 77 direction Forward F0=Off F1=Off F2=Off F3=Off F4=Off.`
`[A0 4D 5A 48]  Set speed of loco in slot 77 to 90.`

It is presumed that those other scripts which are not modified for the "Steal" feature will behave in similar ways.

pabender, 2017-09-05:
Any script that extends AbstractAutomaton (and most of them do) will actually perform an automatic steal because #3904 put this implementation of notifyStealThrottleRequired in place:

@Override
public void notifyStealThrottleRequired(jmri.DccLocoAddress address){
          // this is an automatically stealing impelementation.
          InstanceManager.throttleManagerInstance().stealThrottleRequest(address, this, true);
}

That bit of code was put into place anywhere we might need to acquire a throttle but don't necessarily have a UI for the throttle.

Most scripts should be able to run in headless mode, so it makes sense for them to implement an automatic steal.

Paul

pabender, 2017-09-05:
What you're reporting above may actually not be because of the steal request being not being processed.

The LocoNet throttle should notify the ThrottleListener objects using notifyFailedThrottleRequest at the point where you see:

[BA 4D 4D 45] Set status of slot 77 to IN_USE.
[B4 3A 00 71] LONG_ACK: The Move Slots command was rejected.

Paul

KenC57, 2017-09-05:
Paul,

 

Can a script be written that lets the script choose if it wants to steal or
not?? I agree that defaulting to stealing is the best match for any of the
existing implementations. But some scripts have an interface, so having the
option is important.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

pabender, 2017-09-05:
@KenC57 

Yes, just override the notifyStealThrottleRequired method and do any script specific steal there.

Paul

devel-bobm, 2017-09-06:
@pabender wrote:

> That bit of code was put into place anywhere we might need to acquire a throttle but don't necessarily have a UI for the throttle.

> Most scripts should be able to run in headless mode, so it makes sense for them to implement an automatic steal.

This is a great way to retain JMRI's historical behavior when operating in an environment which can "Steal".  

But I am not convinced that this is the "right" thing to do for LocoNet connections.  

With LocoNet, "stealing" is really more like "sharing".  And "sharing" causes problems for the user.

The most typical problem is that while one throttle is "actively" controlling the loco, another throttle is occasionally sending "refresh" messages at the last requested speed.  This often causes sudden changes in loco speed, occuring at the rate at which the "other" throttle sends its "refresh" message.

There will also be different system-level behavior if we update JMRI for compatibility with LocoNet "expanded" slots.  This "silent" stealing operation _will_ actually steal from an "expanded" slot user, and the throttle using the "expanded" slot _will_ give a "StealZap" message, and lose control.  But locos in "standard" slots will continue to have the "silent steal" (sharing) behavior described in the previous paragraph.

pabender, 2017-09-06:
On 09/05/2017 08:31 PM, Bob Milhaupt wrote:
> @pabender <https://github.com/pabender> wrote:
>
>     That bit of code was put into place anywhere we might need to
>     acquire a throttle but don't necessarily have a UI for the throttle.
>
>     Most [...]
 >
> With LocoNet, "stealing" is really more like "sharing". And "sharing"
> causes problems for the user.
 >
> The most typical problem is that while one throttle is "actively"
> controlling the loco, another throttle is occasionally sending "refresh"
> messages at the last requested speed. This often causes [...]

Yes, I understand that problem, and I know you're trying to prevent JMRI 
from causing this behavior.  As long as there are hardware throttles 
(e.g. the Zephyr) that cause the same issue, even in the absence of 
JMRI, we can't eliminate that behavior completely.

I am also looking at this from a more global support perspective in that
1) for scripts we distribute which don't use any system specific code, 
we need to make sure they work the same way on different systems.
2) on headless systems, we have no way to ask the user if they want to 
steal.  Scripts we distribute need to run on headless systems unless 
they explicitly require a UI (there are some scripts we distribute that 
create windows)

> There will also be different system-level behavior if we update JMRI for
> compatibility with LocoNet "expanded" slots. This "silent" stealing
> operation /will/ actually steal from an "expanded" slot [...]

Think about, for example, the BackAndForthTimed.py script we distribute.
If one of our users has a headless system with a LocoNet connection, 
then the right behavior is ALWAYS to steal the slot if that is required. 
If it doesn't, the the user is going to be confused about why his engine 
that always moves back and forth stopped moving back and forth.

In this case, we don't really care if the steal is from a "standard" or 
"expanded" slot.

Paul

Pugwash1, 2017-09-06:
Using the latest master I cannot get control of any loco stolen or not. Factory reset DCS50, start JMRI, create new throttle. Get throttle but nothing happens. Look at slotmon and the slot has a throttle ID but status is "Free".  I can acquire that loco on a dt401 throttle, no stealing required. But the slot is broken, no dcc signals are sent out on the track.

devel-bobm, 2017-09-07:
@Pugwash1, I cannot duplicate your problem.

If you cannot control a loco from a physical throttle, then you should not expect JMRI to do any better!  Start by solving that problem, then try JMRI again.

devel-bobm, 2017-09-07:
@pabender,

I give up.  Bad LocoNet behavior wins, just because "that's the way we've always done it".  That's just plain wrong.

I will stop banging my head against the wall, because the wall always wins.

closed 2017-09-07 by devel-bobm

KenC57, 2017-09-07:
Bob M,

 

I think part of this is that the steal concept is the first time our
throttles have the idea of 'No'. Prior to this, it never had to handle the
idea of asking for a throttle slot and being told 'No'.

 

My thought about this is that it should, but there are likely things still
in place that make this not work out. Once the whole concept of steal being
a yes-no result, and handling that right; then we should be able to handle
the CS telling us no too.

-ken c

pabender, 2017-09-07:
On Thu, Sep 7, 2017 at 11:01 AM, Ken Cameron <notifications@github.com>
wrote:

> Bob M,
> I think part of this is that the steal concept is the first time our
> throttles have the idea of 'No'. Prior to this, it never had to handle the
> idea of asking for a throttle slot and being told 'No'.
>

Actually, that's not entirely true.   The throttleListener objects have
always had the notifyFailedThrottleRequest() callback that indicates a
request has failed.  Handling that is separate from handling the steal
though.

Paul

pabender, 2017-09-07:
Just so everyone sees the sequence we are talking about here, I went ahead and (finally) documented the throttle creation sequence.

[image]

Note that only the part between "Optional Steal" and "Creating the throttle" was added recently.