JMRI/JMRI #4075: Econami UK version

AdrianGunzburg, 2017-09-09:
Soundtraxx now have a UK version of their Econami Diesel Decoder. DecoderPro is set up only for the US version and there are several of the Indexed CV’s that are particular to the UK sounds, eg. selection of horns (currently only US horns are included), and disabling functions, eg. CV 1.297 and 1.313. (they have a two-tone horn). 
Can we have a new version of DecoderPro specifically for the UK Econami please.
The Econami User Manual clearly identifies the separate features unique to the UK version.

AlanUS, 2017-12-10:
@AdrianGunzburg 
Cannot find data sheets with the specific UK sounds. Difficult to define a decoder definition without such information.

AdrianGunzburg, 2017-12-11:
Hi Alain,

George Bogatiuk at Soundtraxx will be able to supply this information.

georgeb@soundtraxx.com

Regards,

Adrian Gunzburg

AlanUS, 2017-12-12:
SoundTraxx Customer support just redirect me to their specific UK website:
http://soundtraxxuk.com/manuals.php#econami
where all required information is available.
Quite confidential website as none of my internet queries ever gave me a hit on it.
Anyway good enough to attempt new definitions.

AdrianGunzburg, 2017-12-13:
Hi Alain,

Yes I have a copy of the User Manual with the notes for UK variants.

However, from what I can see they don’t include the Prime Mover choices, so I’ve attached them for you from the Econami package as this will be needed for a usable DP version.

 

I’ve successfully installed and modified one of them into my Western Australian loco which has an English Electric prime mover. I managed to eliminate one of the horn tones (the lower one I think) as our loco only had one.

 

See how you go. I’m happy to act as a sounding board, although I’m not strong on Java programming. Happy to “Beta Test” it for you !

 

Cheers,

Adrian

 

http://soundtraxxuk.com/manuals.php#econami

AlanUS, 2017-12-13:
@AdrianGunzburg 
Attachments are removed when posting on GitHub.
However the data sheets with the UK prime movers and horns are available on the very same page I've indicated:
http://soundtraxxuk.com/manuals/Sound-Selection_UKSteam.pdf
http://soundtraxxuk.com/manuals/Sound-Selection_UKDiesel.pdf
So unless you have something different, no need to resend. Otherwise, please send a direct email to my email address, not going through GitHub.
Note that decoder definitions are written in XML. So no Java programming knowledge needed.
I'll work on that during the week-end to come.

AdrianGunzburg, 2017-12-13:
Hi again Alain,

One of the guys in Perth, Western Australia, has produced a 3D printed kit of this loco and is suggesting Loksound or Econami UK as suitable decoders. So, there could be a demand for a DecoderPro for the UK sounds from at least them, not to mention the UK users themselves.

Cheers,

Adrian

 

http://soundtraxxuk.com/manuals.php#econami

AdrianGunzburg, 2017-12-13:
Hi Alain,

The pages you sent me links to do contain all the info I sent you, plus more, ie. air compressors.

 

So I don’t need to send it to you.  You didn’t give me your direct email address anyway.

 

I don’t have XML expertise either !  Look forward to hearing how you go with these new definitrions.

Regards,

Adrian

 

http://soundtraxxuk.com/manuals/Sound-Selection_UKSteam.pdf
http://soundtraxxuk.com/manuals/Sound-Selection_UKDiesel.pdf

dheap, 2017-12-13:
Attachments can be added to a comment on [GitHub](https://github.com/JMRI/JMRI/issues/4075).

AlanUS, 2018-01-06:
Econami UK added to JMRI  #4710

AdrianGunzburg, 2018-01-06:
Hi Alain,

WONDERFUL !!  Many thanks ! I’m sure the UK users will be pleased !  I’ll update my version.

Cheers,

Adrian

AlanUS, 2018-01-07:
@AdrianGunzburg Thanks. This will require testing as there are some unclear or contradictory information in SoundTraxx documentation.

This issue can be closed now.

AdrianGunzburg, 2018-01-07:
We are going to be away for a week, so will have a detailed look when we return.

Cheers,

Adrian

AlanUS, 2018-01-07:
No hurry. As I wrote, I expect an update in a week or so. SoundTraxx already replied they would look at my questions next week.

AlanUS, 2018-01-28:
Got replies from SoundTraxx. Adjustments made.
Now available in JMRI 4.11.2 though not credited yet.
http://jmri.org/releasenotes/jmri4.11.2.shtml

AdrianGunzburg, 2018-01-28:
Thanks !  

 

http://jmri.org/releasenotes/jmri4.11.2.shtml

AlanUS, 2018-05-19:
@AdrianGunzburg 
Thanks for the feedback. Good to hear that it works as expected.

closed 2021-04-14 by stale[bot]