JMRI/JMRI #4080: Loading Panels i18n

silverailscolo, 2017-09-10:
After changing Locale, loading a LE Panel from file gives a list of (different language) color names as not recognized.
Panel colors are store in xml as English lower case strings, so I will try to sort out why this happens. I18N preferably should store in default values to make changing a Locale back and forth with existing items is fluent (at least while developing).

geowar1, 2017-09-10:
Currently LE colors are saved using ColorUtil.colorToString which maps the exact colors (black, darkGray, gray, lightGray, white, red, pink, orange, yellow, green, blue, magenta) to their (hardwired English) names. On load ColorUtil.stringToColor is used to do the opposite. The only exception is the background color which is saved and loaded using it's red, green & blue components. I recently changed its loading code to use ColorUtil.colorToColorName because it generates hex strings (#rrggbb) for colors other than the named colors above. One way to avoid the local issue would be to always save these as hex strings. The ColorUtil.stringToColor method (used by all LE color attributes other than background) could be modified to parse hex strings if/when it's unable to recognize a color by its (localized) name.
You may assign me to investigate unless you'd like to do it yourself. ;-)
If you assign it to me could you send me (<geowar 1 at mac dot com>) the file that's not loading?

geowar1, 2017-11-01:
@silverailscolo: can you test if this was fixed by the #4287 merge? And close this if so? Otherwise /poke (eMail) me and I’ll investigate further. Thanks.

silverailscolo, 2017-11-01:
Splendid, thanks guys. Issue closed.
Switching Locales and loading panels and Gui Console prefs works fine & quiet in the console output. Also after switching to a diffferent macOS primary language.
A minor item, just 1 line in fact, that's still there reads: 
``2017-11-01 21:57:21,093 rix.AbstractSimulatorConnectionConfig WARN  - Loading found invalid value for option SensorDefaultState, found "Unknown", setting to "Onbekend" [init prefs]``
But "Onbekend" is the correct choice (in Dutch) for an unknown value.

closed 2017-11-01 by silverailscolo