JMRI/JMRI #4091: How to use angular web app?

c-andrews, 2017-09-11:
Hi, Could you tell me how to use the angular web application. I load it up using **http://localhost:12080/app/** but i get a server error looking for **http://localhost:12080/app/script/**?

Also Angular gives me a modual load error:
jquery.min.js:2 Uncaught Error: [$injector:modulerr] http://errors.angularjs.org/1.6.6/$injector/modulerr?p0=jmri.app....

Any help would be really appreciated.

rhwood, 2017-09-11:
Can you please attach the JMRI logs from this, as well as the console log from your browser?

In Firefox, right click anywhere on the page and select "Inspect Element" and the Console will be one of the tabs in the new window or section of window that opens. That can be cut and pasted into a log.

Google Chrome and macOS Safari are similar to Firefox (I don't have a Windows system to speak to Edge or Internet Explorer).

In the console, when watching the network tab, and reloading the page, what URLs have a 4xx or 5xx status?

c-andrews, 2017-09-12:
Thanks, here you go. Please let me know if you need anything else.

[JMRI_system_console.txt](https://github.com/JMRI/JMRI/files/1295296/JMRI_system_console.txt)
[image]
[image]

rhwood, 2017-09-12:
@c-andrews Can you please zip the directory /Users/Chris/Library/Preferences/JMRI/My_JMRI_Railroad and attach it? This will allow me to see what unexpected value in your configuration is causing this, as I cannot recreate it on my own.

JMRI does not like something in the "Railroad Name" setting you have in the configuration profile, and before you make a change to that, I would like to see what is currently there, so that future versions of JMRI can be coded defensively around that.

c-andrews, 2017-09-12:
Of course, here you go

[My_JMRI_Railroad.zip](https://github.com/JMRI/JMRI/files/1295551/My_JMRI_Railroad.zip)

c-andrews, 2017-09-14:
@rhwood any luck with the issue?

rhwood, 2017-09-14:
I have been unable to recreate this specific error with 4.8 or the current nightly builds. I have, however, found some other problems with 4.8 that are fixed in the nightly builds (and will be fixed in 4.9.4 when it is released).

It might, unfortunately, be best to add a note to the release notes and just not use the app with 4.8. Sorry about that.

c-andrews, 2017-09-14:
No problem, i will hold off until 4.9.4 is released.
Thanks for taking a look

c-andrews, 2017-09-16:
Just to let you know that I have managed to get the Angular app working now. I re-downloaded the src from Git. As i'm on a Mac i had to copy the vecmath-1.5.2.jar into ~/Library/Java/Extensions

closed 2017-09-16 by c-andrews