JMRI/JMRI #4122: Masts and Mast Logic Silently not Loading

KenC57, 2017-09-16:
I believe I have a panel that works with 4.9.2, silently fails to load signal mast logic in 4.9.3 and silently fails to load masts and logic in 4.9.4ish.

I'm not 100% about the 4.9.2 as I'm not able to confirm on that system right now. It may be 4.8 or 4.9.1, I will find out later. When I say silent, I mean nothing is logged to the console. In 4.9.4ish, should I try to look at the signal masts, I get a dump from a null being seen during table rendering. Here' the top:

2017-09-15 21:34:28,705 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.NullPointerException
	at jmri.jmrit.beantable.RowComboBoxPanel.getRendererComponent(RowComboBoxPanel.java:260)
	at jmri.jmrit.beantable.RowComboBoxPanel.getTableCellRendererComponent(RowComboBoxPanel.java:235)
	at javax.swing.JTable.prepareRenderer(Unknown Source)
	at javax.swing.plaf.basic.BasicTableUI.paintCell(Unknown Source)

[CnyMod50.zip](https://github.com/JMRI/JMRI/files/1308274/CnyMod50.zip)

silverailscolo, 2017-09-17:
After removing the Panels and Logix, most Turnouts and Heads, with DEBOG on for beantables gives:
```
2017-09-17 08:38:07,069 beantable.SignalMastTableAction       DEBUG - setMenuBar number of menu items = 1 [AWT-EventQueue-0]
2017-09-17 08:38:07,140 beantable.BeanTableDataModel          DEBUG - RCBP getRendererComponent (row=0, value=null) [AWT-EventQueue-0]
2017-09-17 08:38:07,141 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.NullPointerException
	at jmri.jmrit.beantable.RowComboBoxPanel.getRendererComponent(RowComboBoxPanel.java:260)
	at jmri.jmrit.beantable.RowComboBoxPanel.getTableCellRendererComponent(RowComboBoxPanel.java:235)
```
There's a clear null ther.

silverailscolo, 2017-09-17:
Fix for Signal Mast Table not opening is in PR #4127

KenC57, 2017-09-17:
So still have issue with mast logic. Attempting a 'discover pairs' on the panel ends up with a out of heap error. It doesn't cleanly figure its way out of that and takes process killing.
At least #4127 lets the mast table display with the 'unknown' aspects.

silverailscolo, 2017-09-17:
Seeing the same (at least the Discovering pane freezes) after changing some paths (on a Mac) in your Panel and activate debugging.
On the LE panel, manually trying to add SML for mast B10B11 (protecting Block 11, facing Block 10), another NPE occurs:
```
2017-09-17 19:01:20,947 implementation.DefaultSignalMastLogic DEBUG - setSignalAppearance called for B10B11 [AWT-EventQueue-0]
2017-09-17 19:01:20,949 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.NullPointerException
	at jmri.jmrit.signalling.SignallingSourcePanel.propertyChange(SignallingSourcePanel.java:195)
	at java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:335)
	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:327)
	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:263)
	at jmri.managers.DefaultSignalMastLogicManager.firePropertyChange(DefaultSignalMastLogicManager.java:311)
	at jmri.managers.DefaultSignalMastLogicManager.discoverSignallingDest(DefaultSignalMastLogicManager.java:431)
	at jmri.jmrit.signalling.SignallingSourcePanel.discoverPressed(SignallingSourcePanel.java:174)
```

silverailscolo, 2017-09-17:
When using the Signal Mast Logic Table > Tools > Auto Generate Signaling Pairs menu is invoked on Ken's panel containing some 20 turnouts, 410 sensors, 50 heads and 28 masts, after some time (2 minutes?) an ``Unrecoverable Erro Encountered`` pane forces PanelPro to exit. Console says:
```
2017-09-17 19:39:22,161 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [Discover Signal Mast Logic]
java.lang.OutOfMemoryError: GC overhead limit exceeded
	at java.lang.AbstractStringBuilder.<init>(AbstractStringBuilder.java:68)
	at java.lang.StringBuilder.<init>(StringBuilder.java:89)
	at java.lang.Object.toString(Object.java:236)
	at jmri.util.SystemNameComparator.compare(SystemNameComparator.java:32)
	at java.util.TreeMap.put(TreeMap.java:552)
	at java.util.TreeSet.add(TreeSet.java:255)
	at java.util.AbstractCollection.addAll(AbstractCollection.java:344)
	at java.util.TreeSet.addAll(TreeSet.java:312)
	at jmri.managers.AbstractProxyManager.lambda$getNamedBeanList$1097(AbstractProxyManager.java:420)
	at jmri.managers.AbstractProxyManager$$Lambda$112/902478634.accept(Unknown Source)
	at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1374)
	at java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:580)
	at jmri.managers.AbstractProxyManager.getNamedBeanList(AbstractProxyManager.java:419)
	at jmri.util.swing.JmriBeanComboBox.<init>(JmriBeanComboBox.java:80)
	at jmri.jmrit.display.layoutEditor.LayoutEditorTools.<init>(LayoutEditorTools.java:4379)
	at jmri.jmrit.display.layoutEditor.LayoutBlockManager.getFacingBean(LayoutBlockManager.java:1591)
	at jmri.jmrit.display.layoutEditor.LayoutBlockManager.getFacingSignalMast(LayoutBlockManager.java:1447)
	at jmri.jmrit.display.layoutEditor.LayoutBlockManager.getFacingSignalMast(LayoutBlockManager.java:1437)
	at jmri.jmrit.display.layoutEditor.LayoutBlockConnectivityTools.findBestHop(LayoutBlockConnectivityTools.java:620)
	at jmri.jmrit.display.layoutEditor.LayoutBlockConnectivityTools.getLayoutBlocks(LayoutBlockConnectivityTools.java:448)
	at jmri.jmrit.display.layoutEditor.LayoutBlockConnectivityTools.checkValidDest(LayoutBlockConnectivityTools.java:305)
	at jmri.jmrit.display.layoutEditor.LayoutBlockConnectivityTools.checkValidDest(LayoutBlockConnectivityTools.java:360)
	at jmri.jmrit.display.layoutEditor.LayoutBlockConnectivityTools.discoverPairDest(LayoutBlockConnectivityTools.java:791)
	at jmri.jmrit.display.layoutEditor.LayoutBlockConnectivityTools.discoverValidBeanPairs(LayoutBlockConnectivityTools.java:733)
	at jmri.managers.DefaultSignalMastLogicManager.automaticallyDiscoverSignallingPairs(DefaultSignalMastLogicManager.java:449)
	at jmri.jmrit.beantable.SignalMastLogicTableAction$5.run(SignalMastLogicTableAction.java:535)
	at java.lang.Thread.run(Thread.java:745)
```
This looks like the Layout Editor is getting stressed out trying to keep updating its GUI comboboxes. Between 4.9.2 and 4.9.4 a lot of changes were made to LE, so I hope @geowar1 is willing to chime in and advice on how we can have LE wait until the SML autocreation is completed before - only once - updating the comboboxes in the LE toolbar.
There is threading involved as well in the start up of SML autocreation (too advanced stuff for me).

geowar1, 2017-09-17:
Line 449 is LayoutBlockConnectivityTools LayoutBlockConnectivityTools… not LayoutEditor (I haven't touched that…).
When this was first posted I wondered if the custom JmriBeanComboBox render (EnabledComboBoxRenderer)  I wrote was the issue… but the RowComboBoxPanel isn't a JmriBeanComboBox (it's just a JComboBox). Sorry… :(

silverailscolo, 2017-09-17:
Thanks, @geowar1

silverailscolo, 2017-09-17:
@geowar1 Then again, I was hinting at ``at jmri.jmrit.display.layoutEditor.LayoutEditorTools.<init>(LayoutEditorTools.java:4379)``, about halfway the last error code block quoted. That's wat causes the memory issue.
Plus I don't understand why asking LE for connections would cause those JmriBeanComboBox to be built at all. We're not in LE Edit mode.
I also saw your name alongside line 305 of LayoutBlockConnectivityTools (a recent change of ``blockList`` as a List instead of an ArrayList).

geowar1, 2017-09-17:
`<init>(LayoutEditorTools.java:4379)` _may_ be a bit of a red herring here… creating a LayoutEditorTools means instantiating a bunch of controls (including JmriBeanComboBoxes). Those could be lazy instantiated (deferred to the actual method calls that use them) to conserve memory. Another option might be to make LayoutEditorTools a singleton class (so there can only ever be one).

... line 305 of LayoutBlockConnectivityTools: [`List vs. ArrayList`](https://stackoverflow.com/questions/147468/why-should-the-interface-for-a-java-class-be-preferred
)

geowar1, 2017-09-17:
Looking at the sources… the only reason that code is instantiating a LayoutEditorTools instance is to call isAtWestEndOfAnchor()… I looked at making that static but it has calls that use layoutEditor (a non-static class variable). It's not dependent on anything else in the LayoutEditorTools class… one fix (to avoid all the extra baggage you get with LayoutEditorTools) would be to move isAtWestEndOfAnchor into the LayoutEditor class (since the calling code already has one of those).

geowar1, 2017-09-17:
Nope: LayoutBlockManager calls isAtWestEndOfAnchor and doesn't have a reference to the current LayoutEditor… so can't do that… :(

silverailscolo, 2017-09-17:
I also came across the isAtWestEndOfAnchor() method. Debug shows that from there autocreating SML pairs causes a huge call on resources. To discover SML for just 3 signal masts, there are 91 calls for turnout1ComboBox alone, 4,950 in total without any graphic use.
[image]
Only building the JmriCombos when a toolbar is visible in LE might do the trick. Or maybe we can add some filter for graphics between LayoutBlockManager and LayoutEditor. 
For tonight I'll only add the debug lines in PR #4129.

geowar1, 2017-09-17:
Then again… isAtWestEndOfAnchor only needs the LayoutEditor reference to call getCoords… and the only LayoutEditor instance variable is the layout name in a debug statement… If we remove that then getCoords and isAtWestEndOfAnchor could both be static methods and LayoutEditorTools wouldn't have to be instantiated to call isAtWestEndOfAnchor.

geowar1, 2017-09-18:
PR #4131 has been merged; Let me know if it helps(/solves) this issue. ;-)

silverailscolo, 2017-09-18:
Can confirm that the out of memory issue is fixed using the supplied panel xml file after PR #4131 . The whole process is also 10-100 times faster, and finishes cleanly.

geowar1, 2017-09-18:
Thinking about the issue… I think that their is still more that can be done… currently the showPopup code is per-instance (of LayoutTrack)… And the (LayoutTrack) Edit dialogs also. Those can be made per-LayoutEditor instance singletons (one per layout instead of one per LayoutTrack instance). That should free up a lot of duplicate variables.

rhwood, 2017-09-18:
Fixed by #4122

closed 2017-09-18 by rhwood