JMRI/JMRI #4123: PipeReaderTest incorrect?

bobjacobsen, 2017-09-16:
(See #4121 CI result)

There may be an error in PipeReaderTest's logic.  The code is:
```
        JTextArea jta = new JTextArea();
        PipedWriter wr = new PipedWriter();
        PipedReader pr = new PipedReader(wr,1);
        PipeListener t = new PipeListener(pr,jta);
        t.start();
        wr.write("Test String");
        wr.flush();
        jmri.util.JUnitUtil.waitFor(()->{return !(pr.ready());},"buffer empty");
        Assert.assertEquals("text after character write","Test String",jta.getText());
```
which fails occasionally as 
```
[ERROR] PipeListenerTest.testWrite:37 text after character write expected:<Test Strin[g]> but was:<Test Strin[]>
```

But note that the `waitFor` is only waiting for the `PipedReader` to be empty; not for it to be _finally_ empty.  Some of the characters might arrive before others.

Unless there's some semantics here that I'm missing, the test for equality should be moved into another `waitFor`.

pabender, 2017-09-16:
@bobjacobsen,  The wait for is actually correct.once the flush is executed on wr, all the data should be flushed into the read buffer for pr, and pr's ready 

I think the real question is how quickly does the text area value ( which is what we're looking at ) get written in the class under test.  The two lines in PipeListener that impact this are:

                    c[0] = (char) pr.read();
                    ta.append(new String(c));   // odd way to do this, but only

Those two lines are not atomic, and because we are testing a thread that does a task, There is opportunity for interleaving the test operations with the class operations

pabender, 2017-09-16:
I updated the cinitial comment and the title to reflect the test we are really looking at here.

bobjacobsen, 2018-01-23:
Has anybody seen this CI failure recently?
```
[ERROR] PipeListenerTest.testWrite:37 text after character write expected:<Test Strin[g]> but was:<Test Strin[]>
```
If not, I think this #4123 is resolved and can be closed.

pabender, 2018-01-23:
I can't say I have seen it recently

rhwood, 2018-01-23:
I haven’t seen it recently.

closed 2018-01-23 by bobjacobsen