JMRI/JMRI #4132: Data Entry Error with cmri.CMRISystemConnectionMemo

CCatania01, 2017-09-18:
Test environment
PanelPro version 4.9.5ish+jenkins+20170917T1343Z+R41490fd
Java 1.8.0_144
Mac OS X x86_64 v10.9.5
MacBookPro

Starting with a clean profile, a single CMRI connection is created.
Added one SMINI node, address 20.
Opened Sensor table and added three sensors for bits 1, 5, and 10.
Opened JMRI System Console

Clicked Add and brought up the Add New Sensor window.
Selected System Connection C/MRI
Keyed in 200  on the fourth keystroke a 0, the first four warn messages appeared in the log.
WARN  - bit number not in range 1 - 999 in CMRI system name: CS2000 [AWT-EventQueue-0]
Keyed in 1 as the final entry and clicked Create.
CS20001 was added.

Cleared 20001 by backspacing, and keyed in 20010, no warning message, clicked Create.
CS20010 was added.

Cleared 20010 by backspacing, and keyed in 2000, next four warn message appeared, keyed 5 to complete the name, and Clicked Create.
CS20005 was added.

Saved panel, and quit.

Started PanelPro, selected profile.
Added a second CMRI connection with node 30.

Created two sensors for second CMRI connection.
Same Warn issue, plus Illegal character in System name.

Anyone else seeing this behavior with 4.9.5ish?

Screen shots attached.
[image]
[image]
[image]
[image]
[image]

bobjacobsen, 2017-09-18:
Are you seeing any issues other than the console log?  E.g. the object isn't being created, or similar?
(sorry, wrong button - reopening)

dsand47, 2017-09-18:
The warning message occurs in CMRISystemConnectionMemo#validSystemNameFormat due to continuous input validation.

Specifically line 317:   **if ((num - ((num / 1000) * 1000)) == 0) {**

When a CMRI node id ends with a zero, such as 30, entering 30+001 looks like 3+000 when the third zero is entered.  

The plus characters represent the node id boundary for explanation purposes, they are not actually entered in the input field.

Dave Sand

CCatania01, 2017-09-18:
The error occurs with two digit node addresses regardless of the value of the units digit.  The 4th digit being 0 zero is failure mode. The System name can be saved and looks proper in the xml.

Upon loading a panel with a System name which was previously flagged as an input error, produces the following errors.

silverailscolo, 2017-09-19:
The checking of user input and (if considered invalid) writing multiple warnings to the Console is temporarily skipped with PR #4141 
Remaining would be the format check itself (flagging valid items as false according to last item reported by @CCatania01 above) and ``Sensor C2S111001 refers to an undefined Serial Node.`` (which is not related to validSystemNameFormat()

CCatania01, 2017-09-19:
I agree with temporarily skipping that validation step.  A three digit node address fails also. 
 The test case connection did not have node 111 defined, however, bits 1 and 2 were defined in the panel xml.
2017-09-19 07:03:56,823 PanelPro.PanelPro                     INFO  - Main initialization done [main]
2017-09-19 07:04:07,634 cmri.CMRISystemConnectionMemo         ERROR - illegal character in number field of CMRI system name: C2S111001 [AWT-EventQueue-0]
2017-09-19 07:04:07,635 cmri.CMRISystemConnectionMemo         ERROR - illegal character in number field of CMRI system name: C2S111001 [AWT-EventQueue-0]
2017-09-19 07:04:07,636 cmri.CMRISystemConnectionMemo         ERROR - illegal character in number field of CMRI system name: C2S111001 [AWT-EventQueue-0]
2017-09-19 07:04:07,636 cmri.CMRISystemConnectionMemo         ERROR - illegal character in number field of CMRI system name: C2S111001 [AWT-EventQueue-0]
2017-09-19 07:04:07,637 cmri.CMRISystemConnectionMemo         ERROR - illegal character in number field of CMRI system name: C2S111001 [AWT-EventQueue-0]
2017-09-19 07:04:07,638 cmri.CMRISystemConnectionMemo         ERROR - illegal character in number field of CMRI system name: C2S111001 [AWT-EventQueue-0]
2017-09-19 07:04:07,638 cmri.CMRISystemConnectionMemo         ERROR - illegal character in number field of CMRI system name: C2S111002 [AWT-EventQueue-0]
2017-09-19 07:04:07,638 cmri.CMRISystemConnectionMemo         ERROR - illegal character in number field of CMRI system name: C2S111001 [AWT-EventQueue-0]
2017-09-19 07:04:07,639 cmri.CMRISystemConnectionMemo         ERROR - illegal character in number field of CMRI system name: C2S111002 [AWT-EventQueue-0]
2017-09-19 07:04:07,639 cmri.CMRISystemConnectionMemo         ERROR - illegal character in number field of CMRI system name: C2S111001 [AWT-EventQueue-0]
2017-09-19 07:04:07,639 cmri.CMRISystemConnectionMemo         ERROR - illegal character in number field of CMRI system name: C2S111002 [AWT-EventQueue-0]
2017-09-19 07:04:07,640 cmri.CMRISystemConnectionMemo         ERROR - illegal character in number field of CMRI system name: C2S111001 [AWT-EventQueue-0]
2017-09-19 07:04:07,640 cmri.CMRISystemConnectionMemo         ERROR - illegal character in number field of CMRI system name: C2S111002 [AWT-EventQueue-0]
2017-09-19 07:04:07,641 cmri.CMRISystemConnectionMemo         ERROR - illegal character in number field of CMRI system name: C2S111001 [AWT-EventQueue-0]
2017-09-19 07:04:07,641 cmri.CMRISystemConnectionMemo         ERROR - illegal character in number field of CMRI system name: C2S111002 [AWT-EventQueue-0]

CCatania01, 2017-10-06:
Needs to be tag as a bug and be addressed.  I thought it would be corrected in 4.9.4.

silverailscolo, 2017-10-07:
Updated in PR #4182. Remaining console logging is the same as with other systems. I thought that wasn't a bug but will take us some more time.

bobjacobsen, 2017-10-07:
@silverailscolo has written PR #4182 to make progress on this, but I'd like to comment more generally.

@CCatania01, I'm hoping that you still don't understand how things work around here.  (If I'm wrong, and you do understand and yet have decided to reject how we do things, well, then I guess you'll just have to [see figure 1](http://www.dslreports.com/forum/r1181086-UNIX-Humor-See-Figure-1))*  Standalone statements like:

> Needs to be tag as a bug and be addressed. I thought it would be corrected in 4.9.4.

imply a command & control approach that doesn't work, and are not intended to ever work. They just end up being borderline offensive. 

Egbert is putting his own time into improving an aspect of the user experience, and in the process there's some logging messages.  You don't like those, but your dislike does _not_ become a "needs to be tag as a bug and be addressed".  Particularly if it's something as minor a logging message, the way to get something done is to put your own time in on it.

It looks to me like PR #4182 is a reasonable approach to this; there might still be corner cases where log messages are emitted, but as a work in progress, I think that the merged #4182 closes this.  Thanks, Egbert. Chuck, if you think more is needed, create a PR.

Bob

* I picked that example of "see figure one" because of the AT&T connection.  The [version that seems most common](http://www.dourish.com/goodies/see-figure-1.html) refers to DEC VAX Software Problem Reports (SPRs). The [original](http://www.snerk.net/humor/figure1.txt), which came from MIT in the 70's, has too much [jargon](http://catb.org/jargon/html/distinctions.html) in it now to be really effective.

closed 2017-10-07 by bobjacobsen

CCatania01, 2017-10-07:
4182 is reasonable. I understand Egbert is working on this. It was not clear to me when it could be done. 
Sorry for any angst. 
Chuck