JMRI/JMRI #4133: Unexpected sensors and turnouts added.

dsand47, 2017-09-18:
As of 4.9.3, the following very strange behavior occurs:

1. Select Add sensor.  Select Internal.
2. Enter 101 and click on Create.
3. IS101 created.
4. Click on Close to close the Add panel.

Repeat using 201.
IS201 and IS202 created.

Repeat using 301
IS301, IS302, and IS303 created.

etc.

Each time the Add panel is closed and opened, the generated count goes up.  This applies to at least turnouts and sensors.  Changing connection type has no impact.  If I follow up and add a CMRI sensor, I will end up with 4 CMRI sensors.

I was researching the issue described in #4132 when I discovered this problem.  

Dave Sand

bobjacobsen, 2017-09-18:
Just to make it even weirder, on my Mac, I can see the effect described above if I click the "Close" button.  But if instead I close the Add window with the target on the upper-left corner of the window, e.g. the window decoration, the number doesn't seem to increase.

A bit of debugging:  `jmrit.beantable.AddNewHardwareDevicePanel` gets a `cancelListener` object from its constructor for handling the Cancel button.  SensorTableAction attaches that to its `cancelPressed(..)` method. That (line 175) sets the enclosing frame invisible and call's `dispose()`. It doesn't look like it's the close that's doing this.

What I think is happening (but somebody else will have to confirm) is that closing with the Cancel button causes the next open to reuse the same panel and/or frame, so that the "Create" button ends up with multiple listeners: The old one(s), plus one more.  Those all get fired when  "Create" is pressed.

bobjacobsen, 2017-09-18:
More:  `SensorTableAction` at line 103 creates and holds an `addButton` object. The same one is used every time "Add" is pressed, e.g. the window is created.  It passes that to the `AddNewHardwareDevicePanel` constructor along with an `okListener`.

`AddNewHardwareDevicePanel` line 89 adds `okListener` to 'addButton`, without removing any previous ones.  So there's your problem.

- Either recreate the GUI objects (preferably, all of them) each time through the GUI construction or
- clean up the GUI objects by removing all pre-existing configuration, listeners, etc.

silverailscolo, 2017-09-18:
I remember seeing that the close button on the pane as not following the cancel button when I was working on the Add new item interface. Sorry to have missed bug. At the time it stopped creating multiple items but apparently that effect has returned. Will try to fix that.

rhwood, 2017-09-18:
Fixed by #4135

closed 2017-09-18 by rhwood