JMRI/JMRI #4161: Incorrect initilization of jmri.jmrix.dccpp.dccppovertcp.Server

pabender, 2017-09-22:
Travis headed restarted a second time after
As seen in #4160 

[INFO] Running jmri.jmrix.dccpp.dccppovertcp.ServerFrameTest
ERROR - Proceeding to initialize class jmri.jmrix.dccpp.dccppovertcp.Server but initialization is marked as complete [main] jmri.InstanceManager.?()
java.lang.Exception: Thread "main"
at jmri.InstanceManager.getNullableDefault(InstanceManager.java:242)
at jmri.InstanceManager.getDefault(InstanceManager.java:193)
at jmri.jmrix.dccpp.dccppovertcp.ServerFrame$Initializer.getDefault(ServerFrame.java:169)
at jmri.InstanceManager.getNullableDefault(InstanceManager.java:271)
at jmri.InstanceManager.getDefault(InstanceManager.java:193)
at jmri.jmrix.dccpp.dccppovertcp.ServerFrame.getInstance(ServerFrame.java:114)
at jmri.jmrix.dccpp.dccppovertcp.ServerFrameTest.getInstanceTest(ServerFrameTest.java:21)
at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
at java.lang.reflect.Method.invoke(Method.java:498)
at org.junit.runners.model.FrameworkMethod$1.runReflectiveCall(FrameworkMethod.java:50)

rhwood, 2017-09-23:
Keeping open until issue not observed in testing over a few days period.

rhwood, 2018-03-16:
Assignee is no longer involved.

commit 06d9f917 (2017-09-23): Merge pull request #4167 from JMRI/rhwood-patch-1

Delay initialization of DCC++ over TCP server
Fixes #4161
closed 2021-04-14 by stale[bot]