JMRI/JMRI #4173: Create Test Release 4.9.5

bobjacobsen, 2017-09-24:
This is the fifth release of the 4.9/4.10 cycle. It's intended to be built around October 7th 2017 from HEAD of master.

bobjacobsen, 2017-10-08:
I’m seeing several outstanding issues with the 2nd version of the 4.9.5 files. In no particular order:

A) LocoNet specific problem with throttles (see #4226)

B) LayoutEditor grid / scrolling

C) SPROG-specific problem with Command Station mode and/or multiple connections and/or some computers and/or ...

It looks like #4226 is a possible solution for (A), but it has the comment "This works, but whether its the best way?”

I’m not sure whether there’s active work on (B), nor whether it’s considered significant enough to hold the test release

And, at least to me, it looks like there’s not enough understanding of what the real issue is in (C) to know what’s going on or how big an effect it is.  Two people have reported it, but they seem to be reporting different things, and the symptoms are not yet clear.

So what should we do with the 2nd set of files?

geowar1, 2017-10-08:
I'm working on B… but I'm not seeing the issue… (can't reproduce).

geowar1, 2017-10-08:
Ok, got it: see PR https://github.com/JMRI/JMRI/pull/4230

geowar1, 2017-10-08:
On Oct 8, 2017, at 10:32 AM, George Warner <notifications@github.com> wrote:
> I'm working on B… but I'm not seeing the issue… (can't reproduce).

Ok, think I’ve got it… BUGFIX LayoutEditor grid / scrolling <https://github.com/JMRI/JMRI/pull/4230>
-- 
Enjoy,
George Warner
eMail/iChat: <geowar1@mac.com>

bobjacobsen, 2017-10-11:
The third set of 4.9.5 files are available at

http://jmri.tagadab.com/jenkins/job/TestReleases/job/4.9.5/

(use build 5).  I’d like to release these late Thursday, if at all possible.

bobjacobsen, 2017-11-07:
The current head of master (1fcd01d1a66d099863fb668bb0f015074f07f0fd ) was just now merged into the release-4.9.5 branch. The next set of files will be building shortly as as Build#7 via [the usual Jenkins task](http://jmri.tagadab.com/jenkins/job/TestReleases/job/4.9.5/) 

@JMRI/maintainers Please hold off merging any PRs for a bit (maybe a couple days) while we get this stabilized.  I'd like to be able to easily merge individual outstanding PRs if they're needed before 4.9.5 goes out, and having them not-yet merged makes that much more straight-forward.

bobjacobsen, 2017-11-07:
Suggested to add PR #4382 "Regression: LayoutEditor.LayoutTurnout popup menu “Set Sensors” dialog not getting turnout id from clicked on turnout"  from branch `geowar1/Regression-layout-editor.layout-turnout-popup-menu-“set-sensors”-dialog-not-getting-turnout-id-from-clicked-on-turnout` to this. (That's maybe  reasonable PR title, but it's long for a branch name) The changes would be:

```
% git merge 'geowar1/Regression-layout-editor.layout-turnout-popup-menu-“set-sensors”-dialog-not-getting-turnout-id-from-clicked-on-turnout'
Merge made by the 'recursive' strategy.
 java/src/jmri/jmrit/display/layoutEditor/LayoutBlockManager.java           |   80 +--
 java/src/jmri/jmrit/display/layoutEditor/LayoutEditor.java                 |  274 ++++-----
 java/src/jmri/jmrit/display/layoutEditor/LayoutEditorBundle.properties     |   29 +-
 java/src/jmri/jmrit/display/layoutEditor/LayoutEditorChecks.java           |    2 +-
 java/src/jmri/jmrit/display/layoutEditor/LayoutEditorFindItems.java        |   15 +-
 java/src/jmri/jmrit/display/layoutEditor/LayoutEditorTools.java            | 1374 +++++++++++++++++++------------------------
 java/src/jmri/jmrit/display/layoutEditor/LayoutSlip.java                   |  168 +++---
 java/src/jmri/jmrit/display/layoutEditor/LayoutTurnout.java                |  187 +++---
 java/src/jmri/jmrit/display/layoutEditor/LayoutTurntable.java              |    6 +-
 java/src/jmri/jmrit/display/layoutEditor/LevelXing.java                    |   47 +-
 java/src/jmri/jmrit/display/layoutEditor/PositionablePoint.java            |  333 +++++------
 java/src/jmri/jmrit/display/layoutEditor/TrackSegment.java                 |  135 ++---
 java/src/jmri/jmrit/display/layoutEditor/configurexml/LayoutEditorXml.java |    8 +-
 java/src/jmri/jmrit/display/layoutEditor/configurexml/TrackSegmentXml.java |   20 +-
 java/test/jmri/jmrit/display/layoutEditor/loadref/LayoutEditorTest.xml     |   36 +-
 15 files changed, 1211 insertions(+), 1503 deletions(-)
```

geowar1, 2017-11-07:
How about "LayoutEditor: popup menus' set signal (mast & head) and sensor dialogs not getting correct object id's"?

bobjacobsen, 2017-11-07:
> How about "LayoutEditor: popup menus' set signal (mast & head) and sensor dialogs not getting correct object id's"?
> 

I don’t know what that means.

geowar1, 2017-11-07:
> That's maybe reasonable PR title, but it's long for a branch name
>>How about "LayoutEditor: popup menus' set signal (mast & head) and sensor dialogs not getting correct object id's"?

>I don’t know what that means.

It's hard to condense it without losing meaning… ;-)
Or: "LayoutEditor: Popup menus' "Set Signal [Mast/Head]" and "Sensor" dialogs not getting correct object id's"?

bobjacobsen, 2017-11-08:
It's the "How about" part that I don't understand.

bobjacobsen, 2017-11-08:
Ah, now that I'm looking on a real screen, not a phone, I think I see.  You're referring the branch name! I thought this was about what to include in 4.5.9

Branch names are _names_, not descriptions nor genealogies.  A branch name should be a short, convenient handle.  Preferably without spaces and punctuation, so that it can be easily handled as a single text item in scripts, cut&paste, etc.

bobjacobsen, 2017-11-08:
More content was added to the-branch-in-PR-4382 (#4382), so I've merged that into the release-4.9.5 branch. Jenkins will be remaking files as build 9 at [the usual place](http://jmri.tagadab.com/jenkins/job/TestReleases/job/4.9.5/)

bobjacobsen, 2017-11-09:
Released.

closed 2017-11-09 by bobjacobsen