JMRI/JMRI #4185: Throttles are too DCC centric

pabender, 2017-09-27:
I was working on some findbugs issues in the throttle GUI, and realized we have allowed the throttle interface to become very DCC centric.

The biggest immediate concern at this point is in requesting of throttles.  We require that both the throttle listeners (the front end code) and the throttle managers (the intermediary to the back end throttles) use jmri.DccLocoAddress objects to pass address information.  These references should be modified to reference jmri.LocoAddress objects (from which jmri.DccLocoAddress objects are derived).

The implementation concern here has always been for analog throttles, but we have just as much of a concern about other formats (The LocoAddress object specifies a Protocol enum that lists 9 different protocols, 3 of which are DCC, and analog throttles are not even a part of that...)

Paul

bobjacobsen, 2017-09-27:
While we're at it, we should revisit the design of  protocol handling, particularly how the `LocoAddress.Protocol` enum is embedded.  Its structure was greatly bent by the limitations on enums in earlier versions of Java.  With those gone, LocoAddress could be much cleaner.

danielb987, 2017-10-13:
How about jmri.ThrottleListener? Its method notifyThrottleFound() takes a DccThrottle as parameter.

Should this parameter DccThrottle be changed to Throttle?

pabender, 2017-10-13:
Sent from my iPhone

> On Oct 13, 2017, at 2:59 PM, Daniel <notifications@github.com> wrote:
> How about jmri.ThrottleListener? Its method notifyThrottleFound() takes a DccThrottle as parameter.
> 
There are others as well.
> Should this parameter DccThrottle be changed to Throttle?
> 

Ultimately, yes.

Paul

pabender, 2017-10-15:
Just to get this in the notes here, #4253 only partially addresses this, so when it is merged, we can't close this issue.

While thinking about #4253, the change from DccThrottle to Throttle MAY require some significant rework in the throttle front ends.  That change may be big enough that we should hold off on implementing it until after the next production release.

The particular thought is that while all throttles provide things like speed and direction control, only DCC throttles provide the mix of functions we provide today, so we need to be able to tell the DCC throttle front ends (the UI, but also the various servers that deal with throttles) what controls are supported for a given throttle type, which isn't something we do today.

the easiest place to see why this is necessary (and why a given throttle manager may actually handle different types) is in the SRCP client, where the protocol specifies multiple possible protocols, each with different parameters (and we only support one of them...)

devel-bobm, 2017-10-15:
I've been wondering about how the "Throttle" deals with the differences between "long" and "short" addresses.  
 
The case I wonder about is a system like NCE.  There, with a physical throttle, the address "044" is different from address "44".  If I recall correctly, one is for "short" addressing and the other is for "long" addressing.   The difference is important and necessary.  I wonder how "Throttle" would have to pass an "address" down to the NCE implementation of a "DCCThrottle".
 
Regards,
Bob M.

bobjacobsen, 2017-10-15:
> On Oct 15, 2017, at 3:27 PM, Bob Milhaupt <notifications@github.com> wrote:
> 
> I've been wondering about how the "Throttle" deals with the differences between "long" and "short" addresses.  
> 
> The case I wonder about is a system like NCE. There, with a physical throttle, the address "044" is different from address "44". If I recall correctly, one is for "short" addressing and the other is for "long" addressing. The difference is important and necessary. I wonder how "Throttle" would have to pass an "address" down to the NCE implementation of a "DCCThrottle".

“44 and “044” differ in the NCE (physical) throttle user interface, but nowhere downstream of that. Once they become actual addresses (e.g. bits/numbers), they’re flagged as different types.

The same is true in JMRI:  More info is used.  Sometimes it’s passing a long/short boolean value:

public boolean requestThrottle(int address, boolean isLong, ThrottleListener l);

Sometimes you pass a DccLocoAddress value which is either created with a boolean flag, or with an explicit protocol value:

public DccLocoAddress(int number, boolean isLong)

public DccLocoAddress(int number, LocoAddress.Protocol protocol)

The user interface then has to work it out.  Sometimes, JMRI asks you to select specifically:

Other times, it takes a single integer and asks the ThrottleManager (really, command-station-specific code) to sort it out.  

public boolean canBeLongAddress(int address);

public boolean canBeShortAddress(int address);

public boolean addressTypeUnique()

There’s lot of detail here, lots of history, but it’s really _not_ a good design.  All of this stuff is system-specific, and we should design the support to handle that.  (Java has come a _long_ way from when these structures were first created, as has JMRI, so this isn’t a criticism of that early effort; code wins, and that code has served well. But if we’re going to redo this, let’s _redo_ it)

I think the right structure now would be system-specific ThrottleManagers that you ask for various “selector” implementations:

- One or more Swing ones that can be embedded into GUI.  They know about long-vs-short in NCE vs Digitrax vs other things, but they’d _also_ be able to provide custom support for TMCC, OpenLCB, and anything else that comes along: Just implement GUI selectors that work with the specifics capabilities of those.
- Another one that’s meant for e.g. scripts, Warrants and other programmatic things that takes parameters to set it up.  An integer (and perhaps long/short selector) for DCC, but something else for other things.  
- Another one that works via JSON for the web
- Any others that are needed

The system-specific selector implementations then provide the Throttle implementation once an address has been selected and requests, hiding all that away from any abstract least-common-denominator (there might be an abstract base for the DCC cases, but that doesn’t have to stretch to do anything else too)

The Throttle implementation can then repeat the pattern:  It can provide a couple implementations for Swing vs programmatic speed setting, so that different systems can use e.g. 1-100 vs 1-128 vs 0-1.0 for speed.  Functions could be handled via array bound variables e.g. “getNumMomentaryFunctions” , "getNameMomFunction(int)”, “getValMomFunction(int)” along with setters; plus “getNumOnOffFunctions()”, “getNumAnalogFunctions()”, etc.

Bob
--
Bob Jacobsen
jacobsen@mac.com +1-510-708-5988 AIM, Skype JacobsenRG

danielb987, 2017-10-20:
One additional thought:

jmri.Throttle has hardcoded the ability to have 29 function keys. Function 0 - 28. It's true for DCC today, but it may be different for another system. And even DCC may be extended some day because of the desire for more functions on the engine, like more sounds. If you run your engine with a smartphone, it's possible to make a user interface that can handle more than 29 functions.

For me it would make sense to change:
void setFn(boolean f0); --> void setF(int function, boolean f0); 
boolean getFn(); --> boolean getF(int function);
setFnMomentary(boolean f0Momentary); --> setFMomentary(int function, boolean f0Momentary);
boolean getFnMomentary(); --> boolean getFMomentary(int function);
where Fn is F0, F1, F2, ...

And that way, the values of the function keys could be stored in a single list instead of 29 variables.

pabender, 2017-12-18:
#4565 does NOT address all of the issues here.  See my comments above from 
October 15th.

commit f7504bbe (2017-12-18): Merge pull request #4565 from danielb987/DccLocoAddressToLocoAddress4185

Use LocoAddress instead of DccLocoAddress
Closes #4185
closed 2021-04-14 by stale[bot]