JMRI/JMRI #4187: Proposed change to Manager name-validity-check routines

bobjacobsen, 2017-09-27:
I'd like to change the interface for checking system names (and perhaps eventually user names) so that they can be "partially checked" as entered.

I do _not_ think it's a good approach to expose regex's for this; aside from design issues, it's simply the case that regex's are an impractical way to do a complete job of partial validation without false positives and false negatives.

So I propose building the "check for partial validity" into the code by adding a routine (the second one in what follows) and changing a return type to express "OK for now, but not yet completely valid". For example, in TurnoutManager, this involves adding/changing it to:

```
    enum NameValidity {
        /**
         * Indicates the name is valid as is, 
         * and can also be a valid prefix for longer names
         */
        VALID,
         /**
          * Indicates name is not valid as-is, nor
          * can it be made valid by adding more characters;
          * just a bad start.
          */
        INVALID, 
        /**
         * Indicates that adding additional characters might (or might not)
         * turn this into a valid name; it is not a valid name now.
         */
        VALID_AS_PREFIX_ONLY
    }
    
    /**
     * Test if parameter is a properly formatted system name.
     *
     * @param systemName the system name
     * @return enum indicating current validity, which might be just as a prefix
     */
    @CheckReturnValue
    public NameValidity validSystemNameFormat(@Nonnull String systemName);

    /**
     * Test if parameter is a properly formatted hardware address
     *
     * @param systemName the system name
     * @return enum indicating current validity, which might be just as a prefix
     */
    @CheckReturnValue
    public NameValidity validAddressFormat(@Nonnull String address);
```

Happy to do the work for this, suggestions on how to improve it welcome.  I can make the time to implement it this weekend.

Bob

silverailscolo, 2017-09-27:
Please let me know how we can integrate this in the GUI front end. Will pause on this part of my PR #4182 .

bobjacobsen, 2017-09-28:
Hearing a bit of enthusiasm and no concerns, I'm going ahead and making this change.  The return-type change touch a lot of implementing methods, which I'm working through.  Expect to get done in a couple days.  Will post a branch once I've got something to show.

bobjacobsen, 2017-09-28:
@silverailscolo:
> Some validation is handled by getBitFromSystemName(), also called from ValidSystemNameConfig(). Would it be possible to pick up its detailed findings via an exception (parameter)?

Not sure what you mean there, and having seen that yet.  (Working from the `Manager` class downwards) Can you point me to an example?  Maybe the thing to do is change the return there too.  Using Exceptions to pass what's really returned information can get complicated fast.

silverailscolo, 2017-09-28:
Only interpreted from what I read that an Exception is easier to pass on (by not catching it) than a return value. Most calls to GetBitFromSystemName() are from VSNConfig(), but here are 2 examples from a VSNFormat():
``JMRI/java/src/jmri/jmrix/nce/NceSensorManager.java`` line 490
and 
``jmri/jmrix/dccpp/DCCppTurnoutManager.java`` line 155
and quite often the code in GBFSN() is duplicated in VSNF() so I suggest to insert it as in https://github.com/silverailscolo/JMRI/blob/5d015dd481e1355f4b347adcd010db66746e24a1/java/src/jmri/jmrix/maple/SerialAddress.java#L83

bobjacobsen, 2017-09-30:
After some thought, and an initial implementation, I think adding the `validAddressFormat` method is a mistake:
- It's not really needed, because single-system managers can just do `validSystemNameFormat(makeSystemName(suffix))` to get exactly the same result
- There's no useful implementation of that for the various proxy managers:  The hardware address if system specific
- (and it should have been called `validSuffixFormat` in any case, because the last part of the system name is really an arbitrary suffix:  It's not always a hardware address)

So I've omitted that from the PR I'll do shortly.

bobjacobsen, 2017-09-30:
On the question of Exception:  I think we're perhaps still not agreeing on the design.  

The code should use `validSystemNameFormat` when it wants to check. It's perfectly fine for that to return any of the possible return values: VALID, VALID_AS_PREFIX_ONLY, or INVALID.  _None_ of those are errors or require warnings.  They just are what they are.  The code that invokes `validSystemNameFormat` can show something to the user, or log a message, or do something else; that's up to whoever is working on that code.

So this is encapsulating the validation in a single, well-defined routine (which can invoke other routines, and slowly get smarter about what it checks) and separating out how to deal with that validation.

bobjacobsen, 2017-09-30:
PR #4189 implements this now.  There's more work to be done because the code that implements validSystemNameFormat was migrated, but not improved. This means that

- [ ] It only returns VALID or INVALID the same way it would have returned true or false; the algorithms have not been updated to detect that they've been asked to check a partial name which can be tagged as VALID_AS_PREFIX_ONLY. That's a very system-specific thing that somebody else will have to eventually do.
- [ ] The logging of warnings and errors are still in place (mostly). This needs to get refactored so they're no longer in validSystemNameFormat as warnings and errors, but rather become debug messages there; meanwhile, the e.g. getTurnout(..) and provideTurnout(..) messages that invoke them should log that an improper name was provided. (This allows GUI code to intercept invalid names before calling get or provide, leaving the normal user-level log completely clean) I'll do this on a time-available basis over the next few days in a PR that follows #4189.

closed 2019-08-14 by bobjacobsen