JMRI/JMRI #4207: Panel Validation fails with trackSegmentConnect2NameValueExists error

KenC57, 2017-10-04:
Here are two files from the user. One was an earlier version (Mk14) and the current one that fails the validation (Mk17).  He says that turning off the xml validation doesn't prevent it from failing. Since the CAN simulator is broken, I can't test further right now. The line number reported is the end of the LayoutEditor segment of the file.

2017-10-04 09:29:40,048 configurexml.ErrorHandler             ERROR - Parse error while parsing file /C:/Users/kcameron/JMRI/PanelProCanSim/Dartmouth_Mk17.xml Exception: org.jdom2.input.JDOMParseException: Error on line 1073: cvc-identity-constraint.4.3: Key 'trackSegmentConnect2NameValueExists' with value 'TO4' not found for identity constraint of element 'LayoutEditor'.
See http://jmri.org/help/en/package/jmri/configurexml/ErrorHandler.shtml for possibly more information. [AWT-EventQueue-0]
org.jdom2.input.JDOMParseException: Error on line 1073: cvc-identity-constraint.4.3: Key 'trackSegmentConnect2NameValueExists' with value 'TO4' not found for identity constraint of element 'LayoutEditor'.
	at org.jdom2.input.sax.SAXBuilderEngine.build(SAXBuilderEngine.java:232)

The files are in the zip.

[Dartmouth.zip](https://github.com/JMRI/JMRI/files/1355932/Dartmouth.zip)

bobjacobsen, 2017-10-04:
The errors are validation errors, so "turning off the xml validation doesn't prevent it from failing" probably means that the user didn't really turn off the validation.  Perhaps the turn-off script was run too late?

What the error is saying is that a problem was found in the test to "Insist that each tracksegment connection2name attribute value refer to an existing ident" by checking that the `connect2name` attribute of the `tracksegment` element exists as an `ident` attribute in one (exactly one, but that's another test) `tracksegment`, `positionablepoint`, `layoutSlip`, `levelxing`, `layoutturntable` or `layoutturnout` element. The specific value that appears as a "connect2name`, but not an `ident`, is "TO4".

Does the code normally ensure that condition is true? (That's a statement about how things are entered in LayoutEditor)

Note that the schema validation doesn't (but could) check that things like Turnouts, Sensors, etc are defined. That's because they could be defined in a another file.  But it does check that for layout editor, because all the layout editor information is written as a single top-level "LayoutEditor" element.

Paul-T, 2017-10-04:
" Perhaps the turn-off script was run too late?"
I am the user with this problem
The turn off script is the only script in Startup

rhwood, 2017-10-04:
If that script is run after the panels are loaded in startup, it is too late. You might want to ensure that script is the first item in the list of startup items.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

Paul-T, 2017-10-05:
There is only the turnoff script in Prefs-Startup.
Panel is launched manually

dsand47, 2017-10-07:
The problem is that all of the <layoutturnout> elements are missing from the <layoutEditor> element in Dartmouth_Mk17.xml.  TO4 is an ident for a layout turnout element.

If I copy the layout turnout elements from Mk14, then the panel loads.  There are lots of bad bean errors, but those are caused by missing turnout names in the copied layout turnout elements.

bobjacobsen, 2017-10-07:
(I edited the above comment to put back-quotes around the element references.  Entering something like `<layoutturnout>` without the back-quotes makes it looks like an HTML formatting command, so it doesn't get displayed)

KenC57, 2017-10-07:
What sort of thing is that element?? Normal turnout, slip, or what?? That
would be a clue for us.

-ken c

dsand47, 2017-10-07:
The problem is that **ALL** of the layout turnout elements are missing in Mk17.  Effectively, all of the turnout icons were deleted from the panel.  It appears that something failed in the save process.  Manually deleting a turnout from a panel also deletes the attached track segments.  In this case, the track segments still exist which causes the error during panel loading.

geowar1, 2017-11-01:
So, if I followed all the above correctly, the issue seems to be that the layout turnouts didn’t get saved. I'll have to investigate. (It's “in the queue “)

closed 2021-04-14 by stale[bot]