JMRI/JMRI #4229: SPROG Command Station Mode Intermittent Failure

dheap, 2017-10-08:
As per a discussion in jmriusers https://groups.yahoo.com/neo/groups/jmriusers/conversations/topics/141522

I've been able to (intermittently) reproduce the problem, but it seems to be timing-related. Turning on DEBUG for jmri.jmrix.sprog made it less likely to occur.

Log attached.

[messages.log](https://github.com/JMRI/JMRI/files/1367231/messages.log)

silverailscolo, 2017-10-09:
The messages regarding "Unable to convert ST258 to a valid hardware address" result from the SPROG managers only expecting a fixed length (1 char) prefix. Fix for now is to overwrite the "S2" suggestion in prefs and use a different single letter prefix for the second SPROG connection.
I'm working through all jmrix packages to fix the multi char prefix handling but that's not done yet. Thanks for pointing to SPROG; that put it higher on the list and has my attention.

crosland, 2017-12-27:
@dheap Wayne's latest post about his problem shows the same IO exception that I have reported on the jmriusers group. I have replied as such with the suggested workaround of enabling purejavacomm logging, which fixes the issue for me.

I have not gotten very far in debugging due the same fact that turning on logging makes the problem disappear. Following Dan's suggestion on jmriusers I cloned the pjc repo and started rebuilding from source. The problem is an error code returned from JTermios. That's as far as I have got so far.

crosland, 2017-12-27:
I have opened an issue on the purejavacomm project.

dheap, 2017-12-27:
@crosland See my note on the [jmriusers thread](https://groups.yahoo.com/neo/groups/jmriusers/conversations/messages/144179).

crosland, 2017-12-28:
My previous "improvements" to the SPROG command station code have actually made things worse. I think I understand what is going on now, and how to fix it.

crosland, 2018-01-13:
This has taken longer than I anticipated but I hope to have things working well enough to push my branch to git soon. The problem with pjc is fixed by changes to the SPROG command station code. The command station will now run in its own thread as it always used to, but was broken by my earlier changes. I also had to adress issues caused by asynchronous messages from, e.g., the power control being interleaved at the traffic controller with messages from the command station. The command stations messages/replies could get out of synch.

crosland, 2018-01-14:
I have now committed my branch crosland_sprog_command_station

closed 2021-04-14 by stale[bot]