JMRI/JMRI #4237: Add LayoutEditor Checks…

geowar1, 2017-10-11:
In the course of working on jmri I've collected and developed a lot of useful (external) scripts and tools to parse layout panel files and find, report & fix common mistakes. This is an effort to internalize those into the Layout Editor. Initially (this issue/PR) I'm going to add a Tools menu, Check item with sub-menu items "Disconnected Tracks" and "Non-Blocked Tracks". Initially these will just report their findings (via console)… but as I figure things out I'll add GUI to aid in the repair of these issues (maybe something as simple as just showing the relevant object's popup menu or editor).

(If you have or can think of things that you would like the Layout Editor to check (& repair?) please mention them here or email me privately (<geowar 1 at mac dot com>) if you prefer.)
And as always…
--
Enjoy,

geowar1, 2017-10-12:
https://github.com/JMRI/JMRI/pull/4242

geowar1, 2017-10-12:
Next check… “Check For...” => “Non-contiguous Blocks”... (Can anyone think of a reason why a block wouldn’t be contiguous?)

pnbgit, 2017-10-13:
It should be a warning not an error.  The only time I have used non-contiguous blocks was where I was using the block name so I could highlight or hide all the track segments in the yards (using a script) of one particular layout- the blocks were not being used in the usual way.
It would also be useful in the non-contiguous blocks check if it checked that the same block does not appear on more than one panel (if it doesn't check that already).  I made that mistake once and it caused some problems.

geowar1, 2017-10-13:
Right now it’s not really a warning nor an error… I’m just adding them all to a results menu and they’re disabled (no action). I haven’t thought up yet what I might want to happen when one of them is selected (highlight the tracks in the selected block?),

geowar1, 2017-10-14:
It’s not really ether (warning or error); it currently just adds all non-contiguous blocks to a (menu) list; selecting one of these blocks from the list doesn’t do anything (I could select all tracks in the selected block or something... (suggestions welcome).

geowar1, 2017-10-16:
Additional (proposed) checks (/auto-repairs?):
Zero length track segments are a pain to find... #1 contributor to both the “Un-Blocked Tracks” and “Non-Contigious Blocks” check failures. I’d like to add an option to “Repair Un-Blocked tracks” that will assign a block to a (un-blocked) track segment when both blocks for the adjacent track objects are the same.

geowar1, 2017-10-16:
I think that zero-length track segments are auto-created when you attempt to (drag) connect layout tracks (excluding track segments themselves… although you can’t drag a track segment; its endpoints are actually a different layout track (usually an anchor).

geowar1, 2017-10-31:
I'm considering adding "Auto-fix Un-blocked Zero-length Track Segments".
Which block they pick would be somewhat random (probably the 1st adjacent track that has an assigned block).
For non-zero length tracks the user will have to make the decision.

I could also check adjacent tracks blocks and add them a submenu for the user to select from.

pnbgit, 2017-10-31:
Yes, the zero length track segments are typically created when you drag two turnouts (or crossovers etc) together and auto-join them, as there has to be a track segment between them.  As you say, _if_  you have something set in the block name combobox, it will assign the zero length segment to that block.  In practice, this is just as likely to be the wrong block, where the user has forgotten to clear that combobox from the last time he/she used it.  It might be better, when a zero length segment is created by dragging two turnouts together, to assign the block from either one of those turnouts, if set, and ignore whatever is left in the block name combobox.

I like the auto-fix option for zero length segments (take the block name from either side) and for a track segment between two items that have the same block.

geowar1, 2017-10-31:
On Oct 31, 2017, at 10:41 AM, Pete Brownlow <notifications@github.com> wrote:
> I like the auto-fix option for zero length segments (take the block name from either side) and for a track segment between two items that have the same block.

Can you create new incident?

“Enhancement: LayoutEditor, auto-set track segment blocks"
Feel free to be more verbose in the description… something like:

When connecting LayoutTrack objects sometimes a zero-length track segment is created.
Currently it is created with the block selected in the toolbar combobox (if any).
Instead its block should be set to the block of ether of the objects that it is connecting.

When the user is creating a new track segment, the block should be copied from the object that the user is dragging the track segment from.
(IOW they’re extending the block for the newly created track segment.)

You may assign this to me. TIA.
-- 
Enjoy,
George Warner
eMail/iChat: <geowar1@mac.com>

geowar1, 2019-12-22:
Ok, I don't remember the PR but I'm pretty sure this has been fixed (I'm now unable to duplicate: the new (zero length) track segments are picking up the block from the turnout being dragged).
Please confirm fixed and close or if not post a reproducible case here so I can investigate further. TIA.

closed 2020-01-06 by geowar1