JMRI/JMRI #4243: Layout editor: Meta key does not work on Java 9 on Windows 10

danielb987, 2017-10-12:
In the class java.awt.event.MouseEvent, the method isMetaDown() doesn't return TRUE in Java 9 on Windows 10 when the user clicks on the right mouse button. It did that in Java 8. And the method isMetaDown() is used in jmri.jmrit.display.LayoutEditor to move items.

A short term solution is to add a check for "e.getButton() == BUTTON3" to make right mouse button work in Java 9.

geowar1, 2017-10-12:
On Oct 11, 2017, at 9:48 PM, Daniel <notifications@github.com> wrote:
> In the class java.awt.event.MouseEvent, the method isMetaDown() doesn't return TRUE in Java 9 on Windows 10 when the user clicks on the right mouse button. It did that in Java 8. And the method [...]

On macOS isMetaDown maps to the command key and BUTTON3 is the control key.

So… not the same thing here…

I did change the key binding code for menus to use:

        int primary_modifier = Toolkit.getDefaultToolkit().getMenuShortcutKeyMask();

That method is OS agnostic… IIRC it returns command for macOS and control for Windows (and Linux?).

(I probably should make that a static final class variable instead of calling that method every time I need it.)
-- 
Enjoy,
George Warner
eMail/iChat: <geowar1@mac.com>

danielb987, 2017-10-12:
I have done a PR #4246 which solves this problem on Windows.

I included a check for Windows so the change is only on Windows.