JMRI/JMRI #4248: Layout Editor: Suggest new popup menu item for track segment: "Add Block Value"

pnbgit, 2017-10-12:
Related to issue #2910 and #4247,  It would make it much quicker and easier to add block value 
labels if a track segment had an "add block value label" option on the 
popup menu, that added a block value label on top of the track, for the block that track segment is 
assigned to.  It would initially be centred on the track segment, with a fixed width and a visible black border. The user could then drag and adjust it as usual.  I think this would be much more intuitive than the current method.  If #4247 is implemented, then the block value placed in the track segment would use the user assigned default settings.

[image]

geowar1, 2017-10-15:
Please assign to me… TIA!

geowar1, 2017-10-17:
Hummmm… the block value label isn't attached to the track in any way (in software)… so I can't determine if the track already has a label… I suppose I could check the bounds of where I'd put a new label and see if there's one already there... I _think_ that that would be "doable". ;-)
QED otherwise… ;-)

dsand47, 2017-10-17:
The block value label belongs to a block.  A block consists of 1 to n track components.  Adding the **_block label_** option to track segments will confuse some people.  They might be inclined to add it to every track segment.  

Then there is also the question of whether this is included for turnouts, level crossings, etc.  These are frequently individual blocks and in the case of cross overs, slips, and crossings, they can have up to 4 blocks.

pnbgit, 2017-10-18:
At the moment there is no limit as to how many times the same block label is added, we rely on the user putting them where they want them, so I don't think this change, which just makes it quicker and easier to add them, will make it more confusing. 

 I wasn't thinking of adding this option for turnouts, crossovers etc. There isn't really space to overlay it onto those components without it looking cluttered and, as Dave says, some such components can have more than one block.  In most cases, if the block is only a turnout/crossover, the user won't want to clutter it with a block label. Normal usage is just to put block labels at key points, such as platforms, loops, storage sidings etc.  If a user really does want a block label associated with a single piece of pointwork, then they can still add it using the existing method.

This suggestion is all about streamlining the creation of the layout diagram, and I think this would help.

At the moment, adding a block label to a plain track segment is pretty tedious, particularly if you have quite a lot of them to do. I make it, 13 clicks, one double click, typing into 3 edit boxes, a right button drag and finally clearing an edit box again.  (It used to be even more clicks and drags, but George's recent change so that you can place labels over the track has reduced it a fair bit).  This idea would reduce it to 2 clicks, with an optional drag to refine the position once placed.

geowar1, 2017-10-30:
@dsand47: I think what he wants is just an easy way to create a block icon over a track segment; it's not really associated with the track segment (other than it's location). @pnbgit: is my assumption correct?

pnbgit, 2017-10-30:
Yes, that sums it up.  It is associated with the block that the track segment is allocated to, but the user would be free to drag it around after creating it in this way, just like you can move a signal mast or NX sensor after you have associated it with a block boundary.

geowar1, 2017-11-01:
You may want to change the issue name to say “Create Block Value Here”. Otherwise it’s “in my queue”. (Is IMQ an acceptable TLA? ;-)

closed 2021-04-14 by stale[bot]