JMRI/JMRI #4249: Multi-connection system support

silverailscolo, 2017-10-12:
About half of the jmrix systems have been 'migrated' several years ago in order to support more than one connection of the same "type". They have their SystemConnectionMemo ``instance()`` method deprecated or removed and a ``swing`` folder containing typical items. These migrated systems are added to the jmrix.ActiveSystemMenu using InstanceManager.

Is it a long term ambition to also 'migrate' the remaining systems? In that case, can an experienced developer list the essential steps, please?
Or could we decide to freeze some of those old style systems depending on 
A. The number of current users nearing 1, or 2;
B. A system by design not supporting more than one connected to JMRI (even then, one might hope to switch between 2 connections inside the same profile). 
I've searched the Technical Docs but found nothing on this topic. My mini project completing multi character connection prefix support might have to wait for the code migration but I find it scary to start this on my own.

pabender, 2017-10-13:
Yes, the long term plan is to migrate every system to support multiple connections.

Ultimately this is related to #1410.  I stopped working on that issue because it was obvious we didn't have enough tests in place to make sure things were functional even if they compiled without issue.

Paul

bobjacobsen, 2017-10-13:
In the long run, having common structure would be better from and understanding and maintenance perspective. 

In some cases, there’s no other reason to do it: There won’t ever be a JMRI install with two Maple connections. It’s hard to imagine two EasyDCC connections. Etc

I’m not sure where the line is and where it should be. 

A first step might be making the list of existing systems and their current state. 

Bob

pabender, 2017-10-13:
@bobjacobsen 

This should be the list of systems that haven't been migrated yet:
bash-4.3$ grep ActiveFlag ActiveSystemsMenu.java
        if (jmri.jmrix.easydcc.ActiveFlag.isActive()) {
        if (jmri.jmrix.grapevine.ActiveFlag.isActive()) {
        if (jmri.jmrix.oaktree.ActiveFlag.isActive()) {
        if (jmri.jmrix.rps.ActiveFlag.isActive()) {
        if (jmri.jmrix.secsi.ActiveFlag.isActive()) {
        if (jmri.jmrix.tmcc.ActiveFlag.isActive()) {
        if (jmri.jmrix.direct.ActiveFlag.isActive()) {
        if (jmri.jmrix.maple.ActiveFlag.isActive()) {
        if (jmri.jmrix.easydcc.ActiveFlag.isActive()) {
        if (jmri.jmrix.grapevine.ActiveFlag.isActive()) {
        if (jmri.jmrix.oaktree.ActiveFlag.isActive()) {
        if (jmri.jmrix.rps.ActiveFlag.isActive()) {
        if (jmri.jmrix.secsi.ActiveFlag.isActive()) {
        if (jmri.jmrix.tmcc.ActiveFlag.isActive()) {
        if (jmri.jmrix.direct.ActiveFlag.isActive()) {
        if (jmri.jmrix.maple.ActiveFlag.isActive()) {

As far as I know, the only other system still referencing ActiveFlag is NCE.  The NCE code uses ActiveFlag internally to activate some things (but I can't remember what).

pabender, 2017-10-13:
Here is a more condensed version of that list:

bash-4.3$ grep ActiveFlag ActiveSystemsMenu.java |sort |uniq
        if (jmri.jmrix.direct.ActiveFlag.isActive()) {
        if (jmri.jmrix.easydcc.ActiveFlag.isActive()) {
        if (jmri.jmrix.grapevine.ActiveFlag.isActive()) {
        if (jmri.jmrix.maple.ActiveFlag.isActive()) {
        if (jmri.jmrix.oaktree.ActiveFlag.isActive()) {
        if (jmri.jmrix.rps.ActiveFlag.isActive()) {
        if (jmri.jmrix.secsi.ActiveFlag.isActive()) {
        if (jmri.jmrix.tmcc.ActiveFlag.isActive()) {

(removing the duplicates)

So that's only 8 systems.

bobjacobsen, 2017-10-14:
> On Oct 12, 2017, at 6:43 PM, Paul Bender <notifications@github.com> wrote:
> 
> Here is a more condensed version of that list:
> 
> bash-4.3$ grep ActiveFlag ActiveSystemsMenu.java |sort |uniq
> if (jmri.jmrix.direct.ActiveFlag.isActive()) {
> if (jmri.jmrix.easydcc.ActiveFlag.isActive()) {
> if (jmri.jmrix.grapevine.ActiveFlag.isActive()) {
> if (jmri.jmrix.maple.ActiveFlag.isActive()) {
> if (jmri.jmrix.oaktree.ActiveFlag.isActive()) {
> if (jmri.jmrix.rps.ActiveFlag.isActive()) {
> if (jmri.jmrix.secsi.ActiveFlag.isActive()) {
> if (jmri.jmrix.tmcc.ActiveFlag.isActive()) {
> 
> (removing the duplicates)
> 
> So that's only 8 systems.

Thanks.

Nobody is making the RPS hardware currently, and the single layout that had a complete set no longer uses it.  It’s OK to keep it around in case a vendor starts creating it, I guess.

Maple and Oaktree are in similar situations:  Limited use, but no real expectation of new need for multiple systems.

SECSI and Grapevine might be like that too, I don’t know.

I don’t think “direct” even works, but haven’t checked.

TMCC is clearly in active use, but I’m not sure there’s a need for multi-system connections.

So the question becomes “what’s the cost and benefit of simplifying the overall structure by updating these, and does somebody think it’s worth that effort?”

Bob

--
Bob Jacobsen
jacobsen@mac.com +1-510-708-5988 AIM, Skype JacobsenRG

silverailscolo, 2017-10-14:
Apart from support for actual use with multiple systems, "migrating" would serve to remove bad examples from jmrix and also prevent the old style systems to choking when a user tries to change the prefix. Try to add a second E to the EasyDCC prefix box and watch the Turnout table fail. I'm willing to spend time migrating the old style packages.
We might add the above system support status in the jmrix technical docs and/or hardware help page.

Do I miss something when I apply the following migration steps:

- add a swing folder with an xComponentFactory class
- update the xMenu class to use this
- register xComponentFactory with InstanceManager in the xSystemConnectionMemo ctor
- add multichar prefix support in the xAdapters and xManagers, using the xSystemConnectionMemo
- start the xManagers from xSystemConnectionMemo.configure() using ``new Manager()`` instead of ``instance()``
- depracate instance() in the managers
- update the respective calls in jmrix.ActiveSystemsMenu and SystemsMenu (some already migrated system menus are still invoked using the old style in jmrix.SystemsMenu)

pabender, 2017-10-14:
Sent from my iPad

> On Oct 14, 2017, at 4:28 AM, Egbert Broerse <notifications@github.com> wrote:
> We might add the above system support status in the jmrix technical docs and/or hardware help page.
> 
> Do I miss something when I apply the following migration steps:
> 
> add a swing folder with an xComponentFactory class
> update the xMenu class to use this
> register xComponentFactory with InstanceManager in the xSystemConnectionMemo ctor
Ultimately, you also want to move GUI objects into that swing directory.  
> add multichar prefix support in the xAdapters and xManagers, using the xSystemConnectionMemo
> start the xManagers from xSystemConnectionMemo.configure() using new Manager() instead of instance()
With appropriate changes so they are connected to the memo. Typically, they need a reference to either the memo or the traffic controller.

Also, any object that is currently referencing either an instance method or pulling a default from the instance manager needs to be re-written so they either get passed the object at construction time or pull it from the memo when they need to.

This applies both to objects requiring the traffic controller and objects requiring managers.  Pulling the default from the instance manager can also apply to the system connection memo.
> depracate instance() in the managers
This should have already have been done, but if it hasn't, yes.
> update the respective calls in jmrix.ActiveSystemsMenu and SystemsMenu (some already migrated system menus are still invoked using the old style in jmrix.SystemsMenu)
You also need to remove all reference to the ActiveFlag if it still exists.  ActiveFlag has already been deprecated and can be removed.

And of course you MUST update the tests, and we really should test with real hardware ( because our tests aren't sufficient to catch problems we introduce as we go about doing this ).

Paul

silverailscolo, 2017-10-14:
Thanks @pabender. I added your steps to the first list.

jmrix migration progress:
- [ ] direct
- [ ] easydcc (migration under way by silverailscolo for 4.9.6ish)
- [ ] grapevine
- [ ] maple (no hardware?)
- [ ] oaktree (no hardware?)
- [ ] rps (no manufacturer, no hardware?)
- [ ] secsi
- [ ] tmcc

bobjacobsen, 2017-10-14:
I was going to create some profile startup tests for those (you out a sample profile in java/tests/apps/PanelPro/profile that holds the settings, then the test launches with those), but in the process I noticed that not all of them have a simulator connection option. It’s probably not worth adding one, though. 

Then I noticed that a couple of these don’t even ask for serial port options. So there might be some breakage already present...

silverailscolo, 2019-05-05:
Closing this issue now that all connection support multi-character prefixes, multiple connections and issue #1410 closed. If an additional Simulator is desired, it could be added based on this.

closed 2019-05-05 by silverailscolo