JMRI/JMRI #4251: Layout Editor need an "Edit" menu

geowar1, 2017-10-13:
Undo/Redo
Cut/Copy/Paste/
Delete/Duplicate
Find
Select All/Invert Selection/
[image]
[image]
[image]
[image]

geowar1, 2017-10-13:
I mean it _IS_ an editor… right? ;-)

pnbgit, 2017-10-13:
Yes please!  Undo is probably one of the biggest things missing from Layout Editor.

geowar1, 2017-10-17:
I don't know what build it support is in Java/Swing, etc... but I do know how it was implemented on macOS… and I can alway re-invent that wheel (but only if necessary). ;-)

closed 2020-05-04 by geowar1