JMRI/JMRI #4252: Add ability to rotate entire layouts orthogonality (for modular layout)

geowar1, 2017-10-13:
with an option to re-orientate all labels, train, blocks icons, etc.

pnbgit, 2017-10-13:
I know somebody who would be over the moon if he could do such a thing.  :-) I also recall having to completely redraw a panel from scratch when it was decided to put the display screen on the opposite side of the layout room.

Cliff-Anderson, 2017-10-13:
This capability would have been a big help when our club converted several Panel Editor and/or Control Panel Editor created Panels from conventional computer screen images with North at the top, to WiFi device screen images with the aisle at the bottom.  

A few of those Panels were relatively simple, but several were rather complex and as they say, "Murphy was an optimist."

geowar1, 2017-10-16:
Please assign to me… (so it will be in my “queue” ;-)

danielb987, 2017-10-16:
I'm currently working on the back end part of it (handling the drawing on screen and handling mouse events). I'm not sure about menu handling and xml files so I leave that part.

My current code is here:
https://github.com/danielb987/JMRI/tree/RotatePanel-%234252

I have made some design decisions and I'm not sure if I have made the correct ones. Maybe things should be done in another way.

I support rotation by 0, 90, 180 and 270 degrees, and mirroring the panel. It's possible to mirror and rotate the panel at the same time. I'm currently testing LayoutEditor and PanelEditor and both seems to work. There are still some bugs, popup menus is at the wrong places and icons are not always updated correctly.

A big concern is text. If the panel is mirrored I suppose that the text at least sometimes should not be mirrored. I think the best way is to mirror the label, but I'm not sure about the best way to handle this from a design perspective (how to not mess upp the code).

The calls to setOrientation in PanelEditor at line 427 and in Editor at line 429 is for testing only and shall be removed.

danielb987, 2017-10-16:
The changes I have done is in jmri.jmrit.display there I have added a method setOrientation to the inner class TargetPane that sets the orientation of the panel.

The method paint in TargetPane transforms the graphics to be drawn on before it paints the icons.

And I intercept the mouse events so I can transform the coordinates before the mouse event is sent to the icons on the panel.

geowar1, 2017-11-05:
Daniel's new branch: <https://github.com/danielb987/JMRI/tree/RotateAndMirrorPanel-%234252>

geowar1, 2019-12-22:
See PR's  #7734, #7740 and #7753. ;-)

bobjacobsen, 2020-01-13:
@geowar1 Can we close this one?

geowar1, 2020-01-13:
All those PR's have merged… so IMHO I think so

closed 2020-01-13 by geowar1