JMRI/JMRI #4259: Unable to create a layout block for a turnout after 4.9.4

dsand47, 2017-10-15:
After creating a layout block for a turnout, saving the changes and loading the panel, the following errors occur:

>     [java] 2017-10-14 22:22:45,462 layoutEditor.LayoutEditor             ERROR - LayoutBlock 'BL-TL-Left' not found when panel loaded [AWT-EventQueue-0]
>     [java] 2017-10-14 22:22:45,464 layoutEditor.LayoutTurnout            ERROR - bad blockname 'BL-TL-Left' in layoutturnout TO0 [AWT-EventQueue-0]

The xml file contains a reference to the layout block in the layout turnout object, but the layout block entry does not exist.

The cause is that LayoutBlockMangerXml was not writing the new layout block because the use count was zero.

The use count is being set to zero by LayoutTrackEditors:turnoutEditBlockPressed.

The problem was introduced by commit 75f9b83.  This commit migrated code from LayoutTurnout to LayoutTrackEditors.  

The original code:
```java
             // decrement use if block was already counted
             if ((block != null) && ((block == blockB) || (block == blockC) || (block == blockD))) {
                 block.decrementUse();
             }
```
block is not equal to block[BCD] for standard turnouts.

The new code:
```java
            if ((layoutTurnout.getLayoutBlock() != null)
                    && ((layoutTurnout.getLayoutBlock() == layoutTurnout.getLayoutBlockB())
                    || (layoutTurnout.getLayoutBlock() == layoutTurnout.getLayoutBlockC())
                    || (layoutTurnout.getLayoutBlock() == layoutTurnout.getLayoutBlockD()))) {
                layoutTurnout.getLayoutBlock().decrementUse();
            }
```
getLayoutBlock[BCD] returns getLayoutBlock if any of them are null, therefor always true for standard turnouts since they don't have [BCD] layout blocks.

There are a number of occurrences of this structure in LayoutTrackEditors that will need to be evaluated and addressed.

Dave Sand

geowar1, 2017-10-15:
Hey DaveS!
If you want assign this to me and I'll (un-)fix it… ;-)
(Why you shouldn't _always_ use the accessors...)

geowar1, 2017-10-15:
Ok, I'm about to commit my work here… Looking thru this code I decided to move the conditional decrementUse calls into the setLayoutBlock[BCD|AC|BD] methods of their respective layout track subclasses. (IMHO the "old way" required that the LayoutTrackEditor had to know too much about the object it was editing.) I'd appreciate a code review to make sure I didn't fub anything up (further). TIA!

geowar1, 2017-10-16:
@dsand47: When you get a chance regress with a build that includes #4242. If that fixes it please close this incident; If not then let me know so that I may investigate further.

dsand47, 2017-10-16:
The original problem is fixed.  However, it is not possible to easily fix the resulting inconsistency.

Trying to edit the turnout results in a NPE:
>
     [java] 2017-10-16 14:13:15,047 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
     [java] java.lang.NullPointerException
     [java] 	at jmri.jmrit.display.layoutEditor.LayoutTurnout.showPopup(LayoutTurnout.java:2402)
     [java] 	at jmri.jmrit.display.layoutEditor.LayoutEditor.showEditPopUps(LayoutEditor.java:5855)
     [java] 	at jmri.jmrit.display.layoutEditor.LayoutEditor.mouseReleased(LayoutEditor.java:5685)

```java
            if (getBlockName().isEmpty()) {
                jmi = popup.add(Bundle.getMessage("NoBlock"));
            } else {
>>>>            jmi = popup.add(Bundle.getMessage("MakeLabel", Bundle.getMessage("BeanNameBlock")) + getLayoutBlock().getDisplayName());
            }
```
The bad block name is still attached to the TOnn element but the layout block does not exist.  One fix is to temporarily assign the block to another track component which results in the creation of the missing layout block.  The other is to edit the xml and remove the bad reference.

Dave Sand

geowar1, 2017-10-16:
On Oct 16, 2017, at 12:39 PM, Dave Sand <notifications@github.com> wrote:
> The original problem is fixed. However, it is not possible to easily fix the resulting inconsistency.
> Trying to edit the turnout results in a NPE:
> 
>  [java] 2017-10-16 14:13:15,047 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
>  [java] java.lang.NullPointerException
>  [java] 	at jmri.jmrit.display.layoutEditor.LayoutTurnout.showPopup(LayoutTurnout.java:2402)
>  [java] 	at jmri.jmrit.display.layoutEditor.LayoutEditor.showEditPopUps(LayoutEditor.java:5855)
>  [java] 	at jmri.jmrit.display.layoutEditor.LayoutEditor.mouseReleased(LayoutEditor.java:5685)
>             if (getBlockName().isEmpty()) {
>                 jmi = popup.add(Bundle.getMessage("NoBlock"));
>             } else {
> >>>>            jmi = popup.add(Bundle.getMessage("MakeLabel", Bundle.getMessage("BeanNameBlock")) + getLayoutBlock().getDisplayName());
>             }
> The bad block name is still attached to the TOnn element but the layout block does not exist. One fix is to temporarily assign the block to another track component which results in the creation of the missing layout block. The other is to edit the xml and remove the bad reference.
> 
?!? Why doesn’t it exist? If the user doesn’t enter a block then it should be set to null… if he enters a invalid block it will be set to null; if he enters a valid block (existing or not) then it will be set to non-null (usage count == 1).

I tried every combo I could and I’m unable to reproduce…

Can provide the steps to reproduce this?

Perhaps this was a layout that was saved while the bug was there?

(In which case a manual edit is probably your only choice.)

-- 
Enjoy,
George Warner
eMail/iChat: <geowar1@mac.com>

dsand47, 2017-10-16:
If you run 4.9.4 or 4.9.5, you can re-create the problem by adding a new block to a turnout, save the panel and then re-start JMRI.  

The easy fix using the latest Jenkins is add the bad block to any other track component.  That will create the missing layout block and the reference from TOnn will now be valid.

Dave Sand

geowar1, 2017-10-16:
On Oct 16, 2017, at 3:04 PM, Dave Sand <notifications@github.com> wrote:
> If you run 4.9.4 or 4.9.5, you can re-create the problem by adding a new block to a turnout, save the panel and then re-start JMRI.
> The easy fix using the latest Jenkins is add the bad block to any [...]

IIRC this was fixed (in Add LayoutEditor Checks… <https://github.com/JMRI/JMRI/pull/4242>) when you mentioned that blocks weren’t getting saved…  (wrong usage count).

-- 
Enjoy,
George Warner
eMail/iChat: <geowar1@mac.com>

geowar1, 2017-10-29:
@dsand47 Can you confirm if this has been fixed and close this issue? TIA.

dsand47, 2017-10-29:
This has been fixed.

closed 2017-10-29 by dsand47