JMRI/JMRI #4261: Loop during shutdown with LocoNet throttle active

bobjacobsen, 2017-10-15:
Report of current HEAD going into a loop if you shutdown with a LocoNet throttle active:

```
	at jmri.jmrix.loconet.LocoNetThrottle.throttleDispose(LocoNetThrottle.java:346)
	at jmri.jmrix.loconet.LnThrottleManager.disposeThrottle(LnThrottleManager.java:285)
	at jmri.jmrix.AbstractThrottle.removePropertyChangeListener(AbstractThrottle.java:419)
	at jmri.jmrit.throttle.AddressPanel.notifyThrottleDisposed(AddressPanel.java:273)
	at jmri.jmrit.throttle.AddressPanel.propertyChange(AddressPanel.java:676)
	at jmri.jmrix.AbstractThrottle.notifyPropertyChangeListener(AbstractThrottle.java:470)
	at jmri.jmrix.loconet.LocoNetThrottle.notifyChangedSlot(LocoNetThrottle.java:436)
	at jmri.jmrix.loconet.LocoNetSlot.notifySlotListeners(LocoNetSlot.java:533)
	at jmri.jmrix.loconet.LocoNetThrottle.throttleDispose(LocoNetThrottle.java:346)
```

The problem is that `LocoNetThrottle.throttleDispose` notifies all `SlotListeners`, which goes through the `AbstractThrottle` implementation to `AddressPanel`, which then tells the `LnThrottleManager` its done, which does another `LocoNetThrottle.throttleDispose` od infinitum.

devel-bobm, 2017-10-17:
I will look into this.

In addressing the LocoNet throttle acquire issues(pull-request #4269), I removed some slot notificaiton code from `LnThrottleManager.java` .  This change might have solved the problem.  With that pull-request applied, and a live JMRI throttle open, if I exit using the PanelPro "Quit" button, I do not see anything evidence of such a loop in the log, and do not experience any lock-up

Lionel642, 2017-11-06:
Hi;

I provided a quick fix in this pull request : https://github.com/JMRI/JMRI/pull/4364

rhwood, 2020-05-10:
Fixed by #4364

closed 2020-05-10 by rhwood