JMRI/JMRI #4263: Error message during LocoNet throttle acquisition

bobjacobsen, 2017-10-15:
I've been trying to help a user with a recent JMRI build,  DCS50 Zephyr and LocoBuffer USB who's seeing this message sequence when acquiring a throttle:

```
loconet.LnThrottleManager             WARN  - Slot [1} has not gone IN-USE in required time, slot is in unreliable state. [AWT-EventQueue-0]
throttle.AddressPanel                 INFO  - Asked for 18 got 18(L) [AWT-EventQueue-0]
loconet.LnThrottleManager             WARN  - slot 1 address 18 is  already in-use. [AWT-EventQueue-0]
```

The slot was definitely _not_ in use before the attempt to acquire address 18 started.

The 'Slot [1] has not gone IN-USE in required time, slot is in unreliable state' message is also issued 19 times when running `jmri.jmrix.loconet.PackageTest`.

`git bisect` says that the PackageTest behavior started with commit [35b1f4a39725bc6e715762ece584b6edd6fac977] "Only set the slot information after the slot move is successful" which is part of the "Only set the slot information after the slot move is successful" PR #4226 which part of solving #4225.

Pugwash1, 2017-10-16:
Can you assign to me. Thanks. 
A listen process needs be killed before the slot move. I will also he the allowance for the slot going in use. 
We have never experienced a problem as a result of these messages.

devel-bobm, 2017-10-16:
Some comments related to this:

- The "SlotManager" manages the _beginning_ of the throttle acquisition process, and only references LnThrottleManager once a candidate slot is presented by the command station.

- For some unknown reason, there seems to be a non-trivial delay between when JMRI sends a request for a specific loco address ("OPC_LOCO_ADR" message) and when JMRI sees the LocoNet message response (as shown in LocoNet Monitor.  This is unexpected, and somewhat concerning.

- The delay noted immediately above _might_ be the source of the "WARN  - Slot [1} has not gone IN-USE in required time, slot is in unreliable state. [AWT-EventQueue-0]" log warning.  This delay may also be the source of the problem.

Investigation is ongoing.

devel-bobm, 2017-10-16:
An example of the unexpected delay, from LocoNet Monitor:

`19:34:51.320: [BF 02 08 4A]  Request slot for loco address 264.`
`19:34:52.443: [E7 0E 07 03 08 00 00 47 00 02 00 00 00 5F]  Report of slot 7 information:
	Loco 264 is Not Consisted, Free, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x00 0x00 (0).`

Note that more than _a second_ has elapsed between sending the address request and receiving the command station's reply.

devel-bobm, 2017-10-16:
I think that something is blocking the AWT-EventQueue thread.

If I have 2 copies of JMRI running on the same PC, with two LocoBuffer-USBs, I find that LocoNet traffic is slow in the JMRI instance which attempts to acquire the address, but is actually happening at the expected speed in the JMRI instance that is NOT attempting to acquire the loco.

Here's the LocoNet monitor from the JMRI instance where I acquired the loco (with more than a second's worth of delay between the first and second LocoNet messages):
```
19:44:00.847: [BF 14 51 05]  Request slot for loco address 2641.
19:44:01.955: [E7 0E 08 03 51 00 00 47 00 14 00 00 00 1F]  Report of slot 8 information:
	Loco 2641 is Not Consisted, Free, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x00 0x00 (0).
19:44:01.955: [BA 08 08 45]  Set status of slot 8 to IN_USE.
19:44:01.955: [E7 0E 08 33 51 00 00 47 00 14 00 00 00 2F]  Report of slot 8 information:
	Loco 2641 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x00 0x00 (0).
19:44:01.955: [BF 14 51 05]  Request slot for loco address 2641.
19:44:01.955: [E7 0E 08 33 51 00 00 47 00 14 00 00 00 2F]  Report of slot 8 information:
	Loco 2641 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x00 0x00 (0).
```

Here's the LocoNet Monitor log from the JMRI instance which only had LocoNet Monitor running, without any unexpected delay):

```
19:44:00.847: [BF 14 51 05]  Request slot for loco address 2641.
19:44:00.863: [E7 0E 08 03 51 00 00 47 00 14 00 00 00 1F]  Report of slot 8 information:
	Loco 2641 is Not Consisted, Free, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x00 0x00 (0).
19:44:00.878: [BA 08 08 45]  Set status of slot 8 to IN_USE.
19:44:00.909: [E7 0E 08 33 51 00 00 47 00 14 00 00 00 2F]  Report of slot 8 information:
	Loco 2641 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x00 0x00 (0).
19:44:01.845: [BF 14 51 05]  Request slot for loco address 2641.
19:44:01.877: [E7 0E 08 33 51 00 00 47 00 14 00 00 00 2F]  Report of slot 8 information:
	Loco 2641 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x00 0x00 (0).
```

devel-bobm, 2017-10-17:
Use of `Thread.sleep()` in LnThrottleManager per #4226 may be the source of the delay problem.  This is delaying the AWT-EventQueue.

devel-bobm, 2017-10-17:
I have reproduced the initial problem reported here with a DCS51.  

I believe that the use of Thread.sleep() in #4226 resulted in a race-condition - either the DCS51 responds before the "Thread.sleep" takes place, or it doesn't.  If the command station response is received before, then the acquire succeeds (with a one second delay).  If the command station response is received during the Thread.sleep() sequence, then the acquire fails after a the one-second (cumulative) delay.

closed 2021-04-14 by stale[bot]