JMRI/JMRI #4264: Layout Editor: Implement independent resize & rotate methods for Layout Tracks.

geowar1, 2017-10-16:
Right now dragging an endpoints of layout turnouts, slips, level crossings and crossovers combine rotating and resizing… and the results can be somewhat... wonky at times.

 I’d like to split these into two different operations perhaps selected via an additional keyboard modifier. I’ve already “borrowed’ the option key for “toggle align to grid behavior” while adding (with the shift key) and dragging (with the meta key) layout tracks objects so I’ll have to use something other than that. Since the “meta” key (different for macOS vs Windows) is used to move stuff and the shift to add… that leaves the non-“meta”, non-shift and non-option key… (not that I know how to determine what that is... (yet)).

I’d prefer that the behavior be that clicking on an object select it and clicking off an object starts a (drag) selection rectangle. Having the shift key down would toggle select/deselect objects. 

Click dragging endpoints would move them (resizing the object without rotation); with the command key down it would rotate it.

Or: selecting an object would display a selection rectangle around that object. Click-dragging in the object would move it and click-dragging the corners would rotate it.

Thoughts?

danielb987, 2017-10-16:
For experienced users different key combinations works fine. For newbies there is two good options: strictly follow the operating system standard (Windows standard, Linux standard, Mac standard) or give the user the option to choose in menues.

I personally thinks it's more important to get a consistent user interface thru JMRI that follows the operating system standard. And while doing that, removing the use of the meta key. Since the meta key doesn't exist, it causes confusion, at least for new users and it makes it hard to remember.

My thought is that if more key combinations are added now, it will be even more difficult to do the change to get a consistent user interface. At least for the users thas have to learn yet Another key combination.

geowar1, 2017-10-16:
I assume the “meta” key on Windows is ether the control key or the Windows key. On macOS it’s definitely the command key. In ether case it definitely exists. JMRI (Layout Editor) uses it to move stuff (by meta-dragging it). (And has been the case since long before I started working on JMRI.)

JMRI modifier key useage is definitely not macOS standard. Which is (indirectly) my point. I may have my preference (macOS) but as a realist I know I have to be OS agnostic. 

If I want to avoid overloading additional modifier combinations the only solution I can think of would be to in addition of displaying the selection rectangle (as JMRI does now) also allow that box to be rotated by the corners (by meta-dragging (again, a JMRI thing; not my or macOS thing)).

danielb987, 2017-10-16:
The meta key is broken on Java 9 on both Windows 10 and Linux. But you can use my helper function isMetaDown in the LayoutEditor class. My function uses the right mouse button as the meta key on Windows and Linux.

Se #4246 and #4266.

bobjacobsen, 2017-10-16:
I think the starting point is to have LayoutEditor use the standard Java tools for recognizing drag operations:  `Toolkit.createDragGestureRecognizer()` et al.  Then, if it's really needed, add the standard modifiers to that.

geowar1, 2017-10-17:
Oh the subject of “inconsistent user interface”: currently background Images, fast clocks,turntables and reporter label are added via the Options menu. They then ether prompt for a location or place it in the center of the layout where you then have to drag them where you want them.

I propose that these be moved out of the Options menu and instead added to a popup menu along with “Add <selected track type from toolbar>” that shows up when you right-, or control- click on the layout editor view where there isn’t already an object (otherwise you’ll get the popup for that object instead).
This would provide a consistent UI for all parts and free up the shift key for other UI needs (like rotate perhaps? ;-).

geowar1, 2017-10-17:
BTW: I played around with createDragGestureRecognizer.  I added it to the. Layout Editor, installed the. Listener and implemented the appropriate callback methods. But I can’t seem to get the callback method to fire.. Ether I don’t know the (magic?) gesture or the already Installed mouse listener handlers are interfering with the gesture recognizer. Any clues?

danielb987, 2017-10-17:
> thread [2].
> Links:

Here is a working example that I used for my experiments:

http://www.java2s.com/Code/JavaAPI/java.awt.dnd/implementsDragSourceListener.htm

On the page, below the class MainClass, there is a class DragLabel which 
has a call to createDefaultDragGestureRecognizer in its constructor.

If you run this code, you will get a frame with a label and a button. If 
you drag the label to the button, the button will get the label:s text. 
Press the button to get the label back again. The label itself is never 
moved while dragging in this example, but the mouse cursor changes while 
dragging.

------------

The change I did to get rid of the threshold was to do my own 
DragSource.createDefaultDragGestureRecognizer method that creates an 
instance of my own MouseDragGestureRecognizer class.

Regards,
Daniel

2017-10-17 10:50 skrev George Warner:

danielb987, 2017-10-17:
> added to a popup menu along with “Add ” that shows up when you

It's a good idea. But what happens if there are items all over the page? 
Like a big background image that fills the entire space. On Windows, 
right click on an item gives a popup menu of that particular item. And 
if there is items all over the page, like a big background image, there 
is no empty space to click on to get the popup menu for adding these 
items. And it would be very confusing if clicking on one item brings up 
a popup menu there you can add other items to the page.

One possible solution is that you select for example a turntable on the 
main menu, it doesn't get placed on the page immediately, but follows 
the mouse when you move the mouse. You then has to left click the mouse 
on the spot there you want the turntable. If you want to abort, you 
press the <Esc> key.

Regards,
Daniel

2017-10-17 10:44 skrev George Warner:

geowar1, 2017-10-17:
Thanks! That definitely helps.
Somehow my brain didn't register "Drag" as part of "drag-n-drop". I was thinking about drag moving objects in the (Layout Editor) view.
Once the caffeine kicked in it's use case was obvious: I've been writing macOS DND code for years.
Duh! ;-)
So, unless I want to re-write the toolbar as a palette (which I don't… perhaps…) I don't know how the DragGestureRecognizer is going to help... :(

geowar1, 2017-10-17:
Yeah, background images would mess with the contextual-menu idea… unless I add the "Add..." items above the background image popup items? IOW: Contextually-clicking on a background image would popup a dialog that includes both the "add <object selected from toolbar>", "Add <XXX>" for the items currently in the "Options" menu and then the background image specific options?

geowar1, 2017-10-17:
Then again, if I did add a tools palette it would free up the (round) radio button choices from the toolbar AND the "shift to add" modifier key. Plus I could add other edit modes (move, resize (scale), rotate, measure (ruler), shapes (paint?), etc.) there and avoid the need for modifiers altogether.

That in combination with using the object bounds to scale (by dragging the edges) & rotate (by dragging the corners) the onscreen objects would mean I don't need another modifier key. Click-dragging the center of the object could move it (so no "meta" key there also).
I'd probably keep the options key to mean invert the "add/move align to grid behavior" (I'd also use it to invert the "rotate octagonally" behavior).
Thoughts?

rhwood, 2017-10-17:
> On Oct 17, 2017, at 05:50, George Warner <notifications@github.com> wrote:
> 
> Then again, if I did add a tools palette it would free up the (round) radio button choices from the toolbar AND the "shift to add" modifier key. Plus I could add other edit modes (move, resize (scale), rotate, measure (ruler), shapes (paint?), etc.) there and avoid the need for modifiers altogether.
> 
> That in combination with using the object bounds to scale (by dragging the edges) & rotate (by dragging the corners) the onscreen objects would mean I don't need another modifier key. Click-dragging the center of the object could move it (so no "meta" key there also).
> I'd probably keep the options key to mean invert the "add/move align to grid behavior" (I'd also use it to invert the "rotate octagonally" behavior).
> Thoughts?
> 
You could also put a tree of questionable objects (the results of checks) in a palette where it is always visible.

geowar1, 2017-10-17:
On Oct 17, 2017, at 2:55 AM, Randall Wood <notifications@github.com> wrote:
> You could also put a tree of questionable objects (the results of checks) in a palette where it is always visible.

Not a bad idea… I didn’t really like putting them as additional menus but at the time didn’t see any alternative (that wasn’t a _LOT_ more work that it was worth).

All in all the “checks” stuff turned out better than I had expected… (even with the in-menu results). I’ve gotten a few offline suggestions of other things to check (some over my head… I haven’t gotten my head completely around all the Logix & SSL (SML?), or enter/exit (?) code to know it well enough to write checks for it).

I think my next “check” would be an option to (auto-?)repair un-blocks tracks adjacent to a blocked track (with that block).

Or after selecting a set of (contiguous) un-blocked tracks an option to assign a block to all of them.
And/or the ability to assign a block to all currently selected tracks.
-- 
Enjoy,
George Warner
eMail/iChat: <geowar1@mac.com>

geowar1, 2017-10-17:
https://github.com/JMRI/JMRI/issues/4273

closed 2021-04-14 by stale[bot]