JMRI/JMRI #4268: Regression: Set sensors option missing from anchor point popup menu

pnbgit, 2017-10-16:
I'm seeing this on Windows 10,   JMRI.4.9.5+Rb2db16, set sensors menu is missing from anchor point popup, it should be present when the anchor point is a block divider. 

I'm pretty sure it was there on the previous 4.9.5 set of files, so seems to be a recent regression. Sensors previously set on anchor points still seem to work ok, but you cannot add new ones or edit settings for existing ones.  Set sensors is still there on the popup menu for turnouts and buffer stops/end bumpers

[image]

geowar1, 2017-10-16:
On Oct 16, 2017, at 2:52 PM, Pete Brownlow <notifications@github.com> wrote:
> I'm seeing this on Windows 10,   JMRI.4.9.5+Rb2db16, set sensors menu is missing from anchor point popup, it should be present when the anchor point is a block divider.
> I'm pretty sure it was there [...]

I’m not seeing it in the code for anchors… but it’s there for end bumpers… easy enough to copy-n-paste it to the anchor section…

I don’t remember explicitly removing it… but I may have inadvertently done so… :(

I’ll get a PR up asap.

-- 
Enjoy,
George Warner
eMail/iChat: <geowar1@mac.com>

pnbgit, 2017-10-28:
I thought I saw an email in the developers list saying this one was fixed, but I can't see a PR here for it. Set sensors is still missing from block divider anchor points as of 4.9.6ish package build #1976, 27th Oct 2017,

dsand47, 2017-10-28:
This problem was introduced by commit f4fc88f on Aug 9th.  The affected class is PositionalPoint.

Dave Sand

pnbgit, 2017-10-29:
IMHO this one should be high priority, unless the user is prepared to edit the XML, then NX routing is pretty much unusable with this bug.

Just to confirm it was there in 4.8, here is a screenshot showing "Set sensors" from 4.8:

[image]

pnbgit, 2017-10-29:
See also #4329, where you can't set NX sensors at turnouts either.

dsand47, 2017-10-29:
It appears that George is rather busy at the moment.  I have a fix for this problem which I will create a pull request for shortly.

Dave Sand

dsand47, 2017-10-29:
PR #4337 has been created.

Dave Sand

geowar1, 2017-10-29:
See PR https://github.com/JMRI/JMRI/pull/4339

pnbgit, 2017-10-30:
Tested on package build # 1987 - set sensors is now back on anchor point menu and tested working.

closed 2017-10-30 by pnbgit